bluez/bluez #552: [5.68] Xbox Series X controller Failed to pair: org.bluez.Error.AuthenticationCanceled

marvix97, 2023-07-03:
I'm trying to pair an xbox series x controller and I get this error when I do "pair MAC" with bluetoothctl. 

Distro
`Arch Linux`
uname -r
`6.4.1-zen1-1-zen`

This is what I get in journalctl when trying to pair

> lug 03 12:02:08 marvix-pc kernel: Bluetooth: hci0: unexpected cc 0x0c7c length: 1 < 3
lug 03 12:02:09 marvix-pc bluetoothd[695]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
lug 03 12:02:09 marvix-pc bluetoothd[695]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
lug 03 12:02:09 marvix-pc bluetoothd[695]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
lug 03 12:02:09 marvix-pc bluetoothd[695]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
lug 03 12:02:09 marvix-pc bluetoothd[695]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error

And in dmesg

> [ 1217.328347] Bluetooth: hci0: unexpected cc 0x0c7c length: 1 < 3
[ 1220.151628] Bluetooth: hci0: unexpected cc 0x0c7c length: 1 < 3

I can upload the full dmesg if needed. 

I'm also providing the full btmon log created with 'btmon -w'  [btmon_log.zip](https://github.com/bluez/bluez/files/11935747/btmon_log.zip)
but I think the relevant part is this

>HCI Event: Number of Completed Packets (0x13) plen 5                                          #69 [hci0] 20.668049
        Num handles: 1
        Handle: 17 Address: 44:16:22:72:AE:CB (Microsoft Corporation)
        Count: 1
        Latency: 537 msec (83-537 msec ~338 msec)
HCI Event: Disconnect Complete (0x05) plen 4                                                  #70 [hci0] 20.668713
        Status: Success (0x00)
        Handle: 17 Address: 44:16:22:72:AE:CB (Microsoft Corporation)
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                          {0x0001} [hci0] 20.668775
        LE Address: 44:16:22:72:AE:CB (Microsoft Corporation)
        Reason: Connection timeout (0x01)
@ MGMT Event: Command Complete (0x0001) plen 10                                            {0x0001} [hci0] 20.668798
      Pair Device (0x0019) plen 7
        Status: Failed (0x03)
        LE Address: 44:16:22:72:AE:CB (Microsoft Corporation)

I already tried to manually set the agent, as mentioned by someone on an old issue, but it didn't work. 

If you need any other log just ask me and I will provide.

ichijiyujin, 2023-07-18:
I've been struggling with this org.bluez.Error.AuthenticationCanceled error myself for a while. I've tried downgrading bluez to 5.66, 5.64 and even 5.60. I've tried switching the kernel lts, nothing seemed to work. For me, it turned out it's conflicting with my bluetooth mouse. If the mouse is connected already, the controller refuses to connect. If I disconnect the mouse, the controller connects normally, and then I can reconnect the mouse too without any issues. 

This didn't use to be an issue in the past, and I honestly can't tell when it started.

marvix97, 2023-07-18:
>For me, it turned out it's conflicting with my bluetooth mouse. If the mouse is connected already, the controller refuses to connect. If I disconnect the mouse, the controller connects normally, and then I can reconnect the mouse too without any issues.

I tried that too, because I have a bluetooth mouse too, but haven't had much luck. Also, on a fresh install, I couldn't even pair the mouse itself.
My workaround was to boot an unupdated live iso of fedora 37, pair the devices and then transplant the keys from /var/lib/bluetooth to my arch installation.

This made the devices connect without issues, so the problem is in the pairing function somewhere I guess.

> 
> This didn't use to be an issue in the past, and I honestly can't tell when it started.

As I said, an unupdated iso of fedora 37 works fine, but then when updated the problem reappears, this could be a starting point.

ichijiyujin, 2023-07-18:
> pair the devices and then transplant the keys from /var/lib/bluetooth to my arch installation

In my case I had the pairing working before, for a long time, but didn't use the controller anymore in recent months. And when I tried to connect it again now, it just refused to connect. So I though if I just remove the device and re-pair it from scratch, it will solve the hiccup. But it just got stuck at org.bluez.Error.AuthenticationCanceled. 

I might try removing and reconnecting all my devices, to see if there's any other odd behavior or if it changes anything. And I'll try to take a look at some older live ISO. Having to disconnect and reconnect my mouse every time I want to use my controller is annoying.