bluez/bluez #554: [5.68] Audio stops when playing multiple sources and during Volume change

ShiroYasha139, 2023-07-04:
When I play audio from say a music player and from another app or browser, the audio just stops and interferes unless I only keep it playing from one source.
And decreasing System Volume to 0% and increasing it results in audio loss. I have to reconnect my bluetooth device to get audio again. 

I've switched back to 5.66 where I have so such issues.

brianmsm, 2023-07-07:
Same problem here. If I'm listening to music or playing something on youtube and a telegram notification arrives for example, even though the bluetooth device doesn't disconnect, the audio is lost.

defunctl, 2023-07-07:
Are you both using Pipewire? 

It's likely to do with this issue: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3316

brianmsm, 2023-07-07:
Yes, I am using pipewire. In fact, when I reverted bluez to 5.66 I still had problems with losing audio when a telegram notification came in. Yes it looks like the problem could be pipewire from what they mention in that thread.

ShiroYasha139, 2023-07-08:
> Are you both using Pipewire?
> 
> It's likely to do with this issue: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3316

Yes I'm using pipewire. I thought downgrading bluez fixed it at first but I was having issues again so I was going to add info here again.
Thanks for sharing the issue. I'll downgrade pipewire itself and test out.

ShiroYasha139, 2023-07-09:
No issues with downgraded pipewire for a whole day, I'm closing the issue since its not related. Regardless I hope it helps people find the pipewire issue.

closed 2023-07-09 by ShiroYasha139