bluez/bluez #555: [5.68] Kernel panic (possibly) after connecting Logitech MX Keys (Lenovo IdeaPad 5 Pro 14ACN6)

holzbrenner, 2023-07-04:
The issue **does not happen with bluez 5.66** on the same system and with the same versions of all other packages, kernel version, Linux firmware, etc. With 5.68 it is always reproducible. Kernel panic happens soon after the GDM login screen is displayed, presumably after the MX Keys keyboard is connected, resulting in a blinking caps lock light on the notebook built-in keyboard and a frozen system. I am also using a Bluetooth mouse (MX Vertical) and Bose QC Earbuds II headphones - both of these devices have been paired previously and were turned on, waiting to be connected, when the problem happened. Perhaps it's one of these other BT devices that triggers the problem (before the system is able to write anything to the kernel log). On the other hand, it could be a BT controller related problem and the peripherals don't matter. I will investigate this further when I have the time to check which of these devices causes the crash.

The last entries in the **journalctl --dmesg** before the crash are:

```
jul 03 19:12:34 (hostname) kernel: input: MX Keys Keyboard as /devices/virtual/misc/uhid/0005:046D:B35B.0003/input/input17
jul 03 19:12:34 (hostname) kernel: input: MX Keys Mouse as /devices/virtual/misc/uhid/0005:046D:B35B.0003/input/input18
jul 03 19:12:34 (hostname) kernel: hid-generic 0005:046D:B35B.0003: input,hidraw1: BLUETOOTH HID v0.13 Keyboard [MX Keys] on 14:5a:fc:12:72:e4
```

OS: **Arch Linux, Gnome 44.2**

`Linux 6.4.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 01 Jul 2023 16:17:21 +0000 x86_64 GNU/Linux`
`linux-firmware 20230625.ee91452d-3`

**Hardware** : [Lenovo IdeaPad 5 Pro 14ACN6](https://wiki.archlinux.org/title/Lenovo_IdeaPad_5_Pro_14ACN6)

**inxi --bluetooth**
```
Bluetooth:
  Device-1: Qualcomm Atheros QCA61x4 Bluetooth 4.0 driver: btusb type: USB
  Report: bt-adapter ID: hci0 state: up address: 14:5A:FC:12:72:E4
```

**hwinfo --bluetooth**
```
04: USB 00.0: 11500 Bluetooth Device                            
  [Created at usb.122]
  Unique ID: eqBD.UCUpDou4XhB
  Parent ID: uIhY.DIk+WevO348
  SysFS ID: /devices/pci0000:00/0000:00:08.1/0000:04:00.4/usb3/3-4/3-4:1.0
  SysFS BusID: 3-4:1.0
  Hardware Class: bluetooth
  Model: "Qualcomm Atheros QCA61x4 Bluetooth 4.0"
  Hotplug: USB
  Vendor: usb 0x0cf3 "Qualcomm Atheros Communications"
  Device: usb 0xe300 "QCA61x4 Bluetooth 4.0"
  Revision: "0.01"
  Driver: "btusb"
  Driver Modules: "btusb"
  Speed: 12 Mbps
  Module Alias: "usb:v0CF3pE300d0001dcE0dsc01dp01icE0isc01ip01in00"
  Driver Info #0:
    Driver Status: btusb is active
    Driver Activation Cmd: "modprobe btusb"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
```

Changes of **/etc/bluetooth/main.conf** compared to the default values:
```
Disable = headset
MultiProfile = multiple
AutoEnable=true
```

Vudentz, 2023-07-05:
@holzbrenner are you sure you are running with the same kernel both 5.68 and 5.66? Perhaps this is related to some change in the input subsystem/driver, the diff from 5.66..5.68 is rather sort for input:

```
git diff 5.66..5.68 profiles/input/
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 021db386f3b7..7ff1ede3db35 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -374,6 +374,15 @@ static void report_value_cb(const guint8 *pdu, guint16 len, gpointer user_data)
                error("bt_uhid_send: %s (%d)", strerror(-err), -err);
 }
 
+static void report_notify_destroy(void *user_data)
+{
+       struct report *report = user_data;
+
+       DBG("");
+
+       report->notifyid = 0;
+}
+
 static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
                                        guint16 plen, gpointer user_data)
 {
@@ -393,7 +402,13 @@ static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
        report->notifyid = g_attrib_register(hog->attrib,
                                        ATT_OP_HANDLE_NOTIFY,
                                        report->value_handle,
-                                       report_value_cb, report, NULL);
+                                       report_value_cb, report,
+                                       report_notify_destroy);
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 021db386f3b7..7ff1ede3db35 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -374,6 +374,15 @@ static void report_value_cb(const guint8 *pdu, guint16 len, gpointer user_data)
                error("bt_uhid_send: %s (%d)", strerror(-err), -err);
 }
 
+static void report_notify_destroy(void *user_data)
+{
+       struct report *report = user_data;
+
+       DBG("");
+
+       report->notifyid = 0;
+}
+
 static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
                                        guint16 plen, gpointer user_data)
 {
@@ -393,7 +402,13 @@ static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
        report->notifyid = g_attrib_register(hog->attrib,
                                        ATT_OP_HANDLE_NOTIFY,
                                        report->value_handle,
-                                       report_value_cb, report, NULL);
+                                       report_value_cb, report,
+                                       report_notify_destroy);
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 021db386f3b7..7ff1ede3db35 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -374,6 +374,15 @@ static void report_value_cb(const guint8 *pdu, guint16 len, gpointer user_data)
                error("bt_uhid_send: %s (%d)", strerror(-err), -err);
 }
 
+static void report_notify_destroy(void *user_data)
+{
+       struct report *report = user_data;
+
+       DBG("");
+
+       report->notifyid = 0;
+}
+
 static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
                                        guint16 plen, gpointer user_data)
 {
@@ -393,7 +402,13 @@ static void report_ccc_written_cb(guint8 status, const guint8 *pdu,
        report->notifyid = g_attrib_register(hog->attrib,
                                        ATT_OP_HANDLE_NOTIFY,
                                        report->value_handle,
-                                       report_value_cb, report, NULL);
+                                       report_value_cb, report,
+                                       report_notify_destroy);
+       if (!report->notifyid) {
+               error("Unable to register report notification: handle 0x%04x",
+                                       report->value_handle);
+               goto remove;
+       }
 
        DBG("Report characteristic descriptor written: notifications enabled");
 
@@ -1798,7 +1813,11 @@ bool bt_hog_attach(struct bt_hog *hog, void *gatt)
                r->notifyid = g_attrib_register(hog->attrib,
                                        ATT_OP_HANDLE_NOTIFY,
                                        r->value_handle,
-                                       report_value_cb, r, NULL);
+                                       report_value_cb, r,
+                                       report_notify_destroy);
+               if (!r->notifyid)
+                       error("Unable to register report notification: "
+                               "handle 0x%04x", r->value_handle);
        }
 
        return true;

```

holzbrenner, 2023-07-06:
> @holzbrenner are you sure you are running with the same kernel both 5.68 and 5.66? Perhaps this is related to some change in the input subsystem/driver, the diff from 5.66..5.68 is rather sort for input:

@Vudentz Thanks for the quick reply. I will check this again and try to narrow the problem down to the specific peripheral that triggers the crash. Could be the headphones, I guess (as I speculate in my original issue description). 

However,  I think I only upgraded the three bluez packages, separately from anything else. I had version 5.66 "pinned" until the 5.67 headset issue was fixed and then, when I noticed that 5.68 is available, I upgraded the three bluez packages separately without anything else. I got kernel panic once login screen appeared after reboot, tried a few times and it always crashed, after which I managed to downgrade them back to 5.66 again and for this action I am quite certain I didn't downgrade or upgrade anything else while doing it; I just did a 'downgrade bluez bluez-libs bluez-utils' and everything works with 5.66 since then (I forgot how I prevented the crash to do the downgrade, either I turned off all BT devices or the service failed to start).

closed 2023-07-10 by holzbrenner