bluez/bluez #558: 5.67 and 5.68 make my audio stutter

sourproton, 2023-07-08:
After upgrading from 5.66, bluetooth audio didn't work correctly anymore when using my Sony XM4 or my JBL charge 2. The quality of the sound was very poor and it was constantly stuttering, both with A2DP and HandsFree profiles. Downgrading `bluez` and `bluez-lib` back to 5.66 solved the problem.

I don't know how to diagnose what the problem was specifically, but I'd be glad to provide more information if guided.

I'm running Arch with Gnome on a Lenovo Yoga Slim 6i gen 8.

```
$ lspci
00:00.0 Host bridge: Intel Corporation Device a707
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:04.0 Signal processing controller: Intel Corporation Device a71d
00:06.0 PCI bridge: Intel Corporation Device a74d
00:06.2 PCI bridge: Intel Corporation Device a73d
00:07.0 PCI bridge: Intel Corporation Device a76e
00:07.2 PCI bridge: Intel Corporation Device a72f
00:08.0 System peripheral: Intel Corporation GNA Scoring Accelerator module
00:0a.0 Signal processing controller: Intel Corporation Device a77d (rev 01)
00:0d.0 USB controller: Intel Corporation Device a71e
00:0d.2 USB controller: Intel Corporation Device a73e
00:0d.3 USB controller: Intel Corporation Device a76d
00:12.0 Serial controller: Intel Corporation Alder Lake-P Integrated Sensor Hub (rev 01)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Device 51f1 (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:15.1 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #1 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:1f.0 ISA bridge: Intel Corporation Device 519d (rev 01)
00:1f.3 Multimedia audio controller: Intel Corporation Device 51ca (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9B1 (rev 02)
```

Vudentz, 2023-07-17:
@sourproton here XM4 is working just fine, do you have any HCI trace? 5.67 is actually crashing so I assume you are using 5.68 already? Are you running with pipewire or pulseaudio, which distro?

sourproton, 2023-07-18:
Since the upgrade I have been running on 5.66 because of the bug. I never suspected the XM4 to be the problem, but rather the bluetooth chip that might be relatively new, given the launch date of the laptop.

After your comment I upgraded again to 5.68 to generate a HCI trace and found the quality is now normal. Probably another package was upgraded in the meantime and now everything is compatible again.

I'll close this issue and reopen it with more info in case I come across the bug again.

closed 2023-07-18 by sourproton