bluez/bluez #559: Connecting to bluetooth headset fails; Hands-Free Voice gateway failed

Daniel451, 2023-07-09:
I am running Manjaro linux on kernel 6.1 and I have installed `pulseaudio` as well as `pulseaudio-bluetooth`. W.r.t. to bluez I have `bluez` 5.66, `bluez-libs` 5.66, `bluez-utils` 5.66, and `bluez-tools` 0.2.0 installed.

I cannot remember changing anything with my audio setup or updating but I guess something did change because my headset (EDIFIER W820NB Plus) was working before and now I either get this error:

> bluetoothd[755]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 64:68:76:5A:E4:F1: Connection timed out (110)

or this error:

> bluetoothd[755]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 64:68:76:5A:E4:F1: Connection refused (111)

or both.

I tried installing `bluez-git`, I tried using `pipewire` instead of `pulseaudio`, I tried multiple bluez Github issues and also everything listed in the ArchWiki w.r.t. bluetooth headsets:
https://wiki.archlinux.org/title/bluetooth_headset#Pairing_works,_but_connecting_does_not

Unfortunately, nothing works. Does someone have any suggestions?

I am quite frustrated with the whole bluetooth experience in Linux tbh. I am running Manjaro with i3, so I am no stranger to config files but connecting a bluetooth headset in 2023 shouldn't be much of a hassle. I also have a Sennheiser Momemntum which actually never worked and cheap Burnester as a fallback solution, which also stopped working - so the problem is definitely not the headset.

Also the bluetooth adapter is definitely working because I have a dual-boot system and in Windows all 3 headsets just work after connecting within seconds.

cartok, 2023-10-13:
Have a similar issue, I am on 6.5.7-arch1-1.

Installed a network card and a new PSU yesterday. Removing the network card doesn't help and under windows everything works as before, I see erros in the bluetooth service. On restart I get:
```
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/faststream
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/faststream_duplex
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSink/opus_05
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/opus_05
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Okt 13 18:24:04 desktop bluetoothd[60468]: Endpoint registered: sender=:1.617 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Okt 13 18:24:10 desktop bluetoothd[60468]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 00:1B:66:BE:C5:89: Permission denied (13)
```

When I manually try to connect with the headset using `bluetoothctl` I get one more error:
````
Okt 13 18:25:03 desktop bluetoothd[60468]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 00:1B:66:BE:C5:89: Too many levels of symbolic links (40)
```
My device is detected, paired, no startup errors only some warnings which propably do not matter and I'm quite sure I had them before (_System does not support csip, micp, vcp, mcp, bass, bap)._

The audio device does not get created cause the connection is not working (nothing in `pw-dump` or `wpctl status`):
```
[bluetooth] connect 00:1B:66:BE:C5:89
Attempting to connect to 00:1B:66:BE:C5:89
[CHG] Device 00:1B:66:BE:C5:89 Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-canceled
[CHG] Device 00:1B:66:BE:C5:89 Connected: no
[CHG] Device 00:1B:66:BE:C5:89 Connected: yes
[CHG] Device 00:1B:66:BE:C5:89 Connected: no
[CHG] Device 00:1B:66:BE:C5:89 Connected: yes
[CHG] Device 00:1B:66:BE:C5:89 Connected: no
``` 

I did not update my system but I updated it today to see if it helps to fix the issue that came out of nothing. - I also changed my PSU but I think I still had sound yesterday. Did I get hacked?... `rkhunter` and `clamav fangfrisch` are looking fine. The headset (Sennheiser PXC 550-II) works fine as be before under Windows.

I don't know if it is related but suddenly the pulseaudio modules `module-always-sink` and `module-switch-on-port-available` can not get loaded anymore, create an error in `pipewire-pulse.service`.
```
pulse.cmd = [
  { cmd = "load-module" args = "module-always-sink" }  # Error :File exists
  { cmd = "load-module" args = "module-switch-on-connect"   } # No Error
  { cmd = "load-module" args = "module-switch-on-port-available" } # Error: No such file or directory
]
```

I already talked to ChatGPT for a while and find no solution. 

Could it be that my headset firmware got auto updated by the Sennheiser Smart Control app that runs on my smartphone and now something is off?

Daniel451, 2023-10-13:
@cartok I tried everything that came to my mind and also had pages over pages with ChatGPT but nothing worked. I would say either the bluez library or linux bluetooth hardware drivers are not really stable.

Under Windows there was no problem whatsoever.

I could only resolve the issue by buying another bluetooth adapter. I am using an ASUS USB-BT500 Bluetooth 5.0 USB Adapter now. That one works well so far, apparently the drivers for this one are better supported.

cartok, 2023-10-13:
@Daniel451 Thanks for info. My internal bluetooth device worked perfectly fine with LL, HFP-HF everything. Much better than the Creative BT-W3 I used on a notebook before (total mess, don't buy Creative stuff)

cartok, 2023-10-13:
@Daniel451 Should I move this to a new ticket as your issue is kinda fixed by changing hardware and you no longer need the topic?

Daniel451, 2023-10-13:
@cartok good question, I think for the matter of consistency I would leave it here since the issue seems similar / same, I just surrendered..

> Is HFP working well with the ASUS USB-BT500 and does it run aptX LL? Low latency and HFP is a "must-have" I think. aptX HD would be a bonus for music for sure but my headset does not support it anyways.

HFP is working fine. How could I check aptX LL? On Windows this is working, I even checked latency manually on Windows by doing sound reaction tests (you press a button when you hear a sound) with myself and 2 other people.

On Linux I just shortly tested but I would say on Linux there is a delay in comparison to Linux. I did not bother to check it as detailed as on Windows though since I use Windows-only for gaming and Linux for working and everything else, thus I have no application on Linux that would require extra low latency.

cartok, 2023-10-16:
@Daniel451 Thx for the infos. Ok sounds like it is LL. It would be possible to check details via pw-dump I suppose but never mind.

I have it working again...

I have to stop scanning, otherwise it could not connect (connected and directl disconnected afterwards.
Not sure if this is intended or a bug.

pma1, 2023-12-26:
I met the same issue:
[bluetooth]# connect 00:1B:66:1C:71:8D 
Attempting to connect to 00:1B:66:1C:71:8D
[bluetooth]# hci0 00:1B:66:1C:71:8D type BR/EDR connect failed (status 0x08, Timeout)
[bluetooth]# hci0 00:1B:66:1C:71:8D type BR/EDR connected eir_len 10
[CHG] Device 00:1B:66:1C:71:8D Connected: yes
[HD 250BT]# hci0 00:1B:66:1C:71:8D auth failed with status 0x05 (Authentication Failed)
[HD 250BT]# Failed to connect: org.bluez.Error.Failed br-connection-unknown
[HD 250BT]# hci0 00:1B:66:1C:71:8D type BR/EDR disconnected with reason 2
[CHG] Device 00:1B:66:1C:71:8D Connected: no
[bluetooth]# hci0 00:1B:66:1C:71:8D type BR/EDR connect failed (status 0x04, Connect Failed)

pma1, 2023-12-26:
kernel: 6.6.8-200.fc39.x86_64
gnome-bluetooth-libs-42.7-1.fc39.x86_64
gnome-bluetooth-42.7-1.fc39.x86_64
bluez-5.71-1.fc39.x86_64
bluez-libs-5.71-1.fc39.x86_64

dal00, 2024-01-02:
@pma1 I had the same issue but with bluez-5.71-2
What solved it for me was to remove the paired headphones and re-pair.

pma1, 2024-01-03:
@dal00 Thanks, just tried bluez-5.71.2, it's fixed.

seamusdemora, 2024-04-09:
I know this isn't a helpful comment, but may be useful for setting expectations. If you spend a week or so chasing Bluetooth problems on a Linux system, you begin to wonder: "Does Bluetooth on Linux just suck?" Unfortunately, I think the answer may be, "Yes, it does suck." I've played audio thru a BT speaker (JBL Flip 5) using my iPhone6 as the music source & speaker control continuously for days without an interrupt. I can occasionally manage an hour or two using Linux with the same Flip 5 speaker. It just sucks - so deal with it.

cartok, 2024-04-10:
> I know this isn't a helpful comment, but may be useful for setting expectations. If you spend a week or so chasing Bluetooth problems on a Linux system, you begin to wonder: "Does Bluetooth on Linux [...]

Are we on reddit?

cartok, 2024-04-24:
I had to create a custom wireplumber config for my headset to make it work ok. Also switched from USB stick to Onboard Intel radio.

Daniel451, 2024-04-24:
@cartok that was actually also the case for me with an old headset I had. Currently, I am using an `Edifier WH950NB`, this worked out-of-the-box without any additional config, just paired and works. Also it connects automatically once the computer is started and I start the headset. Has really nice sound quality, very soft cushioning, and good ANC also.

Ghostbird, 2024-08-07:
This is a real shame.‌ I have the same issue with my TP-Link USB adapter and HUAWEI FreeBuds i6. Pairing works fine, but once the device loses connection for the first time after pairing, it cannot be reconnected. Every attempt to connect gives this error.

It must be removed and re-paired. Which is a too convoluted process to repeatedly do, especially, as it involves removing the buds and putting them back in their case, and even then it's not guaranteed to succeed and often needs several attempts.

For what it's worth, [here's the log](https://gist.github.com/Ghostbird/eed05440bd3211b6eae962b0d017355b) from `bluetoothd`

seamusdemora, 2024-08-07:
> This is a real shame.‌ I have the same issue with my TP-Link USB adapter and HUAWEI FreeBuds i6. Pairing works fine, but once the device loses connection for the first time after pairing, it cannot be reconnected. Every attempt to connect gives this error.

Wow - that's pretty awful! 

But perhaps @Daniel451 has an answer for that issue??  Maybe it's because your hardware isn't on the "community-supported list of hardware"... then again, it could be that there are traces of reddit-like language that are holding things up? 

:-)