bluez/bluez #561: [BlueZ 5.68 Mesh] Fails to receive messages, and send Mesh messages with Kernel 6.2

venngar, 2023-07-17:
In the new release, BT Mesh communication seems to have some serious problems.

1. First of all, it appears callbacks aren't registered in mesh-io.c, which causes Raspberry Pi as a Mesh device to not receive any messages (more details and logs here) https://github.com/bluez/bluez/issues/540#issuecomment-1625719064
2. I've managed to apply the fix locally (see below), but there's another problem. For RPi based on Kernel < 6.1.0, BlueZ successfully sends and receives messages, it's working well. But for Kernel 6.2.0 (Ubuntu 23.04), it doesn't send any messages although the logs show otherwise (I used nRF Bluetooth sniffer to check for Mesh messages and nothing is coming out). Here's the log of sending message:

```
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/node.c:send_call() Send
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/model.c:mesh_model_rx() iv_index 00000000 key_aid = 48
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.934 Used App Key: f43fcd16708552dc056bead675a0b79a
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.934 Clr Rx: c6aeaa0e0000000000
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.935 Clr-Net Tx: 0003002c761281c00048879c0f2a50b94a6becabce19f200000000
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.936 RX: Network [enc] :: 4456667d831cf45f0034ff5ddb4d24059a1a52643a2570acd0bf80
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.936 RX: Network [clr] :: 4403002c761281c00048879c0f2a50b94a6becabce19f2
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.942 Mesh Send Complete: 0e
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.982 Mesh Send Start: 2a4456667d831cf45f0034ff5ddb4d24059a1a52643a2570acd0bf80
Jul 17 15:07:52 gateway401-u23 bluetooth-meshd[12822]: mesh/util.c:print_packet() 02872.989 Mesh Send Complete: 0f
```

Fix used for problem 1, in mesh-io-register-recv-cb function of mesh-io.c:

```
bool res = false;

if (io && io->api && io->api->reg)
     res = io->api->reg(io, filter, len, cb, user_data);

l_queue_push_head(io->rx_regs, rx_reg);

return res;
```

venngar, 2024-10-16:
Any updates on this issue?

badrbouslikhin, 2024-11-06:
I am experiencing the same issue. Is there any known workaround?

closed 2025-08-08 by github-actions[bot]