bluez/bluez #564: Running bluez 5.68 on Raspbian Buster with an older LIBC6 and GLIBC

skast96, 2023-07-26:
It is possible to compile the bluez package with an older version of LIBC, however is it clever to do that or should i consider a dist-upgrade because the older libc is missing some edge cases where the bluez stack is missing shared files?

Vudentz, 2023-07-26:
@skast96 don't think we added any new dependencies, so it is worth checking if it compiles with older libc then I assume you should be fine.

skast96, 2023-08-01:
@Vudentz thank you for your answer it worked installing bluez. However I need a debian package to make the installation process as light as possible do you guys have a script that builds a debian file out of the compiled bluez folder or is this managed by debian itself ?

closed 2023-08-02 by skast96