bluez/bluez #569: Not able to Scan reference device after connecting to an LE device

NJNXP, 2023-08-02:
**Description:**
Linux Bluez DUT stops scanning after connecting to a LE device

**Steps:**
bluetoothctl
	- scan on
	- pair/connect

**Observation:**
	- During connecting the host disables the scan, and later this scan is not automatically enabled. (btmon log)
	< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2  
	        Scanning: Disabled (0x00)
	        Filter duplicates: Disabled (0x00)
	- If user try to execute scan on command following error message is displayed :
	Failed to start discovery: org.bluez.Error.InProgress
	- User has to stop the scan and restart again to connect other devices

**Query:** Is this an expected behavior? In case if scan is disabled then and is not enabled automatically then the scan on command should be executed without error.

[btmon.txt](https://github.com/bluez/bluez/files/12236252/btmon.txt)

dezi, 2023-08-12:
Hi,

i believe that is the same problem as in my issue #572.

The problem is not because of connecting a device, since the
problem is present even w/o connecting anything.

https://github.com/bluez/bluez/issues/572

Regards,
dezi

Vudentz, 2023-09-06:
> **Description:** Linux Bluez DUT stops scanning after connecting to a LE device
>
> **Steps:** bluetoothctl - scan on - pair/connect
>
> **Observation:** During connecting the host disables the scan, and [...]

While connecting we do suspend active scanning since the connection procedure on LE do involve scanning before connecting, once that is completed it shall resume scanning again though.

NJNXP, 2023-11-09:
Hello @Vudentz , 
Following patch added by you is not yet merged. Can you please help to look into it. 
https://patchwork.kernel.org/project/bluetooth/patch/20230816190500.2732009-1-luiz.dentz@gmail.com/

Thanks
Nitin Jadhav