bluez/bluez #570: Writing characteristic of length 512 fails

bp1001, 2023-08-03:
Writing a characteristic of length 512 fails. If I interpret the Bluetooth spec correctly (3.2.9, part F), characteristics (attribute values) of size 512 should be allowed.

It seems that the write fails at write_cb in shared/gatt-server.c at:
`ecode = check_length(length, 0);`

In this case the length passed, is not the length of only the attribute value, but is it also includes the length of the attribute handle.
A similar situation occurs in prep_write_cb(), where the length includes both the attribute handle and offset length.

I fixed this by subtracting the attribute handle length (and offset length) before passing it to check_length():
in write_cb:
`ecode = check_length(length-2, 0);`

in prep_write_cb():
`ecode = check_length(length-4, 0);`

Now writes of 512 length are handled successfully.

Vudentz, 2023-08-03:
@bp1001 are you planning to send a proper patch containing these changes? (see https://github.com/bluez/bluez/blob/master/HACKING#L98)

bp1001, 2023-08-04:
I will send a patch for these changes. Thanks for the link.

closed 2023-08-12 by bp1001