bluez/bluez #571: Hid over Gatt (HOG) depublishing HOG service on dBus.

dezi, 2023-08-12:
Hi,

the HOG module is depublishing the HOG Service and its Charateristics from the dBus Interface.

Why is this necessary?

Background:

I need to attach a Bluetooth Voice Remote, which supplies a HOG service.
This device has additional Report Characteristics, which supply the voice
samples.

When the bluetooth daemon  recognizes and attaches the remote,
I have no otherwise access to the corresponding Report Characteristics
for the voice control and voice samples.

Regards,
Dezi

Vudentz, 2023-09-06:
@dezi not sure exactly what you are after, HOG has a dedicated plugin which claims the service thus why it is not exported, this is done to avoid conflicts with application attempting to operate the attributes, you can however disable the plugin to be able to handle its attributes above, that said HID reports is normally handle by the kernel via HID drivers, so perhaps you need to develop a custom driver if the traffic actually happens over HID.

closed 2023-09-07 by Vudentz

dezi, 2023-09-08:
@Vudentz as i stated, i have a bluetooth remote control with voice. The HOG plugin claims the HOG service and it's characteristics, leaving the Report Charactaristics handling the voice output unreflected. So voice is not working.

I could easily handle the particular characteristic myself, however the whole service is unpublished from dBus.

The original question was: Why is it necessary to unpublish/hide all characteristics of the HOG service?

On the other hand, as i stated on issue #563, disabling the HOG service in configure yields a compile error.

Regards, dezi