bluez/bluez #572: Discovery ceasing to deliver results after short time

dezi, 2023-08-12:
Hi

bluetoothctl scan ceases to deliver results after a short time.

As You can see from the log, after deleting discovered devices
after the delete timeout period, no new advertisments are
discovered.

The controller is still in scan mode.

The full log, including a btmon dump at the same time:

[scratch_108.txt](https://github.com/bluez/bluez/files/12326978/scratch_108.txt)

As You can see from the btmon log at end, the follwing command is issued:

< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                       #150 [hci0] 58.165001
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)

which finally terminates scanning...

See the following log:

=======================
gen8705:~$ hcitool
hcitool - HCI Tool ver 5.68

=======================
gen8705:~$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 84:1B:77:E2:ED:5B  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING
	RX bytes:51544278 acl:255 sco:0 events:1090421 errors:0
	TX bytes:1215549 acl:242 sco:0 commands:36940 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: PERIPHERAL ACCEPT
	Name: 'liesa-gen8705'
	Class: 0x6c0104
	Service Classes: Rendering, Capturing, Audio, Telephony
	Device Class: Computer, Desktop workstation
	HCI Version: 5.2 (0xb)  Revision: 0x237e
	LMP Version: 5.2 (0xb)  Subversion: 0x237e
	Manufacturer: Intel Corp. (2)

=======================
gen8705:~$ bluetoothctl
Agent registered

[CHG] Controller 84:1B:77:E2:ED:5B Pairable: yes
[bluetooth]# version
Version 5.68
[bluetooth]# scan on
Discovery started
[CHG] Controller 84:1B:77:E2:ED:5B Discovering: yes
[CHG] Device D8:0F:99:31:37:5C RSSI: 0xffffffb4 (-76)
[DEL] Device 76:04:8A:D4:AD:69 76-04-8A-D4-AD-69
[DEL] Device 90:98:77:93:24:F2 Susi's Schlafzimmer
[CHG] Device 08:BF:08:5F:59:93 RSSI: 0xffffffc0 (-64)
[CHG] Device D8:0F:99:31:37:5C RSSI: 0xffffffbf (-65)
[CHG] Device D8:0F:99:31:37:5C TxPower: 0x0007 (7)
[CHG] Device 5D:E1:24:8C:5A:D0 RSSI: 0xffffffc1 (-63)
[CHG] Device 5D:E1:24:8C:5A:D0 TxPower: 0x000c (12)
[CHG] Device 0C:96:E6:27:B7:AC RSSI: 0xffffffac (-84)
[NEW] Device 90:98:77:93:24:F2 Susi's Schlafzimmer
[CHG] Device 0C:96:E6:27:B7:AC TxPower: 0x000a (10)
[NEW] Device FE:F7:54:71:C4:F3 iBKS Plus
[NEW] Device 1C:D6:BE:E3:0B:2A TV
[CHG] Device 90:98:77:93:24:F2 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 90:98:77:93:24:F2 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 90:98:77:93:24:F2 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 90:98:77:93:24:F2 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 90:98:77:93:24:F2 UUIDs: 00000000-0000-0000-0000-000000000000
[NEW] Device 76:04:8A:D4:AD:69 76-04-8A-D4-AD-69
[DEL] Device E4:17:C6:E6:62:A7 iBKS Plus
[NEW] Device ED:EB:1D:74:C3:68 ED-EB-1D-74-C3-68
[NEW] Device E4:17:C6:E6:62:A7 iBKS Plus
[CHG] Device 5D:E1:24:8C:5A:D0 RSSI: 0xffffffb6 (-74)
[CHG] Device 08:BF:08:5F:59:93 RSSI: 0xffffffb6 (-74)
[NEW] Device 84:EB:18:5F:35:AA SANITAS SBF70
[CHG] Device 08:BF:08:5F:59:93 RSSI: 0xffffffae (-82)
[DEL] Device 1C:D6:BE:E3:0B:2A TV
[CHG] Device 5D:E1:24:8C:5A:D0 ManufacturerData Key: 0x004c (76)
[CHG] Device 5D:E1:24:8C:5A:D0 ManufacturerData Value:
  10 05 4a 1c c3 1f 65                             ..J...e
[CHG] Device FE:F7:54:71:C4:F3 UUIDs: 0000feaa-0000-1000-8000-00805f9b34fb
[CHG] Device FE:F7:54:71:C4:F3 ServiceData Key: 0000feaa-0000-1000-8000-00805f9b34fb
[CHG] Device FE:F7:54:71:C4:F3 ServiceData Value:
  10 ef 03 61 63 63 65 6e 74 2d 73 79 73 74 65 6d  ...accent-system
  73 07                                            s.
[NEW] Device 1C:D6:BE:E3:0B:2A TV
[DEL] Device 84:EB:18:5F:35:AA SANITAS SBF70
[DEL] Device 08:BF:08:5F:59:93 08-BF-08-5F-59-93
[DEL] Device D8:0F:99:31:37:5C Dezi's Dominator-XL
[DEL] Device 76:04:8A:D4:AD:69 76-04-8A-D4-AD-69
[DEL] Device 5D:E1:24:8C:5A:D0 5D-E1-24-8C-5A-D0
[DEL] Device FE:F7:54:71:C4:F3 iBKS Plus
[DEL] Device 90:98:77:93:24:F2 Susi's Schlafzimmer
[DEL] Device 1C:D6:BE:E3:0B:2A TV
[DEL] Device 0C:96:E6:27:B7:AC Dezi's Gästezimmer groß
[DEL] Device E4:17:C6:E6:62:A7 iBKS Plus
[DEL] Device ED:EB:1D:74:C3:68 ED-EB-1D-74-C3-68
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.InProgress
[bluetooth]#

=================================
Last few lines from btmon dump:

> HCI Event: LE Meta Event (0x3e) plen 26                                                                             #149 [hci0] 54.161889
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x001a
            Props: 0x001a
              Scannable
              Scan response
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: SCAN_RSP to an ADV_IND (0x001a)
          Address type: Random (0x01)
          Address: 62:F1:71:DC:61:EC (Resolvable)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -91 dBm (0xa5)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x00
@ MGMT Event: Device Found (0x0012) plen 14                                                                       {0x0001} [hci0] 54.161980
        LE Address: 62:F1:71:DC:61:EC (Resolvable)
        RSSI: -91 dBm (0xa5)
        Flags: 0x00000004
          Not Connectable
        Data length: 0
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                       #150 [hci0] 58.165001
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Inquiry Complete (0x01) plen 1                                                                           #151 [hci0] 58.198477
        Status: Success (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                           #152 [hci0] 58.205938
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)

dezi, 2023-08-12:
Additional research:

the Start Discovery MGMT Command is repeated / refreshed every 16 seconds.

If the scan ceases to deliver, also the refresh is outstanding (not executed)

=> search for an error condition where the command is not issued.

Regards dezi

@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 745.832501
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
--
@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 761.831308
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
--
@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 777.828046
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
--
@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 793.830363
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
--
@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 809.830755
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
--
@ MGMT Command: Start Discovery (0x0023) plen 1      {0x0001} [hci0] 825.832062
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random

Vudentz, 2023-09-06:
Looks like a duplicated of #510, try with the following change: https://github.com/bluez/bluetooth-next/commit/52bf4fd43f759ac4698f041fff2cb27691698265

asterwyx, 2024-03-20:
I've also met a similar problem. When I run bluetoothctl scan on and try to connect to one found device in other desktop clients. (For example, add Bluetooth device in plasma's settings, just start pairing but never pair them, waiting for the process to timeout.) All devices are removed according to bluetoothctl's log except for the connecting one. I can only see the connecting one in `bluetoothctl devices`.