bluez/bluez #573: Scan with transport=auto delivers unreliable scan results.

dezi, 2023-08-12:
Hi,

i had trouble connecting a ble thermometer to my app under ubuntu/linux.

The device did not reliable show up in the discovery.

Changing the scan parameters from transport=auto to transport=le fixed the problem.

Problem now: I also have to discover br/edr devices at the same time.

Workaround solution:

Switching the DiscoveryFilter-Set from transport=le to transport=bredr
every 250 ms while the discovery is still active.

Interestingly, this workaround also fixes my problem from issue

https://github.com/bluez/bluez/issues/572

Regards, dezi

Excerpt from btmon log:

@ MGMT Command: Start Service Di.. (0x003a) plen 4  {0x0001} [hci0] 2463.114751
        Address type: 0x06
          LE Public
          LE Random
        RSSI: invalid (0x7f)

@ MGMT Command: Stop Discovery (0x0024) plen 1      {0x0001} [hci0] 2463.511880
        Address type: 0x06
          LE Public
          LE Random
< HCI Command: LE Set Extend.. (0x08|0x0042) plen 6  #134809 [hci0] 2463.511920

@ MGMT Command: Start Service Di.. (0x003a) plen 4  {0x0001} [hci0] 2463.516787
        Address type: 0x01
          BR/EDR
        RSSI: invalid (0x7f)
        UUIDs: 0

@ MGMT Command: Stop Discovery (0x0024) plen 1      {0x0001} [hci0] 2463.613490
        Address type: 0x01
          BR/EDR
< HCI Command: Inquiry Cancel (0x01|0x0002) plen 0   #134813 [hci0] 2463.613537
> HCI Event: Command Complete (0x0e) plen 4          #134814 [hci0] 2463.618589

dezi, 2023-09-04:
@jsmif 

I have a self developed long term go process running, which uses the go-package github.com/muka/go-bluetooth.

The package is based on the bluez dBus interfaces and has api methods to start/stop scanning.

Vudentz, 2023-09-06:
Looks like a duplicated of https://github.com/bluez/bluez/issues/510, try with the following change: https://github.com/bluez/bluetooth-next/commit/52bf4fd43f759ac4698f041fff2cb27691698265

closed 2025-08-07 by github-actions[bot]