bluez/bluez #575: [Congestion Environment] Invalid HCI Parameters(0x0d) is returned as status for Scan off, during executing Scan on/Scan off (BT/BLE) in a loop and after that Subsequent Scan(On/Off) not working.

NJNXP, 2023-08-14:
Steps:
1. [bluetoothctl] scan on
2. [bluetoothctl] scan off
Repeat step 1 and 2

Observation:
During one of the scan off command, **Invalid HCI Command Parameters** is sent for **Remote Name Request Cancel**. Later to this scan on/off will not work.

```
@ MGMT Command: Stop Discovery (0x0024) plen 1                            {0x0001} [hci0] 472.216442
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                       #2285 [hci0] 472.216511
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                  #2286 [hci0] 472.217130
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: Remote Name Request Cancel (0x01|0x001a) plen 6               #2287 [hci0] 472.217195
        Address: AC:D6:18:4C:9E:FE (OnePlus Technology (Shenzhen) Co., Ltd)
> HCI Event: Command Complete (0x0e) plen 10                                 #2288 [hci0] 472.217716
      Remote Name Request Cancel (0x01|0x001a) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
        Address: AC:D6:18:4C:9E:FE (OnePlus Technology (Shenzhen) Co., Ltd)
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 472.228893
      Stop Discovery (0x0024) plen 1
        Status: Invalid Parameters (0x0d)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
```

**********Scan On/Off stops working**********

```
@ MGMT Command: Stop Discovery (0x0024) plen 1                            {0x0001} [hci0] 531.399268
        Address type: 0x00
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 531.399281
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x00
@ MGMT Command: Stop Discovery (0x0024) plen 1                            {0x0001} [hci0] 547.223662
        Address type: 0x00
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 547.223675
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x00
```
Recovery: If we send hcitool reset command to controller then we can see **@ MGMT Event: Discovery: Disable**. This MGMT event will recover the issue.

```
~# hcitool cmd 0x03 0x003
< HCI Command: ogf 0x03, ocf 0x0003, plen 0
> HCI Event: 0x0e plen 4
  01 03 0C 00
```

Response in Btmon
```
@ RAW Open: hcitool (privileged) version 2.22                                    {0x0002} 601.429409
@ RAW Close: hcitool                                                             {0x0002} 601.429445
@ RAW Open: hcitool (privileged) version 2.22                             {0x0002} [hci0] 601.429510
< HCI Command: Reset (0x03|0x0003) plen 0                                    #2289 [hci0] 601.429650
> HCI Event: Command Complete (0x0e) plen 4                                  #2290 [hci0] 601.596447
      Reset (0x03|0x0003) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                 {0x0001} [hci0] 601.596501
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
@ RAW Close: hcitool                                                      {0x0002} [hci0] 601.596535
```

Query:
- Why do scan on/off stops working after Invalid HCI Command Parameters event from controller? 
- Recovery explained in the above issue is work around and not a solution.
- Need to check why scan on/off stops responding at first place?

NJNXP, 2023-08-16:
More precise way to reproduce this issue is, 
- Send `scan off `command immediately after observing the log `HCI Event: Command Complete ` for `Remote Name Request Cancel`

[image]

- scan on
- scan off

[image]

NJNXP, 2023-08-16:
After this issue has occurred then following logs are observed for scan on/off command

For `scan off` following logs are observed in bluetoothd and no logs in btmon
Bluetoothd
`bluetoothd[2104]: src/adapter.c:start_discovery() sender :1.68
bluetoothd[2104]: src/adapter.c:update_discovery_filter()
bluetoothd[2104]: src/adapter.c:discovery_filter_to_mgmt_cp()
bluetoothd[2104]: src/adapter.c:update_discovery_filter() filters were equal, deciding to not restart the scan.`

For `scan off` following logs are observed in btmon and bluetoothd
Btmon
`@ MGMT Command: Stop Discovery (0x0024) plen 1`
`Address type: 0x00  `                  
`@ MGMT Event: Command Complete (0x0001) plen 4`
`Stop Discovery (0x0024) plen 1`
`Status: Rejected (0x0b)`
`Address type: 0x00`

Bluetoothd
`bluetoothd[2104]: src/adapter.c:stop_discovery() sender :1.68
bluetoothd[2104]: src/shared/mgmt.c:send_request() [0x0000] command 0x0024
bluetoothd[2104]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0024 complete: 0x0b
bluetoothd[2104]: src/adapter.c:stop_discovery_complete() status 0x0b
bluetoothd[2104]: src/adapter.c:discovery_remove() owner :1.68
bluetoothd[2104]: src/adapter.c:discovery_free() 0xaaaaef28f730`

Vudentz, 2023-08-16:
@NJNXP what kernel version? We might as well ignore errors from Remote Name Request Cancel, btw do you have any explanation on why the controller is generation such error?

Vudentz, 2023-08-22:
@NJNXP could you please check with the above patch.

NJNXP, 2023-08-22:
@Vudentz, Yes, we have tested it and found to be working

closed 2023-08-22 by Vudentz

Sohan-kr, 2024-09-06:
Hi @Vudentz, this is regarding the above issue, you have raised a PR, but it seems the patch failed during CI/CV, so it wasn't merged into TOT. Could you please check why the patch failed and make the necessary corrections?

[image]

Vudentz, 2024-09-06:
> Hi @Vudentz, this is regarding the above issue, you raised a PR for this, but it seems the patch failed during CI/CV, so it wasn't merged into TOT. Could you please check why the patch failed during CI/CV?
> <img alt="image" width="945" src="https://private-user-images.githubusercontent.com/139339641/365046474-c1fdf229-13af-45cd-aee0-b09fd07253d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MzMyMjAsIm5iZiI6MTcyNTYzMjkyMCwicGF0aCI6Ii8xMzkzMzk2NDEvMzY1MDQ2NDc0LWMxZmRmMjI5LTEzYWYtNDVjZC1hZWUwLWIwOWZkMDcyNTNkMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNlQxNDI4NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00Y2UyYzJmMjI1MTk0YzFmMTQwZjVlZWYzYzYzNDU0Nzc4Mzg0M2YyYjdhZmZiN2UyOTNlYWI5Y2FmMGRiMTg0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FyDVi_1DcPrg5eqlFcE7tKW1BdMj-HfY2w_R_4LEPzA">

Patch https://github.com/tedd-an/bluetooth-next/pull/1172 has been merged for a while though.

Sohan-kr, 2024-09-09:
Hi @Vudentz, as per the shared pull request link, it seems that **Tedd-An** created the pull request from the local branch **776737** to the **workflow** branch. However, it appears that the patch was not merged into the **workflow** branch. I have attached the snapshot of the **git blame** for the **hci_sync.c** file from **Tedd-An’s workflow** branch for your reference.
[image]
[image]