bluez/bluez #576: Support for BLE devices that notify with data size greater than MTU

zerotypic, 2023-08-18:
I'm working on communicating with a BLE device that doesn't seem to honour the MTU exchange protocol. During connection (the device is the server), when MTUs are exchanged, the device always returns an MTU of 23. However, it appears to use the client's MTU when transmitting data (e.g. 517). In particular, when it sends a notification on a characteristic, it uses the client's MTU to determine the size of the new value sent. So, although our negotiated MTU should be min(517, 23) == 23, the notification it sends could contain say 200 bytes.

The full data is actually accessible from bluetoothd; in `shared/att.c` `can_read_data()`:

https://github.com/bluez/bluez/blob/60731cab5891ad2be01f5fcdbf5908aff7df4fd1/src/shared/att.c#L1023-L1031

Here, I can read all the sent bytes from `chan->fd` by using a sufficiently large read size (i.e. larger than `chan->mtu`, with the buffer sized accordingly as well). Since the maximum size of a characteristic is 512 bytes, reading into a 512 byte buffer would always read all the data that could have been sent in the notification. I tested this out, and could successfully read the entire notification as sent by the device, with the data making its way up to a D-Bus client.

Is there any interest in supporting such buggy devices, and if so, what would be the best way to do so? Perhaps some option, that if set, would always malloc `bt_att_chan.buf` to 512 bytes, and try to read 512 bytes in the read above? I'm willing to implement it and put up a PR, but would like to know what the preferred approach is.

I'm not sure, however, if I've gotten anything wrong, as I'm only passingly familiar with BLE and bluetoothd's code. For example I'm not sure if my suggested fix would break things when multiple notifications come in quick succession: do the values get appended to the `chan->fd` file descriptor and thus we would need to read them in MTU-sized chunks? Sorry if none of this actually makes sense!

zerotypic, 2023-08-22:
FWIW I found a BLE library that exhibits the buggy behaviour I describe above: https://github.com/currantlabs/ble, specifically the code here: https://github.com/currantlabs/ble/blob/c1d21c164cf83d3360fd5abd73a3405cb588c06c/linux/att/server.go#L207C1-L235

Any devices using this library, for example the Unifi Protect G3 Instant which I am working on now, will send packets with a higher MTU than `min(client_mtu, server_mtu)`. So none of them will work with bluetoothd.

Vudentz, 2023-08-22:
> FWIW I found a BLE library that exhibits the buggy behaviour I describe above: [currantlabs/ble](https://github.com/currantlabs/ble), specifically the code here: [currantlabs/ble@`c1d21c1`/linux/att/server.go#L207C1-L235](https://github.com/currantlabs/ble/blob/c1d21c164cf83d3360fd5abd73a3405cb588c06c/linux/att/server.go#L207C1-L235)
> 
> Any devices using this library, for example the Unifi Protect G3 Instant which I am working on now, will send packets with a higher MTU than `min(client_mtu, server_mtu)`. So none of them will work with bluetoothd.

The problem with allowing buggy behavior is there can be qualification tests that checks we cannot receive more than the MTU negotiated and even if such test doesn't exists today there is nothing stopping Bluetooth SIG to introduce it later, so usually when it is clear that the implementation is not compliant Id recommend follow up with the manufacturer to have it fixed.

zerotypic, 2023-08-23:
Sure, I understand that. What if support was only enabled via an optional flag, would that allow it to still pass qualification tests? The issue with getting the manufacturer to fix it, of course, is that there might be devices that can't be updated, and would be permanently buggy.

On the topic of being compliant, could I also confirm that after MTU exchange, both sending and receiving MTUs must be equal to the minimum? That's my reading of the specification, but I'm not sure if I missed anything out.

closed 2025-08-07 by github-actions[bot]