bluez/bluez #577: ble connecting to new devices is broken with dual transport

hiredman, 2023-08-18:
with bluez 5.68 with ControllerMode = dual, I cannot connect to new(not connected to before or removed) ble devices. if I switch to ControllerMode = le then I can, and once that initial connection happens I can switch back to ControllerMode = dual and connect to and use the device.

Vudentz, 2023-08-18:
@hiredman Any chance to add btmon logs? Are you able to discover them but they don't connect?

hiredman, 2023-08-21:
no, the devices don't show up at all in scans.

btmon output while running `bluetoothctl scan le`:

```
Bluetooth monitor ver 5.68
= Note: Linux version 6.5.0-0.rc6.43.fc40.x86_64 (x86_64)                                                                                             0.731380
= Note: Bluetooth subsystem version 2.22                                                                                                              0.731382
= New Index: 54:8D:5A:75:19:43 (Primary,USB,hci0)                                                                                              [hci0] 0.731383
= Open Index: 54:8D:5A:75:19:43                                                                                                                [hci0] 0.731383
= Index Info: 54:8D:5A:75:19:43 (Intel Corp.)                                                                                                  [hci0] 0.731383
@ MGMT Open: bluetoothd (privileged) version 1.22                                                                                            {0x0001} 0.731384
@ MGMT Command: Set Bondable (0x0009) plen 1                                                                                          {0x0001} [hci0] 6.979031
        Bondable: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                        {0x0001} [hci0] 6.979036
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x00000a91
          Powered
          Bondable
          BR/EDR
          Low Energy
          Secure Connections
@ MGMT Command: Set IO Capability (0x0018) plen 1                                                                                     {0x0001} [hci0] 6.979059
        Capability: KeyboardDisplay (0x04)
@ MGMT Event: Command Complete (0x0001) plen 3                                                                                        {0x0001} [hci0] 6.979060
      Set IO Capability (0x0018) plen 0
        Status: Success (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                                                                                      {0x0001} [hci0] 11.608005
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                                                                                  #1 [hci0] 11.608078
        Address: 3C:2C:C9:DD:F0:34 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                #2 [hci0] 11.706661
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                                                       #3 [hci0] 11.706804
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Active (0x01)
          Interval: 22.500 msec (0x0024)
          Window: 11.250 msec (0x0012)
        Entry 1: LE Coded
          Type: Active (0x01)
          Interval: 22.500 msec (0x0024)
          Window: 11.250 msec (0x0012)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                #4 [hci0] 11.708645
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                            #5 [hci0] 11.708758
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                #6 [hci0] 11.710586
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: Inquiry (0x01|0x0001) plen 5                                                                                                #7 [hci0] 11.710720
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                                                                                                  #8 [hci0] 11.712662
      Inquiry (0x01|0x0001) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                       {0x0001} [hci0] 11.712755
      Start Discovery (0x0023) plen 1
[... 5 lines cut ...]
@ MGMT Event: Discovering (0x0013) plen 2                                                                                            {0x0001} [hci0] 11.712765
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Enabled (0x01)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                            #9 [hci0] 22.011115
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
[... cut ...]
not sure how that all parses. I have a ble device about a foot away from my laptop waiting for someone to connect to it (has a green light that blinks while it is unconnected). I still have the scan running now, and have not seen any scan results show up.

My pi across the room has an older bluez version, so cannot run `bluetoothctl scan le` but `bluetoothctl scan on` shows the ble, plus other devices popping in and out.

The entire time my laptop shows nothing in either `bluetoothctl scan on` or `bluetoothctl scan le`.

So maybe scanning is what is broken with ControllerMode = dual

Vudentz, 2023-08-21:
@hiredman interesting, how does it look like when you set ControllerMode = le? Also what controller and firmware you have? We didn't ask for the controller to be doing any filtering, so it really beats me why it is not sending anything:

`        Filter policy: Accept all advertisement (0x00)`

hiredman, 2023-08-21:
well, now I cannot reproduce anything

1. when I set ControllerMode = le it still isn't showing any scan results (incuding the new ble device)
2. when I removed a ble device I was able to connect to after setting ControllerMode = le the other day, I don't see it in scans and cannot connect to it now :(
3. going back to ControllerMode = dual bredr devices ares showing up in scans fine
4. a ble device I have successfully connected to before, but have not removed, I can reconnect to after disconnecting

So I dunno, ble is just kind of in and out I guess?

```
[    8.872131] Bluetooth: hci0: Found device firmware: intel/ibt-19-32-4.sfi
[    8.872238] Bluetooth: hci0: Boot Address: 0x24800
[    8.872240] Bluetooth: hci0: Firmware Version: 206-22.23
[    8.872242] Bluetooth: hci0: Firmware already loaded
```

pallaswept, 2023-08-22:
The behaviour was totally different when doing this with btmon running. So here's what I did:

sudo btmon
used a 2.4ghz wireless mouse that's working, to drive the GUI, open up the bluetooth config in KDE and click the add device button
turned on my BT mouse (and I saw a bunch of stuff happen in btmon when I just turned it on)
put the mouse in pairing mode (and more stuff happened in btmon, and a very familiar MAC address appeared in the GUI, but the name of the device ("MM731 hybrid mouse", according to the phone never appeared). There was also a second MAC I don't recognise. Probably something from next door.
when it timed out I turned off my mouse and the GUI came up with a "Your device is not pairable" message and a button to make it pairable. Thats totally not normal, it never does that. Is that just btmon interfering?

It's kinda inspiring to see it actually see the mouse, kinda-sorta.... but there's a big antenna on my motherboard and usually when I try to add a device I get literally half of my street's smart watches and refrigerators and various other IoT gizmos in the list, and it's definitely not doing that normal thing. Having btmon running while doing this, made it behave completely differently.

Thanks for your help. I'm going to try the different modes mentioned in the other issue you linked and report back if there's any change.

Vudentz, 2023-08-22:
Looks like I managed to reproduce with Intel AX201 it doesn't resolve the name:

https://gist.github.com/Vudentz/0988a78da9e5658a06edec159847db27

Vudentz, 2023-08-23:
@hiredman any chance to test the changes above (v3).

hiredman, 2023-08-23:
@Vudentz I will start trying to figure out how to test it, but don't hold your breath, it has been a long time since I've built the kernel myself

Vudentz, 2023-08-23:
@hiredman what distro you have? Perhaps you can ask @pallaswept for the kernel image, or build a VM image and just forward the USB device, we could in theory also do something similar with the likes of test-runner (https://github.com/bluez/bluez/blob/master/doc/test-runner.rst) but it currently only support btproxy method which means the guest VM will thing it is using serial UART transport, not USB.

pallaswept, 2023-08-23:
>  will start trying to figure out how to test it, but don't hold your breath, it has been a long time since I've built the kernel myself

If you're running opensuse on x86_64 (I can toggle on other architectures very quickly, the build takes about an hour after that) I have packages you can use for testing.

hiredman, 2023-08-24:
hey, look, I built a kernel for the first time in over a decade! (I am on fedora)

I built the head of bluetooth-next (1098ecdb96b292414e5471262e2b4af5cf377c0c) and I can scan for and connect to new ble devices now. It seems to take longer than I expect for scanning to find the devices(maybe a minute?), but that is entirely subjective, and may just be unreasonable expectations on my part.

thanks!

Vudentz, 2023-08-24:
> hey, look, I built a kernel for the first time in over a decade! (I am on fedora)
>
> I built the head of bluetooth-next (1098ecdb96b292414e5471262e2b4af5cf377c0c) and I can scan for and connect to new [...]

That depends on how frequent the devices are advertising as well, but usually it should come pretty quickly but perhaps this is because you used the interleaved discovery, LE+BR/EDR, try only using bluetoothctl> scan le to see if that gets better.

Anyway this must be something else then the LE Coded PHY being enabled, so I hope we can close this one and if you feel like the discovery is taking too long please create another issue.

hiredman, 2023-08-24:
oh, for sure, thanks again

closed 2023-08-24 by hiredman