bluez/bluez #579: A potential Buffer Overflow bug found in tools/test-runner.c

cascades-sjtu, 2023-08-20:
Hi, I'm currently trying to use the static analysis tool [Infer](https://fbinfer.com) to find uncatched API-misuse bugs in OpenWrt packages, and I find a potential Buffer Overflow in your project, version 5.66.

The bug located in tools/test-runner.c. Firstly, `xdg_runtime_dir` was obtained from environment variable in line 280, and its length is later used as the parameter of `alloca`. While the formatted string is 40-bytes long(except from the %s), the `sprintf` may cause an Off-By-Null issue, as shown in the following code:

```cpp=
xdg_runtime_dir = getenv("XDG_RUNTIME_DIR");
if (!xdg_runtime_dir) {
    fprintf(stderr, "XDG_RUNTIME_DIR not set\n");
    exit(1);
}
audiodev = alloca(40 + strlen(xdg_runtime_dir));
sprintf(audiodev, "id=audio,driver=pa,server=%s/pulse/native", xdg_runtime_dir);
```

I also attached the analysis trace given by Infer FYI:

```json=
       "trace": [
          {
            "file": "tools/test-runner.c",
            "line": 280,
            "col": 21,
            "feature": [ "Input", "getenv" ]
          },
          {
            "file": "tools/test-runner.c",
            "line": 285,
            "col": 14,
            "feature": [ "LibraryCall", "strlen" ]
          },
          {
            "file": "tools/test-runner.c",
            "line": 286,
            "col": 3,
            "feature": [ "BufferOverflow", "sprintf", [ "Var" ] ]
          }
        ],
        "vec": {
          "BufferOverflow": 1,
          "LibraryCall strlen": 1,
          "Input getenv": 1
        },
```

pv, 2023-08-20:
That code no longer exists in latest releases. Testing like this probably makes most sense for upstream if run on the current master branch.

cascades-sjtu, 2023-08-20:
Thanks for your reply, I fetch the bluez source code from OpenWrt packages, so it still may cause some issues in the downstream repo?

pv, 2023-08-20:
I think it's job for the downstream (OpenWRT) to apply any patches if they are using old releases, so you can contact them.

In this case, however, test-runner is a tool for running unit tests in a VM image, so probably not relevant for any downstream distributor.