bluez/bluez #581: Bluetooth: SMP security requested but not available And Set Secure Connection failed

LIUpeirun, 2023-08-21:
Hello,I recently encountered some issues porting Bluez to Linux system.
My Linux kernel version is 4.4.19.
When I used the IOS system phone to connect to the Bluetooth on my board, this exception occurred.
`Bluetooth: SMP security requested but not available
bluetoothd[600]: src/device.c:gatt_debug() (chan 0x172a3b0) Retrying operation 0x17368d0
Bluetooth: SMP security requested but not available
bluetoothd[600]: src/device.c:gatt_debug() (chan 0x172a3b0) Retrying operation 0x17368d0`
When I started Bluetooth, I observed that such an error occurred.
`bluetoothd[2526]: Failed to set privacy: Rejected (0x0b)`
For this error, I tried using btmgmt sc off to close the secure connection and modify privacy=off in main.conf,But they will still face the same mistake.
`[rootaanyka~]$ btmgmt bredr off
Set BR/EDR for hci failed with status 0x0b (Rejected)
[rootaanyka~]$ btmgmt le off
hcio Set Low Energy complete, settings: powered br/edr
[rootaanyka -]$ btmgmt le oepn
hci0 Set Low Energy complete, settings: powered br/edr
[root@anyka~]$ btmgmt bredr off
Set BR/EDR for hci0 failed with status 0x0b (Rejected)
[rootanyka-]$ btmgmt bredr open
Set BR/EDR for hci failed with status 0xb(Rejected)
[rootaanyka -]$ btmgmt sc open
Set Secure Connections for hcio failed with status 0x0b (Rejected)
[root@anyka ~]$ btmgmt sc off
Set Secure Connections for hci0 failed with status 0x0b (Rejected)
[rootaanyka ~]$ btmgmt privacy open 
Set Privacy for hci0 failed with status 0xb (Rejected)
[rootaanyka~]$ btmgmt privacy off
Set Privacy hci0 failed with status 0xOb (Rejected)`
I don't know how to solve this problem, so I would like to ask for your advice. Thank you very much.

LIUpeirun, 2023-08-21:
I feel that this should be related to a certain configuration in the kernel, but I don't know how to solve it or which configuration is related. This is my personal guess

closed 2023-08-30 by LIUpeirun

Aye-zang, 2024-07-25:
you need let the power off first. 
use` btmgmt power off` then ` btmgmt privacy on`