bluez/bluez #582: Unable to pair any new devices, errors in journalctl, new since 5.68

pallaswept, 2023-08-22:
I'm able to reconnect to any devices I've already paired, but I can no longer pair any new devices. The list just shows up empty, as though there's nothing there. My phone can see those devices trying to pair, no problem. the PC, just nothingness. Same deal whether I use KDE's bluedevil-wizard or bluetoothctl CLI and 'scan on', just nothing happens.

I've noticed this:

```
Aug 22 09:00:51 localhost.localdomain systemd[1]: Starting Bluetooth service...
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: Bluetooth daemon 5.68
Aug 22 09:00:51 localhost.localdomain systemd[1]: Started Bluetooth service.
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: Starting SDP server
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support csip plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support vcp plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support mcp plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support bass plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support bap plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: src/plugin.c:plugin_init() System does not support admin plugin
Aug 22 09:00:51 localhost.localdomain bluetoothd[4880]: Bluetooth management interface 1.22 initialized
```

All these 'system does not support * plugin" messages are new, I never used to get all that. I have a btrfs snapshot from July that works fine, and I don't see these messages there. Since that time I've had new kernels, new firmware, and new bluez, so I am not sure where to start, other than these error messages. I tried rolling back the firmware to no avail, and /var/log/messages reports no errors other than the same as the above (this is where any bluetooth-related messages from that file begin):
```
Aug 22 09:00:51 localhost kernel: NET: Registered PF_BLUETOOTH protocol family
Aug 22 09:00:51 localhost kernel: Bluetooth: HCI device and connection manager initialized
Aug 22 09:00:51 localhost kernel: Bluetooth: HCI socket layer initialized
Aug 22 09:00:51 localhost kernel: Bluetooth: L2CAP socket layer initialized
Aug 22 09:00:51 localhost kernel: Bluetooth: SCO socket layer initialized
Aug 22 09:00:51 localhost kernel: input: Microsoft X-Box One S pad as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-9/1-9.1/1-9.1.3/1-9.1.3:1.0/input/input35
Aug 22 09:00:51 localhost kernel: usbcore: registered new interface driver xpad
Aug 22 09:00:51 localhost mtp-probe[4800]: checking bus 1, device 10: "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-9/1-9.1/1-9.1.3"
Aug 22 09:00:51 localhost mtp-probe[4800]: bus: 1, device: 10 was not an MTP device
Aug 22 09:00:51 localhost kernel: Intel(R) Wireless WiFi driver for Linux
Aug 22 09:00:51 localhost kernel: usbcore: registered new interface driver btusb
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: enabling device (0000 -> 0002)
Aug 22 09:00:51 localhost systemd[1]: Starting Load/Save RF Kill Switch Status...
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: Detected crf-id 0x3617, cnv-id 0x100530 wfpm id 0x80000000
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: PCI dev 2723/0084, rev=0x340, rfid=0x10a100
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: Direct firmware load for iwlwifi-cc-a0-78.ucode failed with error -2
Aug 22 09:00:51 localhost kernel: usb 1-9.1.4: Found UVC 1.00 device USB Camera (0c45:6366)
Aug 22 09:00:51 localhost systemd[1]: Starting Bluetooth service...
Aug 22 09:00:51 localhost systemd[1]: Started Load/Save RF Kill Switch Status.
Aug 22 09:00:51 localhost kernel: igc 0000:06:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)
Aug 22 09:00:51 localhost kernel: igc 0000:06:00.0 eth0: MAC: 3c:7c:3f:1c:e5:49
Aug 22 09:00:51 localhost kernel: igc 0000:06:00.0 enp6s0: renamed from eth0
Aug 22 09:00:51 localhost bluetoothd[4880]: Bluetooth daemon 5.68
Aug 22 09:00:51 localhost systemd[1]: Started Bluetooth service.
Aug 22 09:00:51 localhost bluetoothd[4880]: Starting SDP server
Aug 22 09:00:51 localhost systemd[1]: Reached target Bluetooth Support.
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support csip plugin
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: api flags index 2 larger than supported by driver
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
Aug 22 09:00:51 localhost kernel: iwlwifi 0000:05:00.0: loaded firmware version 77.2df8986f.0 cc-a0-77.ucode op_mode iwlmvm
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support vcp plugin
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support mcp plugin
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support bass plugin
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support bap plugin
Aug 22 09:00:51 localhost bluetoothd[4880]: src/plugin.c:plugin_init() System does not support admin plugin
Aug 22 09:00:51 localhost bluetoothd[4880]: Bluetooth management interface 1.22 initialized
Aug 22 09:00:51 localhost kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Aug 22 09:00:51 localhost kernel: Bluetooth: BNEP filters: protocol multicast
Aug 22 09:00:51 localhost kernel: Bluetooth: BNEP socket layer initialized
Aug 22 09:00:51 localhost kernel: Bluetooth: hci0: Found device firmware: intel/ibt-20-1-3.sfi
Aug 22 09:00:51 localhost kernel: Bluetooth: hci0: Boot Address: 0x24800
Aug 22 09:00:51 localhost kernel: Bluetooth: hci0: Firmware Version: 206-22.23
Aug 22 09:00:51 localhost kernel: Bluetooth: hci0: Firmware already loaded

```
Any help would be greatly appreciated, I'm disabled and can't move a lot so I kinda need my wireless gizmos working :( I'm kinda terrified that this keyboard gets unpaired somehow because I can't get to the desk to use the wired one. If there's any info/logs/etc that you could use to help me try and work out what's gone wrong here, just let me know.

Vudentz, 2023-08-22:
@pallaswept Lets start with btmon logs, I suspect this is a similar issue to https://github.com/bluez/bluez/issues/577 if the controller is not sending any reports found during scanning.

pallaswept, 2023-08-22:
The behaviour was totally different when doing this with btmon running. So here's what I did:

sudo btmon
used a 2.4ghz wireless mouse that's working, to drive the GUI, open up the bluetooth config in KDE and click the add device button
turned on my BT mouse (and I saw a bunch of stuff happen in btmon when I just turned it on)
put the mouse in pairing mode (and more stuff happened in btmon, and a very familiar MAC address appeared in the GUI, but the name of the device ("MM731 hybrid mouse", according to the phone never appeared). There was also a second MAC I don't recognise. Probably something from next door.
when it timed out I turned off my mouse and the GUI came up with a "Your device is not pairable" message and a button to make it pairable. Thats totally not normal, it never does that. Is that just btmon interfering?

It's kinda inspiring to see it actually see the mouse, kinda-sorta.... but there's a big antenna on my motherboard and usually when I try to add a device I get literally half of my street's smart watches and refrigerators and various other IoT gizmos in the list, and it's definitely not doing that normal thing. Having btmon running while doing this, made it behave completely differently.

Thanks for your help. I'm going to try the different modes mentioned in the other issue you linked and report back if there's any change.

pallaswept, 2023-08-22:
OK So I tried it in le mode, and the add device page instantly lit up with 3 MAC addresses, but still no names, and when I put my mouse in pairing mode, it never appeared in the list. Then the list just slowly emptied out. It seems super-intermittently working. What was interesting is that I just let it sit there, and that "Your device is not pairable" message re-appeared. And then disappeared, and reappeared, over the space of a minute or so. And i literally didn't touch a thing, I was sitting here watching it. btmon was not running when all this happened. Here's a screenshot of the message so you know what I'm talking about:

[image]

pallaswept, 2023-08-22:
Possibly related: several other opensuse tumbleweed users are getting unusual intermittent behaviour since the upgrade, the OP included logs with the same sort of plugin_init() errors I got.
https://www.reddit.com/r/openSUSE/comments/15pz3k3/last_tumbleweed_upgrade_broke_bluetooth/

Vudentz, 2023-08-22:
I think this is specific to some controllers, for instance AX201 seems to have this problem while AX210 doesn't:

https://gist.github.com/Vudentz/0988a78da9e5658a06edec159847db27

With AX201 there is no SCAN_RSP which means there is no name which would explain if only the address are displayed.

pallaswept, 2023-08-22:
> I think this is specific to some controllers

Mine's an AX200 I believe. Greatly appreciate you and Luiz Augusto for helping with this so quickly. If there's a branch I can test for you or if you like I can make sure it's packaged on OpenBuildService so that there's an interim fix for other opensuse users, or anything else that can assist, just say the word. Seems like the heavy muscle is on this and I should get out of the way for now, so, thank you very much :)

Vudentz, 2023-08-22:
> Mine's an AX200 I believe. Greatly appreciate you and Luiz Augusto for helping with this so quickly. If there's a branch I can test for you or if you like I can make sure it's packaged on [...]

Please go ahead and create a test package if you can, I have a limited number of controllers so we probably want to collect more information about other controllers behaving this way. You can find the patch above, note that the fix is in the kernel though not userspace.

pallaswept, 2023-08-22:
I am a little slow at this (re-learning it) but I'm on it.

pallaswept, 2023-08-23:
I am a little slow at this (re-learning it) but I'm on it.

> Please go ahead and create a test package if you can, I have a limited number of controllers so we probably want to collect more information about other controllers behaving this way. You can find [...]

Well the good news is that I know how to do this next time and have figured out how opensuse like to include patches into their kernel builds, and did get the build done and the package installed. The bad news is that it didn't exactly boot. I was able to get to TTY shells, and I tried bluetoothctl there, but there was no change in behaviour (ie, it didn't fix it)

I think the issue is that I'm using a branch of suse's official kernel build which is 6.4.11 at present (no surprise that the other reports have been that their issues started with 6.4), and the patch was made for something newer (probably the latest sources I guess). I had to modify the patch slightly to get the line numbers to match up so that it'd merge, but even still, that patch alone didn't do the trick.

I could try and build from the full source where the patch originated but it'd mean checking off a *long* list of patches to ensure compatibility with the suse build so it's not really practical, and I have a strong aversion to installing software outside of the control of my package manager, it tends to lead to broken boxes.

Not sure if there's anything I can do here. If there's a version of the patch built against linux 6.4 sources, I could give that a shot?

pallaswept, 2023-08-23:
Great news, it works!

I noticed there's a newer version of the patch, the one which modifies 4 files rather than the original 3. I put that in my build, with the same mods to make it merge with the 6.4 sources, and while my graphics problems persisted I used bluetoothctl from the TTY and the mouse I've been using for testing appeared, name and all, along with all the street's IoT devices, just like it used to work. I paired the mouse, and now even back on the original unpatched kernel it still remembers it, which is a handy bonus!

So yes, the second version of the patch definitely works, even on 6.4, with some minor modification. Nice!

For reference, the patch I used this time, without my mods:

```
From c5950023655727a9ee904a14c53c9b7fb943c240 Mon Sep 17 00:00:00 2001
From: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date: Tue, 22 Aug 2023 16:23:40 -0700
Subject: [PATCH] Bluetooth: HCI: Introduce HCI_QUIRK_BROKEN_LE_CODED

This introduces HCI_QUIRK_BROKEN_LE_CODED which is used to indicate
that LE Coded PHY shall not be used, it is then set for some Intel
models that claim to support it but when used causes many problems.

Link: https://github.com/bluez/bluez/issues/577
Link: https://github.com/bluez/bluez/issues/582
Link: https://lore.kernel.org/linux-bluetooth/CABBYNZKco-v7wkjHHexxQbgwwSz-S=GZ=dZKbRE1qxT1h4fFbQ@mail.gmail.com/T/#
Fixes: 288c90224eec ("Bluetooth: Enable all supported LE PHY by default")

---
 drivers/bluetooth/btintel.c      |  6 ++++++
 include/net/bluetooth/hci.h      | 10 ++++++++++
 include/net/bluetooth/hci_core.h |  4 +++-
 net/bluetooth/hci_sync.c         |  5 ++++-
 4 files changed, 23 insertions(+), 2 deletions(-)

diff --git a/drivers/bluetooth/btintel.c b/drivers/bluetooth/btintel.c
index 633e8d9bf58f95..a62145710accda 100644
--- a/drivers/bluetooth/btintel.c
+++ b/drivers/bluetooth/btintel.c
@@ -2780,6 +2780,9 @@ static int btintel_setup_combined(struct hci_dev *hdev)
 			set_bit(HCI_QUIRK_WIDEBAND_SPEECH_SUPPORTED,
 				&hdev->quirks);
 
+			/* These variants don't seem to support LE Coded PHY */
+			set_bit(HCI_QUIRK_BROKEN_LE_CODED, &hdev->quirks);
+
 			/* Setup MSFT Extension support */
 			btintel_set_msft_opcode(hdev, ver.hw_variant);
 
@@ -2851,6 +2854,9 @@ static int btintel_setup_combined(struct hci_dev *hdev)
 		 */
 		set_bit(HCI_QUIRK_WIDEBAND_SPEECH_SUPPORTED, &hdev->quirks);
 
+		/* These variants don't seem to support LE Coded PHY */
+		set_bit(HCI_QUIRK_BROKEN_LE_CODED, &hdev->quirks);
+
 		/* Set Valid LE States quirk */
 		set_bit(HCI_QUIRK_VALID_LE_STATES, &hdev->quirks);
 
diff --git a/include/net/bluetooth/hci.h b/include/net/bluetooth/hci.h
index c58425d4c592c9..87d92accc26ea9 100644
--- a/include/net/bluetooth/hci.h
+++ b/include/net/bluetooth/hci.h
@@ -319,6 +319,16 @@ enum {
 	 * This quirk must be set before hci_register_dev is called.
 	 */
 	HCI_QUIRK_USE_MSFT_EXT_ADDRESS_FILTER,
+
+	/*
+	 * When this quirk is set, LE Coded PHY shall not be used. This is
+	 * required for some Intel controllers which erroneously claim to
+	 * support it but it causes problems with extended scanning.
+	 *
+	 * This quirk can be set before hci_register_dev is called or
+	 * during the hdev->setup vendor callback.
[... 119 lines cut ...]
 
 #define scan_coded(dev) (((dev)->le_tx_def_phys & HCI_LE_SET_PHY_CODED) || \
 			 ((dev)->le_rx_def_phys & HCI_LE_SET_PHY_CODED))
diff --git a/net/bluetooth/hci_sync.c b/net/bluetooth/hci_sync.c
index 0cb78081719837..9b93653c619730 100644
--- a/net/bluetooth/hci_sync.c
+++ b/net/bluetooth/hci_sync.c
@@ -4668,7 +4668,10 @@ static const struct {
 			 "advertised, but not supported."),
 	HCI_QUIRK_BROKEN(SET_RPA_TIMEOUT,
 			 "HCI LE Set Random Private Address Timeout command is "
-			 "advertised, but not supported.")
+			 "advertised, but not supported."),
+	HCI_QUIRK_BROKEN(LE_CODED,
+			 "HCI LE Coded PHY feature bit is set, "
+			 "but its usage is not supported.")
 };
 
 /* This function handles hdev setup stage:

So, big thanks to you all who have worked to fix this up. 

I'm not sure of the correct procedure from here, to create a patch intended for 6.4 and get it pushed out.... Or perhaps it's not too late to get it into 6.5, which I believe is in late RC stages, and wouldn't be too far off for end-user availability? Could you advise any action I should take here, if there is any? I'm sorry for the noob questions, the last time I touched the linux kernel in any serious manner wasn't this century 🤣 

```

pallaswept, 2023-08-23:
> > I'm not sure of the correct procedure from here, to create a patch intended for 6.4 and get it pushed out.... Or perhaps it's not too late to get it into 6.5, which I believe is in late RC stages, and wouldn't be too far off for end-user availability? Could you advise any action I should take here, if there is any? I'm sorry for the noob questions, the last time I touched the linux kernel in any serious manner wasn't this century 🤣
> 
> I hope we can still include this for 6.5 which should then be pickup by backport to be included in 6.4 as well as it includes the fixes tag.

Sounds great, if it can get into 6.5. 

BTW my graphics issues are definitely not the fault of the patch, it's my kernel build, I've got that all sorted out now and the patch is working in GUI land too.

Vudentz, 2023-08-24:
Pull request on the way:
https://patchwork.kernel.org/project/bluetooth/patch/20230824201458.2577-1-luiz.dentz@gmail.com/

pallaswept, 2023-08-26:
> Pull request on the way: https://patchwork.kernel.org/project/bluetooth/patch/20230824201458.2577-1-luiz.dentz@gmail.com/

Fantastic. As far as I'm concerned that patch fixes it and it's all down to kernel.org now so you can resolve this case if you don't need it to track things down the line or whatever.

pallaswept, 2023-09-02:
> Pull request on the way: https://patchwork.kernel.org/project/bluetooth/patch/20230824201458.2577-1-luiz.dentz@gmail.com/

I notice it didn't make it into 6.5 :( What's the plan of attack for those of us suffering this? Custom kernels until 6.6?

Vudentz, 2023-09-06:
> > Pull request on the way: [patchwork.kernel.org/project/bluetooth/patch/20230824201458.2577-1-luiz.dentz@gmail.com](https://patchwork.kernel.org/project/bluetooth/patch/20230824201458.2577-1-luiz.dentz@gmail.com/)
> 
> I notice it didn't make it into 6.5 :( What's the plan of attack for those of us suffering this? Custom kernels until 6.6?

Yep, we were late for 6.5, but it should find its way to backports once its 6.6-rc1 is announced.

pallaswept, 2023-09-07:
So close! Oh well custom kernels it is until then. The code is merged, so if you want to resolve this case I'm OK with that, but perhaps you'd like to keep it open to track it until the merged fixes reach public release. I don't want to close it just in case you need it open.

closed 2023-09-07 by Vudentz

pallaswept, 2023-09-14:
It's in 6.5.3 FWIW.