bluez/bluez #583: Config checking generates erroneous failures

pallaswept, 2023-08-22:
In attempts to pinpoint the problem in #582 I have been altering my config file. In many cases, I simply uncommented the default value, just to be sure it is being set.

The result of this has been several errors such as the following:
```
 bluetoothd[3933]: src/main.c:check_options() Unknown key MultiProfile for group General in /etc/bluetooth/main.conf
 bluetoothd[3933]: src/main.c:check_options() Unknown key RefreshDiscovery for group General in /etc/bluetooth/main.conf
 bluetoothd[3933]: src/main.c:check_options() Unknown key Encryption for group CSIS in /etc/bluetooth/main.conf
 bluetoothd[3933]: src/main.c:parse_config_int() LE.EnableAdvMonInterleaveScan = 0 is out of range (> 1)
 bluetoothd[3933]: src/main.c:parse_config_int() GATT.Channels = 3 is out of range (> 5)
 bluetoothd[3933]: src/main.c:parse_config_int() CSIS.Rank = 0 is out of range (> 255)
 bluetoothd[3933]: src/main.c:parse_config_int() AdvMon.RSSISamplingPeriod = 0 is out of range (> 255)
```
Obviously 0 is not > 255, 0 is not >1, 3 is not >5....and those strings are definitely present in the relevant parts of the code. 

An unusual example was 

`src/main.c:parse_config_int() GATT.KeySize = 0 is out of range (< 7)` 

(The acceptable range is 7-16 but 0 is supposed to be 'don't care' and can't be accepted because it does a simple test for <7

But most of the numeric errors are just strange. I'm thinking maybe it's not handling the cast from long int (returned from strtol()) to int (the format of the comparison)? But I don't know. I just found this so I'm reporting it I'm sorry I can't be more helpful by providing the answers yet.

The string errors aren't making a lot of sense either, as with the numeric values, code-wise everything looks like it should be working, only, it isn't.

Anyway, there's a real problem here because there's no way to test faulty settings because we can't enter valid ones. A good example being GATT channels, the old (5.66) config had it at 3, it worked, 1 doesn't, so I wanted to try 3, but it won't let me.... You get the idea. What do?

Vudentz, 2023-08-25:
@pallaswept I think we had fixed this already:

https://github.com/bluez/bluez/commit/446d8ada5790e5ca2cdc359eafa4bf7136b43ac5

pallaswept, 2023-08-25:
> @pallaswept I think we had fixed this already:
> 
> [446d8ad](https://github.com/bluez/bluez/commit/446d8ada5790e5ca2cdc359eafa4bf7136b43ac5)

Hahaha well that explains why when I glanced at the code it looked fine and I couldn't see why it wouldn't work - it's fixed already! I'll look forward to that one in 5.69 or whatever is next.

What about the strings though?

pallaswept, 2023-08-27:
> What about the strings though?

looking closer, maybe that is fixed with https://github.com/bluez/bluez/commit/2c94ab92cb0e9e8c2493340378d6d29ef467a6ac ?

Vudentz, 2023-09-06:
> Hahaha well that explains why when I glanced at the code it looked fine and I couldn't see why it wouldn't work - it's fixed already! I'll look forward to that one in 5.69 or whatever is next.
>
> What [...]

Do we you have a problem with any of the string parameters? I though this was specific to integer values only.

pallaswept, 2023-09-07:
> Do we you have a problem with any of the string parameters? I though this was specific to integer values only.

It appears so - note the first three errors:
```
 bluetoothd[3933]: src/main.c:check_options() Unknown key MultiProfile for group General in /etc/bluetooth/main.conf
 bluetoothd[3933]: src/main.c:check_options() Unknown key RefreshDiscovery for group General in /etc/bluetooth/main.conf
 bluetoothd[3933]: src/main.c:check_options() Unknown key Encryption for group CSIS in /etc/bluetooth/main.conf

```

All of these strings are simply uncommented defaults from the file.

Vudentz, 2023-09-07:
MultiProfile seems to be working for me, the other 2 are indeed causing these errors although they are parsed they are not listed as supported.

closed 2023-09-07 by BluezTestBot

onsolutionjames, 2023-09-11:
I think this issue might have been prematurely closed. The issue with ints still exists in 5.69, and doesn't seem to be addressed by the above changes.

I've set `TemporaryTimeout=60` in main.conf, and I'm getting `General.TemporaryTimeout = 60 is out of range (> -1)` in the logs, when running 5.69

I suspect the issue is with `parse_config_u32` and `parse_config_int` in `main.c`, from this commit: https://github.com/bluez/bluez/commit/2c94ab92cb0e9e8c2493340378d6d29ef467a6ac

`parse_config_u32` takes a uint32 for the min and max, but I think these are being converted to a signed int when doing the comparison in `parse_config_int`, so it’s actually turning into -1.

I'm not sure what the best solution is here because I'm not a regular C user. Sorry if I'm misinterpreting what's happening though! Happy to provide any more info if needed, or create a quick fix if someone can point me in the right direction.

pallaswept, 2023-09-28:
@Vudentz pardon my @-ing you, but it looks like this needs some more attention, and I thought maybe you're not subbed to it any more since it was resolved, so you hadn't seen the above posts. Apologies if you're already aware. Not trying to rush you at all, just trying to keep you in the loop. Hope I'm not doing it wrong, apologies if so.

Vudentz, 2023-09-28:
> Hi. which version includes the fix? the releases page on this project is strange, i only see 2 releases with 5.68 being the latest. And I see 5.69 as current on most distributions.
> 
> btw, I still see the error on 5.69, several distros.

github serves as a mirror, the releases are done on git.kernel.org, anyway this shall be be fixed in the upcoming 5.70 which we shall be announce shortly.