bluez/bluez #584: Central application fails to connect to remote peripheral, if local peripheral already has a connection.

soulfish5, 2023-08-22:
I have a machine where I am attempting to run both a Peripheral Application(Running a GATT Server) and a Central application (running a GATT Client) using Bluez.
Connections fail, if they are not established in a certain order. **To be clear. If the outgoing connection is established first, there is no issue being connected twice.**

If a remote central is connected to the peripheral application on the machine, the central application is unable to connect to a remote peripheral. When the remote central is disconnected, the local central can once again connect to remote peripherals.  If the events occur in the opposite order, ie - the central application connects to a remote peripheral, the peripheral application is able to receive incoming connections without issue.

In an attempt to take our applications out of the equation, I was able to reproduce the behavior in two others cases.

1. Using the test/example-gatt-server provided by bluez and the ubuntu ui to connect to a remote gatt server.
2. Using the bluetootchctl tool.

### Reproducing the issue with bluetootchctl

If you run through the steps in Setup Advertising, the behavior described above appears. If you run “Connect remote device to local machine” first, followed by “Connect local machine to remote device”, the second connection fails. If you run the steps in the reverse direction, everything works.

**Setup Advertising**

- bluetoothctl
- menu advertise
- name on
- back
- advertise on
- discoverable on

**Connect remote device to local machine.**
Steps for this depend on the remote device being used.

**Connect local machine to remote device**

- scan on
- scan off
- devices ( to find and identify the device to connect to )
- connect "MAC ADDRESS"

I don't have the logs available at the moment. I can attach them later. 
In both cases the log shows "Connection attempt to: ...". This is printed out in the device_connect_le method of the device.c file.
In the case where the connection fails, that is the last message. When thing work, the next log statement is a notification that the device connected. It originates in the connected_callback method of the adapter.c file.

### Hardware/Software Versions.
BLUEZ: 5.50.
Kernel: 5.10.104-tegra.
I have seen the same behavior on two additional machines. An x86 Ubuntu 22.04 VM and a Raspbian Pi4 build. (Will have to add version information later).

hciconfig -a

- hci0:   Type: Primary  Bus: USB
- BD Address: D4:6D:6D:ED:54:99  ACL MTU: 1021:4  SCO MTU: 96:6
- UP RUNNING PSCAN ISCAN
- RX bytes:15062 acl:0 sco:0 events:2432 errors:0
- TX bytes:597226 acl:0 sco:0 commands:2430 errors:0
- Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
- Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
- Link policy: RSWITCH HOLD SNIFF
- Link mode: SLAVE ACCEPT
- Name: 'hostname'
- Class: 0x200000
- Service Classes: Audio
- Device Class: Miscellaneous,
- HCI Version: 4.2 (0x8)  Revision: 0x100
- LMP Version: 4.2 (0x8)  Subversion: 0x100
- Manufacturer: Intel Corp. (2)

Vudentz, 2023-09-06:
@soulfish5 5.50 is really old already, and kernel 5.10 as well, any chance to test with something more recent?

jemmy1794, 2023-10-02:
Have you solve this problem? I faced the same problem.

BLUEZ: 5.66
Ubuntu 22.04 、 Raspbian Pi4

I used the C dbus to register a Service then advertise. If any remote device connect to this device, then this device could not connect to other peripheral. (Call Device1 connect dbus api, get Connect timeout reached)

I've tried the bluetoothctl to test the senario like "Reproducing the issue with bluetootchctl". The problem is reproduced. No connection established until the remote device disconnect from this device.

soulfish5, 2023-10-02:
> @soulfish5 5.50 is really old already, and kernel 5.10 as well, any chance to test with something more recent?

Hi @Vudentz  - I missed the notification on your reply. Stuff ended up getting reworked so the issue didn't need to be solved. If I have some time in the future with a newer OS I will repost here.

soulfish5, 2023-10-02:
> Have you solved this problem? I faced the same problem.
> 
> BLUEZ: 5.66 Ubuntu 22.04 、 Raspbian Pi4
> 
> I used the C dbus to register a Service then advertise. If any remote device connect to this device, then this device could not connect to other peripheral. (Call Device1 connect dbus api, get Connect timeout reached)
> 
> I've tried the bluetoothctl to test the senario like "Reproducing the issue with bluetootchctl". The problem is reproduced. No connection established until the remote device disconnect from this device.

@jemmy1794  Unfortunately I didn't resolve the issue.

Vudentz, 2024-08-12:
https://patchwork.kernel.org/project/bluetooth/patch/20240812150439.283920-1-luiz.dentz@gmail.com/

VinitMehta-NXP, 2024-08-13:
Hello @Vudentz,

After analyzing the issue below is the findings:

The issue is seen because of a specific conditional check in Kernel mentioned below in check_pending_le_conn function in hci_event.c file.

	if (hdev->conn_hash.le_num_peripheral > 0 &&
	    (!test_bit(HCI_QUIRK_VALID_LE_STATES, &hdev->quirks) ||
	     !(hdev->le_states[3] & 0x10)))
	{
		return NULL;
	}

Above conditional check was introduced to allow forming scatternet connections. The quirk HCI_QUIRK_VALID_LE_STATES uses values from HCI_LE_READ_SUPPORTED_STATES command. However, it seems based on specs the values would be available only in case of legacy advertising and not extended advertising.

Below is the link to the patch when the condition was introduced:
https://github.com/bluez/bluetooth-next/commit/4364f2e91f0d44fa0e233d2a55e3ec35053d9bd9

For now, the workaround would be to avoid rejection of create connection command if we bypass the checking for HCI_QUIRK_VALID_LE_STATES quirk.

Regards,
Vinit

Vudentz, 2024-08-14:
@VinitMehta-NXP could you please respond to the thread with a Tested-by?

VinitMehta-NXP, 2024-08-16:
Adding last tested by:
BSP Version: 6.6.23
Bluez Version: 5.72

VinitMehta-NXP, 2024-08-16:
> @VinitMehta-NXP could you please respond to the thread with a Tested-by?

Adding last tested by:
BSP Version: 6.6.23
Bluez Version: 5.72

closed 2024-08-26 by Vudentz