bluez/bluez #585: Random disconnect on Redmi Buds 3 Pro

RodoMa92, 2023-08-23:
Since I've got this bluetooth buds I randomly have disconnections after a while and they will begin to connect one buds first only, then only the other buds while the other will disconnect again and so forth until two possible end state are reached:
- Both buds resynchronize and go back to normal
- No buds resynchronize and I lose bluetooth audio to the buds until a case cycle. They will not respond to any connect event afterwards (not sure if they are always refusing to a manual reconnect, but I can check if that's important).

This happens on a RTL8822CE bluetooth card mounted on a Steam Deck device. This do not happens on my Smartphone (Qualcomm Snapdragon 870 Chipset device).

I've tried to acquire logs with btsnoop, but it always crashes with stack overflow detected or something and doesn't even try to acquire anything. Therefore I used wireshark cli to acquire these logs in background, which seems to work pretty good.

I've trimmed the packet log size, since it was huge (250 MB or something), just as the event happens. The oddities start at frame number 6488, attached below.

LMK if you need additional logs/debugging steps.

[disconnect_reconnect.zip](https://github.com/bluez/bluez/files/12417468/disconnect_reconnect.zip)

RodoMa92, 2023-09-13:
I might have got what the issue is, not sure if it's the same. On a headset removal, the event corresponding to the removal get spammed to the host until a timeout of the event and a disconnection event happens, see image below:
[image]
Couple of seconds later, a disconnect event happens:
[image]
After a reconnection with audio working fine, this spurious events still get spammed from the headset randomly.
[image]

Can we just try to respond to this events as ACK and ignore them? By the way, audio pause do not work on Linux on removal, but it works on Android. I'll try to pcap on Android a pause/playback event to check if these are just how the headset reports pause to the host, but I'm not really an expert on bluetooth protocols, so I might be completely offbase with my speculation.

Pcap attached to this post.
[bluetooth.tar.gz](https://github.com/bluez/bluez/files/12596214/bluetooth.tar.gz)

RodoMa92, 2023-09-13:
Ok, just did additional testing on it, this are 5.2 bluetooth buds that supports multipoint bluetooth audio according to the declared specs.

It seems that on the removal on a single earbud (only one of the two of them) bluetooth data stops for around 10 seconds (using tshark I can see that during that period no bluetooth packets are flowing beetween them, the counter is stuck) until the above disconnect happens. 

I have the feeling that one of the two buds "manages" the connection and the other synchronize on it (I'm not an expert on multipoint, so I might be completely wrong). 

When removed, the buds sends out the above event to signal to the host adapter that it should switch the master to the other one, however it doesn't get any ACK back from the system and therefore kill connection. During this period I have no audio on both buds. If I pull out the non main one nothing happens (but the music doesn't stop), no matter how much I trigger the event reinserting the bud and remove it. If I use a single buds of the two, nothing happens to the audio stream (besides not pausing) on removal.

Since I do not know if this is a bluez bug or a Pipewire audio bug, I'll also open an issue on the pipewire team to see if they can help.

Marco

RodoMa92, 2023-09-13:
Here is an android log of a successful removal with pause and reinsert with resume (it was harder to get than anticipated, thanks Google):
[btsnoop_hci.tar.gz](https://github.com/bluez/bluez/files/12597615/btsnoop_hci.tar.gz)

Vudentz, 2023-09-14:
@RodoMa92 what BlueZ and Kernel versions?

RodoMa92, 2023-09-15:
Kernel 6.4.14-200.fc38.x86_64
Bluez 5.69

RodoMa92, 2023-09-15:
Additional info here https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3508

RodoMa92, 2023-09-15:
Another interesting difference seems that Bluez negotiates AAC at 192 kbit/s, compared to 256 kbit/s of Android. Odd.

By the way, my Android device runs a open source implementation of Android, so no proprietary Bluetooth drivers AFAIK.

RodoMa92, 2023-10-03:
Any additional debugging steps that I can perform?

Vudentz, 2023-10-03:
@RodoMa92 please use btmon to collect the HCI trace.

RodoMa92, 2023-10-05:
Would love to do it, however:
```sudo btmon -i hci0 -w hcidump.log
Bluetooth monitor ver 5.70
*** buffer overflow detected ***: terminated
[1]    30549 IOT instruction  sudo btmon -i hci0 -w hcidump.log
```
Any solutions?

Vudentz, 2023-10-05:
@RodoMa92 run it with valgrind so it is able to generate a backtrace.

RodoMa92, 2023-10-05:
It just get stuck like this forever
```
==23073== Memcheck, a memory error detector
==23073== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==23073== Using Valgrind-3.21.0 and LibVEX; rerun with -h for copyright info
==23073== Command: btmon -i hci0 -w hcidump.log
==23073== 
```
No logfile seems to be generated though.

RodoMa92, 2023-10-05:
Running with root without sudo made it works perfectly fine finally!

Here is the log and an interesting observation made while reproducing the issue.

I've tried twenty times and I couldn't get it to do it. As soon as I've realized that I had a bluetooth keyboard connected and turned it off manually, on the next removal this triggered istantly. After a correct resync and a removal, again immediately happened again but resynced it file. Finally I've kept going until I've got the ping pong issue with final failure on reconnect until a manual reconnection from KDE bluetooth manager.

Log attached:
[btmon.tar.gz](https://github.com/bluez/bluez/files/12822476/btmon.tar.gz)

RodoMa92, 2025-05-29:
I don't have the hardware anymore to be able to test this so I'll close this issue for now.

closed 2025-05-29 by RodoMa92