bluez/bluez #586: [5.68] Logitech MX Series paired but not working

terrancewong, 2023-08-28:
Logitech MX Anywhere3 mouse can be paired, and connected, but not working (none of pointing, wheel, button).

Bluez 5.66 with Kernel 6.1.27 works fine.

# journal when issue happens

```

Aug 28 13:56:22 <hostname> blueman-manager[6189]: blueman-manager 13.56.22 WARNING  ManagerDeviceList:396 _monitor_power_levels: Failed to get power levels, probably a LE device.
Aug 28 13:56:22 <hostname> bluetoothd[1046]: src/service.c:service_accept() input-hog profile accept failed for <M:A:C>
Aug 28 13:56:24 <hostname> bluetoothd[1046]: src/service.c:service_accept() input-hog profile accept failed for <M:A:C>

```

# system info

```
% lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
~/ssh % uname -a
Linux <hostname> 6.4.0-3-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.4.11-1 (2023-08-17) x86_64 GNU/Linux
% apt show bluez 
Package: bluez
Version: 5.68-2
Priority: optional
Section: admin
Maintainer: Debian Bluetooth Maintainers <team+pkg-bluetooth@tracker.debian.org>
Installed-Size: 4,262 kB
Pre-Depends: init-system-helpers (>= 1.54~)
Depends: libasound2 (>= 1.0.17), libc6 (>= 2.34), libdbus-1-3 (>= 1.9.14), libdw1 (>= 0.127), libglib2.0-0 (>= 2.75.3), libreadline8 (>= 6.0), libudev1 (>= 196), kmod, udev, default-dbus-system-bus | dbus-system-bus
Suggests: pulseaudio-module-bluetooth
Homepage: http://www.bluez.org
Tag: admin::hardware, admin::kernel, devel::lang:c, devel::library,
 hardware::bluetooth, implemented-in::c, protocol::TODO,
 role::devel-lib, role::program, use::driver, use::synchronizing
Download-Size: 1,182 kB
APT-Manual-Installed: yes
APT-Sources: http://mirrors.rit.edu/debian unstable/main amd64 Packages
Description: Bluetooth tools and daemons
```

Vudentz, 2023-08-28:
@terrancewong my MX Anywhere 3 is working just fine here on fedora, perhaps this has been fixed  on master, please check if you can still reproduce it with 5.69 that was just released.

terrancewong, 2023-08-28:
#529  might be the same issue

terrancewong, 2023-08-30:
plus hcidump. 

mouse can be paired and connected. events  received. Don't know what keeps it from connecting to xinput.

[20230829.hcidump.log](https://github.com/bluez/bluez/files/12478671/20230829.hcidump.log)

terrancewong, 2023-08-30:
```
modprobe hid-logitech
modprobe hid-logitech-hidpp
```
still the same.

terrancewong, 2023-11-15:
is anything out of ordinary? 

lsmod:

```
# lsmod | grep hid
hidp                   36864  1
bluetooth            1126400  65 btrtl,hidp,btmtk,btintel,btbcm,bnep,btusb,rfcomm
intel_hid              28672  0
sparse_keymap          12288  1 intel_hid
hid_multitouch         32768  0
usbhid                 73728  0
hid_generic            12288  0
i2c_hid_acpi           12288  0
usbcore               409600  12 xhci_hcd,usbnet,snd_usb_audio,usbhid,snd_usbmidi_lib,usb_storage,uvcvideo,btusb,xhci_pci,aqc111,cdc_ether,uas
i2c_hid                36864  1 i2c_hid_acpi
hid                   176128  5 i2c_hid,hidp,usbhid,hid_multitouch,hid_generic
```

`systemctl restart bluetooth`

```
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Terminating
Nov 14 21:01:13 <hostname> systemd[1]: Stopping bluetooth.service - Bluetooth service...
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_hq
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_sq
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_mq
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/aptx_hd
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/aptx_hd
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/aptx
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/aptx
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/sbc
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/sbc
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Endpoint unregistered: sender=:1.67 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Nov 14 21:01:13 <hostname> dbus-daemon[1061]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.67" (uid=1000 pid=2115 comm="/usr/bin/pulseaudio --daemonize=no --log-target=jo") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.167" (uid=0 pid=12448 comm="/usr/libexec/bluetooth/bluetoothd")
Nov 14 21:01:13 <hostname> dbus-daemon[1061]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.67" (uid=1000 pid=2115 comm="/usr/bin/pulseaudio --daemonize=no --log-target=jo") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.167" (uid=0 pid=12448 comm="/usr/libexec/bluetooth/bluetoothd")
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Stopping SDP server
Nov 14 21:01:13 <hostname> bluetoothd[12448]: Exit
Nov 14 21:01:13 <hostname> systemd[1]: bluetooth.service: Deactivated successfully.
Nov 14 21:01:13 <hostname> systemd[1]: Stopped bluetooth.service - Bluetooth service.
Nov 14 21:01:13 <hostname> (uetoothd)[12716]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Nov 14 21:01:13 <hostname> systemd[1]: Starting bluetooth.service - Bluetooth service...
Nov 14 21:01:13 <hostname> bluetoothd[12716]: Bluetooth daemon 5.70
Nov 14 21:01:13 <hostname> systemd[1]: Started bluetooth.service - Bluetooth service.
Nov 14 21:01:13 <hostname> bluetoothd[12716]: Starting SDP server
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support csip plugin
Nov 14 21:01:13 <hostname> bluetoothd[12716]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support micp plugin
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support vcp plugin
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support mcp plugin
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support bass plugin
Nov 14 21:01:13 <hostname> bluetoothd[12716]: src/plugin.c:plugin_init() System does not support bap plugin
Nov 14 21:01:13 <hostname> dbus-daemon[1061]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.170' (uid=0 pid=12716 comm="/usr/libexec/bluetooth/bluetoothd")
Nov 14 21:01:13 <hostname> bluetoothd[12716]: Bluetooth management interface 1.22 initialized
Nov 14 21:01:13 <hostname> systemd[1]: Starting systemd-hostnamed.service - Hostname Service...
[... cut ...]
Nov 14 21:01:15 <hostname> kernel: input: UGREEN B3.0 Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.0/bluetooth/hci0/hci0:256/0005:2B89:3632.0007/input/input20
Nov 14 21:01:15 <hostname> kernel: hid-generic 0005:2B89:3632.0007: input,hidraw2: BLUETOOTH HID v0.01 Mouse [UGREEN B3.0 Mouse] on <MAC>
Nov 14 21:01:19 <hostname> bluetoothd[12716]: src/service.c:service_accept() input-hog profile accept failed for <MAC_Logi>
Nov 14 21:01:21 <hostname> bluetoothd[12716]: src/service.c:service_accept() input-hog profile accept failed for <MAC_Logi>
```

closed 2025-08-07 by github-actions[bot]