bluez/bluez #594: [5.69] Somewhat broken, was the release pulled?

kode54, 2023-09-04:
I was about to make a formal issue report against 5.69, but it appears that no such release exists yet? I was experiencing issues with both my Xbox Series X|S Controller and a DualShock 4 Controller, where they would either randomly disconnect, or fail to connect properly in the first place. Reverting to 5.68 fixed the problem.

lavjamanxd, 2023-09-05:
Just saw your bug report, and can confirm downgrading to 5.68 fixed my DS4 controller as well.

Vudentz, 2023-09-06:
Can you guys try with the latest, I suspect the following change should have it fixed:

https://github.com/bluez/bluez/commit/670f0d0f618d3f122e6d95c34de53f59a343a15f

ComradeClaritux, 2023-09-07:
I have this issue as well with a DS4 and a DS5 controller. Can confirm that it's working again with latest commit

lilxandrias, 2023-09-14:
Can confirm the issue, this is **critical** and Bluez 5.69.1 or 5.70 needs to be released, or else some devices are not working. Audio was working for me, but Dualshock 4 complete blackout.

fionnb, 2023-09-14:
Can confirm the issue, too. Controller not pairing on 5.69; downgraded to 5.68 and everything is fine.

Vudentz, 2023-09-14:
@fionnb can you try with the latest just confirm it has been fixed upstream.

known-as-bmf, 2023-09-20:
Hello and thanks for the hard work 

I think I am also affected by this issue (openSUSE Tumbleweed, bluez 5.69, trying to connect to a Sony DualSense Controller (PS5 gamepad)) 

From this thread, I have the impression that a fix has already been implemented (apparently tested by @lilxandrias [here](https://github.com/bluez/bluez/commit/670f0d0f618d3f122e6d95c34de53f59a343a15f#commitcomment-127311178)), and I would like to know if you have a rough ETA for releasing a new version that includes the aforementioned fix.

Unfortunately I am not an advanced enough user to manually compile / install and try out the current master branchon my system.

Just for reference, here are more details about the issue I am experiencing (I am using GNOME settings to manage my BT devices):
* gamepad in pairing mode is detected properly
* gamepad pairs properly
* when connecting to the device (automatically after pairing or manually when the gamepad is already paired), the gamepad turns off by itself

relevant logs from systemd's `bluetooth.service` (I redacted my machine hostname and gamepad device id)

```
sept. 20 17:28:37 xxxxxxxxxx bluetoothd[1586]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
sept. 20 17:28:37 xxxxxxxxxx bluetoothd[1586]: profiles/input/server.c:confirm_event_cb() Refusing connection from XX:XX:XX:XX:XX:XX: unknown device
```

Sorry if it is too much noise but I would also like to mention @DimStar77 who I think is the maintainer for https://build.opensuse.org/package/show/openSUSE:Factory/bluez because it might be possible to release the fix for openSUSE as a patch somehow?

thanosngplus, 2023-09-23:
Can confirm that the issue still exists in the latest version, and downgrading to 5.68 fixed it!

kimec, 2023-09-24:
Hi, I am reporting Fedora 38 is impacted too. My DS4 knock off gamepads stopped working after the upgrade to 5.69.  They worked flawlessly before that. Downgrade to 5.66 via `dnf downgrade bluez\*` "fixes" the issue.

Natosphere, 2023-09-28:
> Hi, I am reporting Fedora 38 is impacted too. My DS4 knock off gamepads stopped working after the upgrade to 5.69. They worked flawlessly before that. Downgrade to 5.66 via `dnf downgrade bluez\*` "fixes" the issue.

Can confirm. On a new install of Fedora 38 Plasma, my 8bitdo pro 2 bluetooth gave errors in pairing (in all 4 pairing modes). The downgrade to 5.66 was a quick "fix".

Vudentz, 2023-09-28:
https://github.com/bluez/bluez/releases/tag/5.70

closed 2023-09-28 by Vudentz

kimec, 2023-10-27:
Just reporting the Fedora 38 is now shipping bluez 5.70-1 but the version still does not work with my DS4s. I had to downgrade yet again. The last working version on Fedora 38 is still 5.66.

fionnb, 2023-10-29:
I also tried 5.70 on Arch (Endeavour), still not working, had to downgrade 5.70-1 => 5.68-1 again.

mikebm94, 2023-10-29:
I can confirm the issue is not fixed by version 5.70. I still cannot connect a DS4 controller.

IsaacVaughn, 2023-11-03:
Still having this problem on 5.70, controller works properly after downgrading to 5.68. Errors are:
Nov 02 20:06:54 manjaro bluetoothd[2665913]: src/service.c:service_probe() input-hid profile probe failed for E4:17:D8:9D:A6:E7
Nov 02 20:08:21 manjaro bluetoothd[2665913]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
Nov 02 20:08:21 manjaro bluetoothd[2665913]: profiles/input/server.c:confirm_event_cb() Refusing connection from E4:17:D8:9D:A6:E7: unknown device
Nov 02 20:08:47 manjaro bluetoothd[2665913]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
Nov 02 20:08:47 manjaro bluetoothd[2665913]: profiles/input/server.c:confirm_event_cb() Refusing connection from E4:17:D8:9D:A6:E7: unknown device

Vudentz, 2023-11-03:
> Still having this problem on 5.70, controller works properly after downgrading to 5.68. Errors are: Nov 02 20:06:54 manjaro bluetoothd[2665913]: src/service.c:service_probe() input-hid profile probe [...]

Perhaps it is related to https://github.com/bluez/bluez/issues/614?

hawkeye116477, 2023-11-04:
> Still having this problem on 5.70, controller works properly after downgrading to 5.68. 

Looks like that's already fixed, but not released yet, so you can also switch to bluez-git package from AUR and it should also work.

Vudentz, 2023-11-04:
I will check with @holtmann if we can make a new release.