bluez/bluez #597: 2 echo pop, not the same codecs

promeneur, 2023-09-08:
My PC get an Intel AX210 (Bluetooth 5.3)
openSUSE Tumbleweed, KDE, pipewire

Two Amazon Echo Pop connected at the same time to the PC via Bluetooth.
  
Surprisingly, one uses SBC and the other one uses SBC-XQ.

I can't switch to SBC or SBC-XQ according to get the same codec for the twos.

pv, 2023-09-10:
Using same codec for A2DP doesn't work right now, see https://lore.kernel.org/linux-bluetooth/20220605122927.110627-1-pav@iki.fi/T/#u but haven't got around to fixing it so far.

Vudentz, 2023-09-11:
@pv that change has been applied though, but perhaps you are referring to the fact that the spec has some wording saying a third device shouldn't be able to connect which makes A2DP unusable for setting up multiple devices, that said many devices on the market do support the so called multi-point connections: 

https://www.soundguys.com/bluetooth-multipoint-explained-28601/

Now perhaps we should add a property on MediaEndpoint so the endpoint implementation can indicate multi-point is supported, etc, so it can be reflected on AVDTP in-use flag.

ValeZAA, 2025-05-18:
>I can't switch to SBC or SBC-XQ according to get the same codec for the twos.

here I could switch but it caused a broken connect after 4 times. Same happened on windows. I do think it is a different case though, I have a stereopair, i.e. bluez just connect to one of the speakers and 2nd speaker gets right audio in the stereo sound. Completly different.