bluez/bluez #598: Bluez tries to be both BAP Client and Server at the same time

pv, 2023-09-10:
When connecting Pipewire+BlueZ <-> Pipewire+BlueZ, BlueZ acts as both BAP Server and Client on both sides.

This appears not allowed in BAP v1.0.1 Sec 2.4: "A device shall not concurrently occupy the Unicast Client role and Unicast Server role in a connection to a single device because the Unicast Client occupies the LL Central role, and both devices cannot occupy the LL Central role in the same connection, as defined in Volume 6, Part B, Section 1.1.1 in [1]."

In this setup, bluetoothctl shows
```
[fedora]# transport.list 
Transport /org/bluez/hci0/dev_70_1A_B8_98_FF_A2/fd2 
Transport /org/bluez/hci0/dev_70_1A_B8_98_FF_A2/fd3 
Transport /org/bluez/hci0/dev_70_1A_B8_98_FF_A2/pac_source0/fd4 
Transport /org/bluez/hci0/dev_70_1A_B8_98_FF_A2/pac_sink0/fd5 
```
so there are both BAP Client and Server transports simultaneously present for the same device.

In bluetoothd logs on the two sides I see
```
syys 10 19:50:04 monolith bluetoothd[51115]: /home/pauli/prj/external/bluez/buildsan/../src/adapter.c:connected_callback() hci0 device 70:1A:B8:98:FF:A2 connected eir_len 0
syys 10 19:50:04 monolith bluetoothd[51115]: /home/pauli/prj/external/bluez/buildsan/../src/gatt-database.c:connect_cb() New incoming LE ATT connection
syys 10 19:50:04 monolith bluetoothd[51115]: /home/pauli/prj/external/bluez/buildsan/../src/device.c:device_attach_att() Elevating security level since LTK is available
syys 10 19:50:04 monolith bluetoothd[51115]: /home/pauli/prj/external/bluez/buildsan/../attrib/gattrib.c:g_attrib_ref() 0x60700001a030: g_attrib_ref=1
syys 10 19:50:04 monolith bluetoothd[51115]: /home/pauli/prj/external/bluez/buildsan/../src/device.c:device_accept_gatt_profiles() initiator true
```
and
```
syys 10 19:50:03 fedora bluetoothd[1131]: src/adapter.c:connected_callback() hci0 device 84:5C:F3:52:11:91 connected eir_len 3
syys 10 19:50:04 fedora bluetoothd[1131]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
syys 10 19:50:04 fedora bluetoothd[1131]: src/device.c:device_attach_att() Elevating security level since LTK is available
syys 10 19:50:04 fedora bluetoothd[1131]: attrib/gattrib.c:g_attrib_ref() 0x607000015170: g_attrib_ref=1
syys 10 19:50:04 fedora bluetoothd[1131]: src/device.c:device_accept_gatt_profiles() initiator true
```
It looks as if both sides think they are initiator.

Pipewire assumes it is either BAP Client or BAP Server but not both, so fails to work properly in this case (this role also cannot be controlled by sound server which only registers PACs that can be used for either). AX210 also fails Create CIS with Memory Capacity Exceeded when in this state, probably since it only can do one CIG so no client and server at the same time. It works, if I fail `bap_accept` on one side to force it to be server only.

Vudentz, 2023-09-11:
@pv not sure how you managed to reproduce this, are you advertising on both ends and changing who is initiating? Anyway I planning to use the MGMT New Connection flag as initiator, lets see if that is more reliable then checking what key type we got.

closed 2023-09-12 by BluezTestBot

pv, 2023-09-13:
Thanks, works correctly on master now.

Reproduction was: Pipewire+BlueZ running on two different devices (here: host and VM) with separate BT controllers, no special configuration except ControllerMode=le on both sides. Pair devices and power cycle. Start advertising on one side, then the other side autoconnects.