bluez/bluez #602: Reimplementing ESP32 code setting BLE security level with bluez

not-my-profile, 2023-09-16:
I'd like to emulate the BR-E1 wireless remote control from Canon with bluez. My code works fine except that I'm running into the same issue as is described in https://github.com/ArthurFDLR/BR-M5/issues/2 (the camera closes the connection after 3 minutes and only allows reconnecting after re-pairing, restarting the camera also runs into the same issue), the solution that has been figured out there is:

> I've added `BLEDevice::setEncryptionLevel()` section into my code.
> My assumption from the links below is that setting BLE security is required in BLE bounding process.

[CanonBLERemote.cpp] in that repository contains code such as:

```cpp
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT_NO_MITM);
    BLEDevice::setSecurityCallbacks(new SecurityCallback());
```

I'd be curious how such behavior could be replicated with the bluez D-Bus API. Looking at the [device API](https://github.com/bluez/bluez/blob/master/doc/device-api.txt) there do not appear to be any equivalent methods.

The most relevant property appears to be `Bonded`. While `device-api.txt` doesn't explicitly mention encryption:

> **boolean Bonded [readonly]**
> Indicates if the remote device is bonded. Bonded means the information exchanged on pairing process has been stored and will be persisted.

Some [random article](https://community.silabs.com/s/article/x-reference-bonding-encryption-and-mitm-protection-with-ble-modules?language=en_US) defines:

> Bonding is the long-term storage of encryption info (particularly keys) so that the devices “know” each other and can easily reconnect in an encrypted way.

Which does appear to be relevant. However the `Bonded` Device API property has only been introduced in version 5.65, and I am on 5.55-3.1 (Debian 11), so I don't even know how to check if my `Pair()` call resulted in a successful bond.

I'm very much out of my depth when it comes to BLE encryption and would appreciate some pointers on if the logic in [CanonBLERemote.cpp] can be reimplemented with bluez.

[CanonBLERemote.cpp]: https://github.com/ArthurFDLR/BR-M5/blob/02b158d3c842d7ad9a6f3f7b98966e8590d70814/lib/CanonBLERemote/src/CanonBLERemote.cpp

Vudentz, 2023-09-22:
@not-my-profile on LE the pairing procedure is normally triggered on demand based on the GATT attribute permission required:

https://github.com/bluez/bluez/blob/master/doc/gatt-api.txt#L256

If you are triggering the pairing procedure directly via Device.Pair then the IO Capabilities are exchanged and based on that the stack decides on what to do, usually it attempts the highest possible security, but perhaps something is preventing the right level to be selected, or we are attempting MITM which the device doesn't support, anyway we might need the HCI traces to check what is going on (please use btmon to collect that while attempting to pair).

alex-odn, 2024-07-29:
Hi @not-my-profile.
You can use btmgmt to enable privacy. 
After connecting and pairing with a camera do:
  sudo btmgmt
  privacy on