bluez/bluez #604: Roland GO:KEYS MIDI connection support

mwprado, 2023-09-17:
HI, I bought a GO:KEYS MIDI keyboard for my kids. On scaning I can see GO:KEYS MIDI and GO:KEYS AUDIO.

The GO:KEYS AUDIO connection works perfectly, but,  GO:KEYS MIDI midi connection don't  work.

There are some information:

Product:
GO:KEYS - Music Creation Keyboard (GO-61K)
https://www.roland.com/global/products/gokeys_go-61k/specifications/

Main Specs:
61 keys (Ivory Feel and Box-shape Keys with velocity)
Touch Sensitivity: Key Touch: 3 types, fixed touch
Maximum Polyphony: 128 voices
MIDI Format: Conforms to GM2
Recorder
Save Format: Standard MIDI Files (Format 0)
Bluetooth
Bluetooth Ver 4.2:
  Profile Support: A2DP(Audio), GATT(MIDI over Bluetooth Low Energy)
  Codec: SBC (Support to the content protection of the SCMS-T method)
  *This function is not available for GO-61KL.

$ bluetoothctl --version
bluetoothctl: 5.69

[bluetooth]# show
Controller AC:67:5D:FF:E3:C5 (public)
	Name: fedora
	Alias: fedora
	Class: 0x007c010c (8126732)
	Powered: yes
	PowerState: on
	Discoverable: yes
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0545
	Discovering: yes
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0c (12)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded

[bluetooth]# devices
Device 34:81:F4:62:FF:07 GO:KEYS Audio
Device CB:81:F4:62:FF:07 GO:KEYS MIDI
Device A4:AE:11:0E:2C:12 Wireless Controller

[bluetooth]# info CB:81:F4:62:FF:07 
Device CB:81:F4:62:FF:07 (random)
	Name: GO:KEYS MIDI
	Alias: GO:KEYS MIDI
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
	Modalias: bluetooth:v0039p1582d4200
[bluetooth]# info 34:81:F4:62:FF:07 
Device 34:81:F4:62:FF:07 (public)
	Name: GO:KEYS Audio
	Alias: GO:KEYS Audio
	Class: 0x00240414 (2360340)
	Icon: audio-card
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v0039p1582d4200
	ManufacturerData Key: 0x5349 (21321)
	ManufacturerData Value:
  53 43                                            SC 
  
$ rfkill list
0: acer-wireless: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: acer-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
4: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

[image]

[bluetoothd-gokeys.out.log](https://github.com/bluez/bluez/files/12643025/bluetoothd-gokeys.out.log)

mwprado, 2023-09-17:
I've found this thread with some clues.
https://www.spinics.net/lists/linux-bluetooth/msg89475.html

Vudentz, 2023-09-19:
@mwprado well you probably need to recompile with midi support:

https://github.com/bluez/bluez/blob/master/README#L282

Or you can handle the attributes on your own, using the D-Bus API:

https://github.com/bluez/bluez/blob/master/doc/gatt-api.txt

mwprado, 2023-09-19:
Thank you I've recompiled with bluettoth support, but not working yet. I searched in mailing list and you had commented about this keyboard. I trying debug the source to figure out where is the problem. I'll also give a try in gatt-api.

mwprado, 2023-09-20:
Data with gattctl:

$ gattctl --connect cb:81:f4:62:ff:07
Connecting...
Terminate with Ctrl+C
[cb:81:f4:62:ff:07] Discovered, alias = GO:KEYS MIDI
[cb:81:f4:62:ff:07] Connected
[cb:81:f4:62:ff:07] Discovered, alias = GO:KEYS MIDI
[cb:81:f4:62:ff:07] Resolved services
[cb:81:f4:62:ff:07]  Service [0000180a-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a2a-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a23-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a28-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a27-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]  Service [00001801-0000-1000-8000-00805f9b34fb]
[cb:81:f4:62:ff:07]    Characteristic [00002a05-0000-1000-8000-00805f9b34fb]

mwprado, 2023-10-15:
When I run gatttool in interactive mode, I got data below:

[CB:81:F4:62:FF:07][LE]>primaries
attr handle: 0x0031, end grp handle: 0x0041 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x0001, end grp handle: 0x0003 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0007, end grp handle: 0x000f uuid: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
attr handle: 0x0021, end grp handle: 0x0023 uuid: 00001801-0000-1000-8000-00805f9b34fb

[CB:81:F4:62:FF:07][LE]>characteristics
handle: 0x0002, char properties: 0x0a, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0008, char properties: 0x18, char value handle: 0x0009, uuid: 49535343-aca3-481c-91ec-d85e28a60318
handle: 0x000b, char properties: 0x1e, char value handle: 0x000c, uuid: 7772e5db-3868-4112-a1a9-f2669d106bf3
handle: 0x000e, char properties: 0x08, char value handle: 0x000f, uuid: 49535343-8841-43f4-a8d4-ecbe34729bb3
handle: 0x0022, char properties: 0x26, char value handle: 0x0023, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x0032, char properties: 0x02, char value handle: 0x0033, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0034, char properties: 0x02, char value handle: 0x0035, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0036, char properties: 0x02, char value handle: 0x0037, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x0038, char properties: 0x02, char value handle: 0x0039, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x003a, char properties: 0x02, char value handle: 0x003b, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x003c, char properties: 0x02, char value handle: 0x003d, uuid: 00002a28-0000-1000-8000-00805f9b34fb
handle: 0x003e, char properties: 0x02, char value handle: 0x003f, uuid: 00002a23-0000-1000-8000-00805f9b34fb
handle: 0x0040, char properties: 0x02, char value handle: 0x0041, uuid: 00002a2a-0000-1000-8000-00805f9b34fb

In other hand, when I run bluetoothctl and try list attributes, display that info

[GO:KEYS MIDI]# list-attributes 
Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0021
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0021/char0022
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char0032
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char0034
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char0036
	00002a25-0000-1000-8000-00805f9b34fb
	Serial Number String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char0038
	00002a27-0000-1000-8000-00805f9b34fb
	Hardware Revision String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char003a
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char003c
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char003e
	00002a23-0000-1000-8000-00805f9b34fb
	System ID
Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_CB_81_F4_62_FF_07/service0031/char0040
	00002a2a-0000-1000-8000-00805f9b34fb
	IEEE 11073-20601 Regulatory Cert. Data List

I can't figure out why the characteristcs below (one of them is MIDI I/O) are not linked in any service.
handle: 0x0008, char properties: 0x18, char value handle: 0x0009, uuid: 49535343-aca3-481c-91ec-d85e28a60318
handle: 0x000b, char properties: 0x1e, char value handle: 0x000c, uuid: 7772e5db-3868-4112-a1a9-f2669d106bf3
handle: 0x000e, char properties: 0x08, char value handle: 0x000f, uuid: 49535343-8841-43f4-a8d4-ecbe34729bb3

In Android BLE Scanner App scanner that chractheristics are linked to 03b80e5a-ede8-4b33-a751-6ce34ec4c700 service as showed below:
[image]

mwprado, 2023-10-23:
I've collected more data from log with specific errors:
[trace.log](https://github.com/bluez/bluez/files/13064909/trace.log)

mwprado, 2023-11-21:
thi is the btmon log
[btmon-gokeys-disconnect-connect.log](https://github.com/bluez/bluez/files/13431819/btmon-gokeys-disconnect-connect.log)

mwprado, 2023-11-23:
In btmon log, I could see this problem:
= bluetoothd: profiles/midi/midi.c:midi_accept() Could not open ALSA Sequencer: No such file or directory (-2)                                                                                                                      19.072129
= bluetoothd: src/service.c:service_accept() MIDI GATT Driver profile accept failed for CB:81:F4:62:FF:07                                                                                                                           19.072166

mwprado, 2023-11-26:
@Vudentz  
Why does gatttool list charactheristics with handle like in:
handle: 0x000b, char properties: 0x1e, char value handle: 0x000c, uuid: 7772e5db-3868-4112-a1a9-f2669d106bf3

but when I connect with blutoothctl, don't list that characteristic.

mwprado, 2023-12-01:
> In btmon log, I could see this problem: = bluetoothd: profiles/midi/midi.c:midi_accept() Could not open ALSA Sequencer: No such file or directory (-2) 19.072129 = bluetoothd: src/service.c:service_accept() MIDI GATT Driver profile accept failed for CB:81:F4:62:FF:07 19.072166

I put user in audio group and disable SELINUX. There isn't any error now. I can see Go:KEYS in alsa and pipewire.

$ aplaymidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
128:0    FluidSynth GM                    FluidSynth GM
129:0    GO:KEYS MIDI                     GO:KEYS MIDI Bluetooth

$ arecordmidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
129:0    GO:KEYS MIDI                     GO:KEYS MIDI Bluetooth

Unfortunately, I can't play midi on the keyboard or read notes from it.
and I can see in pipewire graph and zrythm but there ins't any sound yet.

[image]

mwprado, 2023-12-12:
Device cache for GO:KEYS shows info below and is diferent from Gatttool posted before.

[1]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=1800
EndGroupHandle=3

[7]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=03B80E5AEDE84B33A7516CE34EC4C700
EndGroupHandle=15

[33]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=1801
EndGroupHandle=35

[49]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=180A
EndGroupHandle=65

desfonema, 2024-04-20:
Hi! Same issue found here. There is a longer discussion at https://linuxmusicians.com/viewtopic.php?t=25227
I can help debugging/troubleshooting if needed. Just let me know what to run and I'll get back with the dumps.

maksut, 2024-05-12:
I believe this is not specific to GO:KEYS. I have Roland FP10. Similarly, bluetooth connection is OK but no midi data received.
Again the midi connection works fine with android. I can also try to help with debugging/troubleshooting.

pavelbrych, 2024-05-16:
Same for me with Roland FP-10, device is there but no midi data. Also can help if any testing/debugging needed.

pavel@localhost:~$ arecordmidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
 32:0    Virtual Raw MIDI 4-0             VirMIDI 4-0
128:0    FP-10                            FP-10 Bluetooth

Vudentz, 2024-05-16:
Perhaps we need to compare the HCI traces from and Android with BlueZ to see what is different.

maksut, 2024-05-16:
If this analysis true then it's Roland being non-compliant: https://www.spinics.net/lists/linux-bluetooth/msg89475.html

The conclusion was that the device uses a mix of different UUID sizes. Not saying bluez should workaround it. But it's interesting that all other OSes seem to tolerate it.

Vudentz, 2024-05-16:
@maksut hmm so there is a problem with discovery then not able to discover certain range when it intermixes UUID16 with UUID128? That would be non-complaint since the spec requires them to not be intermixed like that, but perhaps we could attempt to detect that some how. Can someone here paste the traces using btmon -w <filename> so I can try to find a way to decode that.

pavelbrych, 2024-05-16:
I kept trying and found that it works only when device is paired first time. Not sure what info will be relevant, so:

OS: Fedora 39

Digital piano - Roland FP-10 - latest firmware (there are some fixes for bluetooth!):
https://www.roland.com/global/support/by_product/fp-10/updates_drivers

I compiled and installed bluez from https://mirrors.edge.kernel.org/pub/linux/bluetooth

```
./configure --enable-midi --with-systemdsystemunitdir=/etc/systemd/system
make
sudo make install
``` 

Now, whenever i freshly pair my device (after deleting it from known devices using "forget" button), it works!

But when disconnected and connected again, it connects, it's visible in pipewire, can be connected to but does not work. PC restart does nothing, only after device is deleted and freshly paired it works again.

Attached are btmon -w files from first pairing vs connecting known device:

[btmon_dump.zip](https://github.com/bluez/bluez/files/15342154/btmon_dump.zip)

desfonema, 2024-05-17:
It is interesting that both Apple devices (iOS, and OSX) work out of the box, and also Windows, and Android. I know BLE is somewhat new to BlueZ so maybe there is something others are doing and we are missing.

pavelbrych, 2024-05-17:
Following details may not be related to this bluez issue, but maybe it will be helpful for somebody.

It's also possible that FP-10 problems are not connected to originaly reported bluez issue at this thread at all, only the symptoms are same. I know basically nothing about bluetooth and it's implementation, but if anybody wants some debug data to test anything, I can do that.

Today I tested Roland FP-10 on Windows using MIDIberry app and it behaves basically the same as for Linux - when freshly paired, device is "connected" and works (I can see MIDI data in MIDIberry). When disconnected and connected again, device is there with status "connected" but no MIDI data, until I remove the device and freshly pair it again.

I don't have any Apple devices, but I will check with Android and let you know later.

I observed that FP-10 stay connected only when there is some software using it. I guess this is some general feature of bluetooth, not specific to this particular case, but it has consequences for this issue:

If I pair FP-10 without MIDIberry running, it will connect briefly and when nobody using it, disconnect. If you run MIDIberry afterwards, FP-10 buletooth status changes to connected, but does not work.

If I run MIDIberry before FP-10 pairing, it will grab FP-10 device as soon as connected and it stays connected and works.

Note for anybody who would want to use this as workaround to get FP-10 BLE MIDI working - it seems that while MIDIberry is running, it keeps bluetooth device "in use", regardless of bluetooth status. You cannot remove and successfully pair FP-10 again while MIDIberry is running. To reset everything, you need to close MIDIberry, remove FP-10 from paired devices, open MDIDIberry and pair FP-10 again.

mwprado, 2024-07-16:
> @maksut hmm so there is a problem with discovery then not able to discover certain range when it intermixes UUID16 with UUID128? That would be non-complaint since the spec requires them to not be intermixed like that, but perhaps we could attempt to detect that some how. Can someone here paste the traces using btmon -w so I can try to find a way to decode that.

the Go:keys log from my android phone:
[btsnoop_hci.zip](https://github.com/user-attachments/files/16241736/btsnoop_hci.zip)

mwprado, 2024-07-22:
A related project:
https://github.com/waldt/goplus

desfonema, 2025-05-31:
Issue still present with bluez  5.82-1  

A different MIDI keyboard BLE that I have is working now in this version but the GO:Keys still can't do MIDI

mwprado, 2025-11-09:
fedora 43, same problem.