bluez/bluez #605: Conenct problems with Gamepads (Authentication Failed)

DrKlipper, 2023-09-18:
Hi !
I use a Raspberry Pi Zero 2 which has embedded BT. I want to connect a little gamepad: 
This one : https://www.amazon.de/auvisio-Gamepad-Bluetooth-Mini-Akku-Game-Controller-Fernbedienung-black/dp/B0179P8WMG

It´s working in Windows like a charm and I know it worked ~2 years ago as I bought it. 

But when I try it now it drives me crazy :-) This is my workflow: 

- sudo bluetoothctl
- scan on
- turn on the gamepad
- wait to get the new device message in bluetoothctl
- pair the ID
- trust the ID (but this doesn´t really change anything at all)
- connect the ID

The result is this: 
```pi@Pi2W:~/blue/bluez $ sudo bluetoothctl
Agent registered
[bluetooth]# scan on
hci0 type 7 discovering on
Discovery started
[CHG] Controller B8:27:EB:39:52:35 Discovering: yes
[NEW] Device 2C:90:1F:28:8D:3C 2C-90-1F-28-8D-3C
[NEW] Device 4F:56:CA:87:45:04 4F-56-CA-87-45-04
[bluetooth]# devices
Device 17:05:16:03:35:E1 MOCUTE-032S_A03-35E1
Device 08:A9:5A:91:95:B2 Sony PLAYSTATION(R)3 Controller
Device 2C:90:1F:28:8D:3C 2C-90-1F-28-8D-3C
Device 4F:56:CA:87:45:04 4F-56-CA-87-45-04
[NEW] Device A8:51:AB:24:B7:F6 A8-51-AB-24-B7-F6
hci0 type 7 discovering off
[bluetooth]# remove 17:05:16:03:35:E1
[DEL] Device 17:05:16:03:35:E1 MOCUTE-032S_A03-35E1
Device has been removed
hci0 type 7 discovering on
[CHG] Device 4F:56:CA:87:45:04 RSSI: 0xffffffab (-85)
hci0 type 7 discovering off
[bluetooth]# pair 17:05:16:03:35:E1
Device 17:05:16:03:35:E1 not available
hci0 type 7 discovering on
[CHG] Device 2C:90:1F:28:8D:3C RSSI: 0xffffffaa (-86)
[CHG] Device A8:51:AB:24:B7:F6 RSSI: 0xffffffa2 (-94)
[NEW] Device 17:05:16:03:35:E1 MOCUTE-032S_A03-35E1
[bluetooth]# pair 17:05:16:03:35:E1
Attempting to pair with 17:05:16:03:35:E1
hci0 device_flags_changed: 17:05:16:03:35:E1 (BR/EDR)
     supp: 0x00000001  curr: 0x00000000
hci0 type 7 discovering off
hci0 17:05:16:03:35:E1 type BR/EDR connected eir_len 27
[CHG] Device 17:05:16:03:35:E1 Connected: yes
hci0 new_link_key 17:05:16:03:35:E1 type 0x04 pin_len 0 store_hint 1
[CHG] Device 17:05:16:03:35:E1 Bonded: yes
hci0 device_flags_changed: 17:05:16:03:35:E1 (BR/EDR)
     supp: 0x00000001  curr: 0x00000001
[CHG] Device 17:05:16:03:35:E1 UUIDs: 00000000-deca-fade-deca-deafdecacaff
[CHG] Device 17:05:16:03:35:E1 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 17:05:16:03:35:E1 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 17:05:16:03:35:E1 ServicesResolved: yes
[CHG] Device 17:05:16:03:35:E1 Paired: yes
Pairing successful
[CHG] Device 17:05:16:03:35:E1 WakeAllowed: yes
hci0 17:05:16:03:35:E1 type BR/EDR disconnected with reason 2
[CHG] Device 17:05:16:03:35:E1 ServicesResolved: no
[CHG] Device 17:05:16:03:35:E1 Connected: no
hci0 type 7 discovering on
[CHG] Device A8:51:AB:24:B7:F6 RSSI: 0xffffffab (-85)
[CHG] Device 2C:90:1F:28:8D:3C RSSI: 0xffffffa1 (-95)
[bluetooth]# connect 17:05:16:03:35:E1
Attempting to connect to 17:05:16:03:35:E1
[... 62 lines cut ...]
= bluetoothd: profiles/input/device.c:control_connect_cb() connect to 17:05:16:03:35:E1: Invalid exchange (52)                                                                  68.768268
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                      #156 [hci0] 68.815665
        Handle: 12 Address: 17:05:16:03:35:E1 (OUI 17-05-16)
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                           #157 [hci0] 68.816209
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                      #158 [hci0] 68.915947
        Status: Success (0x00)
        Handle: 12 Address: 17:05:16:03:35:E1 (OUI 17-05-16)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                               {0x0002} [hci0] 68.915982
        BR/EDR Address: 17:05:16:03:35:E1 (OUI 17-05-16)
        Reason: Connection terminated due to authentication failure (0x04)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                               {0x0001} [hci0] 68.915982
        BR/EDR Address: 17:05:16:03:35:E1 (OUI 17-05-16)
        Reason: Connection terminated due to authentication failure (0x04)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                              #159 [hci0] 70.775679
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
```

I hope I got all the information pinned down here so that anyone can have a look at it. If there is anything missing please let me know. I would love to get this running again. 

Dominik

DrKlipper, 2023-09-18:
Just a side note ... Installed Retropie on the Zero2w. 
It uses bluez 5.50 with Kernel 5.10.103-v7+ (Debian Buster). 
It works without any issues. No key is required at all. 
So it must be a problem of a newer bluez version or the new version 6 Kernel.

jdeut, 2023-09-23:
I had the same authentication-failed issue with a PS5  DualSense Controller.

Trusting the device doesn't work immediately. However after rebooting my system my controller connected immediately and everything works fine since then.

DrKlipper, 2023-09-24:
Ok I found a strange workaround. But to be honest, I don't know if this is a real workaround or "works as expected". 

Anyway ... Here is what I can do: 
* Go to the Retropie repo in the bluetooth folder: https://github.com/RetroPie/RetroPie-Setup/tree/master/scriptmodules/supplementary/bluetooth
* Download bluez-simple-agent and save it as bluez-simple-agent.py 
* Download bluezutils.py from the same folder 
* put the two files onto your Pi 
* Start `sudo bluetoothctl` and run `scan on`
* Use a second Terminal and start `python bluez-simple-agent.py`
* Pair, Trust, Connect the gamepad and you are done 

In some way this is how it works in Retropie, too. 

There is still the problem that it won´t reconnect after a reboot. But ok I can get around that problem as described here: 
https://unix.stackexchange.com/questions/334386/how-to-set-up-automatic-connection-of-bluetooth-headset

**Can anyone tell me why this solution is working?** I´m a little bit confused why the bluez stack can handle a simple gamepad not by it´s own.

CristianArean, 2023-10-17:
Yes, it seems to be a problem of newer bluez versions. I did not have this problem in ubuntu 23.04, but I have it in fedora 38 and ubuntu 23.10

DrKlipper, 2023-10-18:
Maybe this issue should get a bug tag assigned ?

Vudentz, 2023-10-24:
Isn't this a duplicated of #614?

RodoMa92, 2023-11-17:
Yeah, it seems that the same bug affects me now. Seems that some regression has hit pairing since the latest 5.70 release.

Vudentz, 2023-11-17:
Perhaps this is due to mgmt submenu registing callbacks and thus interfering with pairing, that was fixed by https://github.com/bluez/bluez/commit/9741e47aebcdcff343dd2ca7dadc3ec41b66744f

geofire, 2024-07-05:
I've come across this issue as well, though with an Xbox Wireless Controller with slightly different symptoms.

- Xbox Wireless Controller
    - Model 1914
    - Firmware version 5.21.3241.0 (latest)
- BlueZ 5.66
- RetroPie 4.8.8 64-bit (commit 03b7224) (built from source on a Raspberry Pi 5; commit date 2024-07-03)
- Raspberry Pi 5 (brand new, has never been paired with this Xbox controller before)

Error: `AuthenticationCanceled` immediately after attempting to pair. Seems to be related to issue #552.

Tried the following as suggested by some, but don't resolve the issue:
- `sudo bash -c 'echo 1 > /sys/module/bluetooth/parameters/disable_ertm'` 
    - (apparently Xbox controllers have issues with ERTM being enabled)
- `Privacy=device`

Symptoms:
- When attempting to pair directly with `bluetoothctl`, the following happens:
    - Xbox controller is picked up, though BlueZ rapidly flicks between `Connected: yes` and `Connected: no` states every half-second; coincidentally matching the Xbox symbol 'breathing' light pattern on the controller indicating pairing in progress.
    - Attempting to pair with the controller by ID, results in an immediate `Failed to pair: org.bluez.Error.AuthenticationCanceled` error message.

cmitu, 2024-07-05:
@geofire the ETRM configs are no longer needed with recent kernels. Try installing the `xpadneo` driver (available under RetroPie's `driver` packages) and re-pair. Make sure your Xbox controller's firmware is updated (with the Windows update application from the store).

anonymouz, 2024-09-11:
Just to inform you; it look like it works on x86 but fails on ARM. If more people see the same, then focus debugging on bluez+ARM compatibility problems. Maybe some bit manipulation differences that lead to problems on ARM!?

DrKlipper, 2024-09-11:
Ok my tests where all based on ARM systems (mainly Raspberry Pi). 
I can do a X86 test and report back.

gelse, 2024-09-27:
I can confirm that problem on Arch Linux on an X86/64 machine with bluez 5.78-1
Just wanted to drop that it most propably is not only related to ARM.

chewi, 2024-11-05:
Confirmed on Gentoo with a Quartz64. Bluez 5.78 doesn't work but 5.68 does. Gentoo is useful for making this kind of direct comparison. :smile:

chewi, 2024-11-08:
I've cracked it, at least in my case. I bisected the issue to between 5.72 and 5.73 with commit 9698870015b0972d3d93f30a6d9de1ed029ba87c. Such a simple commit! I could just change `UserspaceHID` to `false` in input.conf without even having to patch anything. And sure enough, that fixed it! 5.79 now works.

I had previously noticed that Gentoo had warned me that I hadn't enabled `CONFIG_UHID` in the kernel. I'd looked at the description and judged that it was for more obscure use cases, especially given the **N** suggestion.

> Say Y here if you want to provide HID I/O Drivers from user-space. This allows to write I/O drivers in user-space and feed the data from the device into the kernel. The kernel parses the HID reports, loads the corresponding HID Device Driver or provides input devices on top of your user-space device.
> 
> This driver cannot be used to parse HID-reports in user-space and write special HID-drivers. You should use hidraw for that. Instead, this driver allows to write the transport-layer driver in user-space like USB-HID and Bluetooth-HID do in kernel-space.
> 
> If unsure, say N.

Turns out I do have this enabled on my x86_64 system, so that would explain the difference. I'll try enabling it for ARM soon.

As for the rest of you, perhaps you're also lacking this kernel option. I also currently set `ClassicBondedOnly=false` in my input.conf because the script I use to connect my controllers doesn't work without it. I've been meaning to adapt the script, but perhaps this is also necessary for some controllers?

chewi, 2024-11-10:
I can confirm that enabling `CONFIG_UHID` in the kernel was enough to make it work with `UserspaceHID=true`.

bionade24, 2025-05-28:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days.

https://drewdevault.com/2021/10/26/stalebot.html

Vudentz, 2025-05-28:
> > This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days.
> 
> https://drewdevault.com/2021/10/26/stalebot.html

I would have to disagree, users tend to forget about the issues they created, so this is very much intentional to get feedback if they are still experiencing the problem, in fact your very response sort of enforces that tagging with stale does raise the user attention to it.

bionade24, 2025-05-28:
> I would have to disagree, users tend to forget about the issues they created, so this is very much intentional to get feedback if they are still experiencing the problem, in fact your very response sort of enforces that tagging with stale does raise the user attention to it.

As long as the code handling the behaviour wasn't changed severely, the bug either still exists without further responses from the user or the problem was never a bug in this software.  With many people in this thread reporting very similar issues, it's highly likely there is a bug in the software stack that they share, even if it hasn't been reproduced by a developer yet. A project may be understaffed and there really aren't enough maintainers to handle all bugreport related things manually. The stale bot  as a reminder to the bug is imho a horrible way of doing this. 1st, it treats "I probably haven't read any documentation and need help" with only  one comment the same like this issue where the user provided context to work with & other users jumping on the bug thread. Especially with LLMs stuff I believe there could easily be some better bot, which e.g. triggers after a new major release and asks if the issue still can be reproduced with a newer kernel and version x.x. And only then marks such an issues as stale, while closing "pls need help" much earlier.

> are you guys still experiencing this problem 

I still experience the adjacent problem I have and haven't contributed to the thread as I never had anything that I consider relevant, but was interested in the workarounds/fixes for this.  If it didn't affect me anymore, I'd just unsubscribed the thread & not responded to the stale bot thing.

sylv256, 2025-07-25:
I'm experiencing this problem. My Xbox Wireless Controller will not connect no matter what I do, so I guess I just have to use Windows.

ayan4m1, 2025-09-14:
Having this issue with the Zero 2 W, version `Linux retropie 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l` and bluez version 5.50.

cmitu, 2025-09-14:
@ayan4m1 I don't think it's not the same issue - see the 2nd comment of the original poster which had no issue with the older `bluez` version. In your case you may try to install the `xpadneo` driver first.

sylv256, 2025-10-24:
> I'm experiencing this problem. My Xbox Wireless Controller will not connect no matter what I do, so I guess I just have to use Windows.

No idea what I did, but it fixed itself after making the computer forget and reintroducing the controller.

Amaderis, 2025-12-12:
Please help. I have exactly the same issue. It disconnects and reconnects randomly.
I noticed the following in the logs:

```
[SIGNAL] LE.Disconnected - org.bluez.Reason.Unknown, Unspecified
[SIGNAL] Disconnected - org.bluez.Reason.Unknown, Unspecified
```

```
> HCI Event: Disconnect Complete (0x05) plen 4                      #106 [hci0] 216.523611
        Status: Success (0x00)
        Handle: 16
        Reason: LMP Response Timeout / LL Response Timeout (0x22)
@ MGMT Event: Device Disconnected (0x000c) plen 8               {0x0001} [hci0] 216.523620
        LE Address: AC:8E:BD:57:DB:68 (Microsoft Corporation)
        Reason: Unspecified (0x00)

```