bluez/bluez #606: Bluetoothctl not creating an event*

jesusramondovale, 2023-09-18:
I'm using a DEBIX-Board Model-A with a DEBIX-modified Ubuntu22 image installed. I'm trying to connect via Bluetooth an Xbox Series X controller to it (using bluetoothctl on terminal) 

So I started the Bluetooth service, and successfully connected my controller using bluetoothctl as I can see on bluetoothctl console prompt, and also in the Bluetooth monitor (btmon), when I touch my controller I can see all the messages successfully arriving on the board. 

The problem is that no event is created on /dev/input/. 

Does anyone know why is this happening? Must I be creating the event myself? 

EDIT: this is the bluetoothctl info of my Xbox controller:

```
[Xbox Wireless Controller]# info 0C:35:26:49:44:6C
Device 0C:35:26:49:44:6C (public)
        Name: Xbox Wireless Controller
        Alias: Xbox Wireless Controller
        Appearance: 0x03c4
        Icon: input-gaming
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000001-5f60-4c4f-9c83-a7953298d40d)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v045Ep0B13d0517
        ManufacturerData Key: 0x0006
        ManufacturerData Value:
  03 00 80                                         ...
        Battery Percentage: 0x32 (50)

```

Thank you!

Vudentz, 2023-09-18:
@jesusramondovale what version of BlueZ? Note that it seems 5.69 has some problem with input devices: https://github.com/bluez/bluez/issues/594

jesusramondovale, 2023-09-19:
I'm using BlueZ 5.66. Thanks for answering

macmpi, 2023-09-21:
is `uinput` driver loaded on your system? (check `modinfo uinput` or `lsmod | grep uinput`)

closed 2025-08-06 by github-actions[bot]