bluez/bluez #607: ccc_perm settings in gatt-database::parse_chrc_flags missing BT_ATT_PERM_WRITE

klillas, 2023-09-20:
https://github.com/bluez/bluez/blame/7f788a2c5162985eaddc8bb9377b3f1c91f2e041/src/gatt-database.c#L1780-L1797

I'm not a BlueZ expert, hopefully I have understood this correctly...

The new [CCC permission settings interface](https://github.com/bluez/bluez/blob/master/src/gatt-database.c#L1781C20-L1781C20) over dbus is missing the BT_ATT_PERM_WRITE permission.

As far as I can tell in my testing using a characteristic with the encrypt-authenticated-indicate property, [gatt-server::check_permissions](https://github.com/bluez/bluez/blob/master/src/shared/gatt-server.c#L424C12-L424C12) will always fail a write attempt to the CCC descriptor because it expects the BT_ATT_PERM_WRITE to be set, even if a more specialized write permission exists. 

Also other write permission settings in the same interface, like encrypt-authenticated-write, always sets the BT_ATT_PERM_WRITE together with a more specialized write permission flag.

I assume the solution is to add BT_ATT_PERM_WRITE to the ccc_perm as an additional flag, perhaps also BT_ATT_PERM_READ so you can read out the CCC descriptor?

Vudentz, 2023-09-20:
@klillas wasn't that fixed by fbf17ac497bb2e0077c4cfa22583439e157693fe?

klillas, 2023-09-20:
Ah yes, sorry I was running version 5.65. The fix you linked will surely take care of it. Thanks for pointing it out 😄