bluez/bluez #608: Fails to finish scan, wireplumber+Blueman

Kyllingene, 2023-09-21:
I recently reinstalled Arch (Wayland+sway), and opted for pipewire over pulseaudio. Unfortunately, my bluetooth configuration hasn't been working since. Whenever I scan for devices, I find none; the scan also fails to terminate in Blueman, giving the only concrete errors I can find.

When I run `blueman-manager` from the command line, it prints this:

```
blueman-manager 21.10.58 ERROR    Base:85 callback  : Unhandled error for org.bluez.Adapter1.StopDiscovery
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/bluez/Base.py", line 78, in callback
    value = proxy.call_finish(result).unpack()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress (36)
```

In `journalctl -b | grep -i bluetooth`, I found this line repeated many times:

```
Sep 20 21:09:04 kyllingene dbus-daemon[515]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.18" (uid=1000 pid=659 comm="/usr/bin/wireplumber") interface="(unset)" member="(unset)" error name="org.bluez.Profile1.Error.NotImplemented" requested_reply="0" destination=":1.353" (uid=0 pid=112091 comm="/usr/lib/bluetooth/bluetoothd")
```

I'm quite uneducated about the relation between client and daemon in Linux bluetooth, but that looks like a bluez issue. If I'm mistaken, I'd be glad to take this over to the wireplumber repo.

When using `bluetoothctl`, nothing errors, but I don't get any devices.

I have fully updated (and restarted) my system, and reinstalled all related packages. I also tried with `pulseaudio[-bluetooth]`, but no dice.

System info:
	- Arch Linux (kernel 6.5.3-zen1-1-zen)
	- bluez[-utils] 5.69-1
	- wireplumber 0.4.14-1
	- blueman 2.3.5-2

Kyllingene, 2023-09-28:
[Here you go.](https://gist.github.com/Kyllingene/c9b6f14bc1a6a78b19fc8078238fc341) By the way, is it supposed to be so long for so short a capture?

Kyllingene, 2023-10-03:
Just tried `bluetoothctl` again, noticed that `scan off` fails with the same `InProgress` error. Also, at some point, I started getting this when I open it:
```
hci0 new_settings: powered bondable ssp br/edr le secure-conn
Agent registered
[CHG] Controller C0:E4:34:9D:20:FE Pairable: yes
```
Perhaps that can help?

Verymadlibrarian, 2024-10-21:
Hi,just wanted to add to this issue, i'm facing the same issue but changing ControllerMode didn't resolve it.
On my end, sometimes after a reboot, the scanning starts properly in the bluetoothctl utility but no bt devices are shown and if I retry to start the scan again I get : 

`[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Failed to start discovery: org.bluez.Error.InProgress`

and journalctl -xeu gives : 

`oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support bap plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support bass plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support mcp plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support vcp plugin
oct. 21 00:01:57 y13 bluetoothd[433]: profiles/audio/micp.c:micp_in
it() D-Bus experimental not enabled
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support micp plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support ccp plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support csip plugin
oct. 21 00:01:57 y13 bluetoothd[433]: src/plugin.c:init_plugin() Sy
stem does not support asha plugin
oct. 21 00:01:57 y13 bluetoothd[433]: Bluetooth management interface 1.23 initialized
oct. 21 00:01:57 y13 bluetoothd[433]: Battery Provider Manager created
oct. 21 00:01:57 y13 bluetoothd[433]: Failed to set mode: Not Supported (0x0c)`

If I open blueberry, it just says that it is searching for devices, but doesn't not show any