bluez/bluez #612: Support for BAP audio configurations 6(i), 8(i), 9(i), 11(i)

pv, 2023-09-30:
Feature request: It seems BAP audio configurations 6(i), 8(i), 9(i), 11(i) are not supported currently in BlueZ.

These are the configurations where two sink ASEs are configured, with separate CIS for each.

Namely:
- `src/shared/bap.c:bap_parse_pacs` merges PACs with same codec into one PAC (this is OK)
- `profiles/audio/bap.c:bap_ready` creates one MediaEndpoint for each (remote PAC, local PAC) pair
- Each device has then at most one sink MediaEndpoint.
- `profiles/audio/bap.c:bap_config` creates one bt_bap_stream for each MediaEndpoint.
- `src/shared/bap.c:bt_bap_stream_new` associates each stream with a single ASE.
- At most one sink ASE is configured, so configurations with multiple sink ASEs are not set up.

Possibly, each BAP spec audio configuration that the remote seems to support could be exposed as MediaEndpoint.

NerdyProjects, 2023-09-30:
Background: LE audio support for Earfun air pro 3. Likely,all Qualcomm QCC30XX le audio earbuds (using "True wireless mirroring") work that way. https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3531

Vudentz, 2023-10-02:
@NerdyProjects are these the model: https://www.amazon.com/EarFun-Air-Pro-Cancelling-Multipoint/dp/B0BNNMYSGQ?th=1

Vudentz, 2023-10-02:
> It seems BAP audio configurations 6(i), 8(i), 9(i), 11(i) are not supported currently in BlueZ.
> 
> These are the configurations where two sink ASEs are configured (or two source ASEs) for the same device, with separate CIS for each.
> 
> 6(i) and 9(i) are Mandatory for client (BAP Table 4.2).
> 
> Namely:
> 
>     1. `src/shared/bap.c:bap_parse_pacs` merges PACs with same codec into one PAC (this is OK)
> 
>     2. `profiles/audio/bap.c:bap_ready` creates one MediaEndpoint for each (remote PAC, local PAC) pair
> 
>     3. Each device has then at most one sink MediaEndpoint.
> 
>     4. `profiles/audio/bap.c:bap_config` creates one bt_bap_stream for each MediaEndpoint.
> 
>     5. `src/shared/bap.c:bt_bap_stream_new` associates each stream with a single ASE.
> 
>     6. At most one sink ASE is configured, so configurations with multiple sink ASEs are not set up.
> 
> 
> Possibly, we should change step 2. so that instead each possible BAP spec audio configuration that the remote seems to support could be exposed as a separate MediaEndpoint.
> 
> A complication with this is that the PACs indicate configuration of a single ASE, and the Codec_Specific_Capabilities for each ASE is expected (cf. eg. BAP Table 4.8) to allocate it to a different channel, so the SelectProperties API would need to be extended so that configuration for multiple ASEs can be returned. E.g. add integer input parameter `"NumASEs`", and have the callback return an array of configurations.
> 
> An alternative could be to extend the DBus API by exposing the remote ASEs and PACs, and leaving it to the sound server to figure it out.

Perhaps we could use the same logic of 0 length LTV to separate stream configurations, that way we don't need to change the API and it works in the same way as to how we merge the capabilities to merge configurations. Obviously we need the changes to the bap plugin to understand when there are multiple configurations it needs to setup multiple streams, anyway this was in my TODO list so I can look into it once I get a hold of a device that does requires such setup.

NerdyProjects, 2023-10-03:
> @NerdyProjects are these the model: https://www.amazon.com/EarFun-Air-Pro-Cancelling-Multipoint/dp/B0BNNMYSGQ?th=1

Yep, that's the model. They should be actually quite popular, as they provide very good value for the money (with some downsides e.g. quite high background noise, not perfect sound quality in general). Do you have funding/money available to acquire one or would it be useful if I gift one device?

Vudentz, 2023-10-03:
> > @NerdyProjects are these the model: [amazon.com/EarFun-Air-Pro-Cancelling-Multipoint/dp/B0BNNMYSGQ?th=1](https://www.amazon.com/EarFun-Air-Pro-Cancelling-Multipoint/dp/B0BNNMYSGQ?th=1)
> 
> Yep, that's the model. They should be actually quite popular, as they provide very good value for the money (with some downsides e.g. quite high background noise, not perfect sound quality in general). Do you have funding/money available to acquire one or would it be useful if I gift one device?
> 
> Edit: If you get hold of one, they will not support LE Audio with firmware version >0.3.9 (or something). But the support can email you an LE enabled firmware (after telling them which FW you are running) that needs to be flashed via an APK they send you as well... I did that with 1.0.7 and it worked fine.

Can you attach the firmware here, or is that under NDA or something? I will try to get a hold of one.

NerdyProjects, 2023-10-03:
No, I did not sign any NDA or something, so here you are:
[earfun air pro 3 v1.0.7le.zip](https://github.com/bluez/bluez/files/12796747/earfun.air.pro.3.v1.0.7le.zip)
This also includes the APK to flash the bin file, so you need an android phone.
I also found https://cweiske.de/tagebuch/bluetooth-gaia.htm this for the GAIA stuff, but it doesn't look like there is a linux application available to do the flashing...

Vudentz, 2023-10-13:
Weird, it doesn't seem to support pairing for some reason:

```
> ACL Data RX: Handle 2048 flags 0x02 dlen 6                                                                       #195 [hci0] 14:39:03.418094
      SMP: Pairing Failed (0x05) len 1
        Reason: Pairing not supported (0x05)
```

After flashing, do I have to reset it or something?

Vudentz, 2023-10-13:
Ok, got it working, it was connecting to my phone:

```
Endpoint /org/bluez/hci0/dev_70_5A_6F_63_B6_41/pac_sink0
	UUID: 00002bc9-0000-1000-8000-00805f9b34fb
	Codec: 0x06 (6)
	Capabilities:
  03 01 80 00 02 02 13 02 03 01 05 04 4b 00 64 00  ............K.d.
  02 05 02 00 03 01 34 00 02 02 23 02 03 01 05 04  ......4...#.....
  28 00 50 00 02 05 02 00 03 01 b4 00 02 02 23 02  (.P...........#.
  03 01 05 04 1e 00 9b 00 02 05 02                 ...........     
	Device: /org/bluez/hci0/dev_70_5A_6F_63_B6_41
	Locations: 0x00000003 (3)
	SupportedContext: 0x0fff (4095)
	Context: 0x000f (15)
	QoS.Framing: 0x00 (0)
	QoS.PHY: 0x02 (2)
	QoS.Retransmissions: 0x0f (15)
	QoS.MaximumLatency: 0x0fa0 (4000)
	QoS.MimimumDelay: 0x000061a8 (25000)
	QoS.MaximumDelay: 0x00009c40 (40000)
	QoS.PreferredMimimumDelay: 0x00000000 (0)
	QoS.PreferredMaximumDelay: 0x00000000 (0)
```

It does connect one CIS, at least with bluetoothctl:

```
Transport /org/bluez/hci0/dev_70_5A_6F_63_B6_41/pac_sink0/fd3
	UUID: 00002bcb-0000-1000-8000-00805f9b34fb
	Codec: 0x06 (6)
	Configuration:
  02 01 03 02 02 01 03 04 28 00                    ........(.      
	Device: /org/bluez/hci0/dev_70_5A_6F_63_B6_41
	State: idle
	Endpoint: /org/bluez/hci0/dev_70_5A_6F_63_B6_41/pac_sink0
	QoS.CIG: 0x00 (0)
	QoS.CIS: 0x00 (0)
	QoS.Framing: 0x00 (0)
	QoS.PresentationDelay: 0x00009c40 (40000)
	QoS.Interval: 0x00002710 (10000)
	QoS.Latency: 0x000a (10)
	QoS.SDU: 0x0028 (40)
	QoS.PHY: 0x02 (2)
	QoS.Retransmissions: 0x02 (2)
	Location: 0x00000003 (3)
	Links: /org/bluez/hci0/dev_70_5A_6F_63_B6_41/pac_source0/fd2
```

But I guess we want to have stereo support for it as well, I will work on that using bluetoothctl as a reference.

closed 2023-12-15 by BluezTestBot

pv, 2024-01-07:
Pipewire side: https://gitlab.freedesktop.org/pipewire/pipewire/-/merge_requests/1833
Stereo works on Earfun Air Pro 3 LE Audio firmware.