bluez/bluez #614: PS5 controller will no longer pair with bluetoothd

Absolutely-Free, 2023-10-01:
I recently had some time to game but found that my PS5 controller will no longer pair with either of my computers that I game on.

I originally noticed this issue for the first time yesterday, and while troubleshooting I found issue #594 which seems to describe my issue perfectly. If I try to pair the controller from my previously working paired devices using KDE's GUI, I get error messages and the controller never connects to the computer. If I delete the device from the GUI and try to pair the devices for the first time again, the controller will show up when it's in pairing mode while going through the setup wizard but will fail to pair. If I check `systemctl status bluetooth` I see the same errors repeating every time I try to pair the devices (personal info modified):

```shell
Sep 30 20:54:43 HOSTNAME bluetoothd[885]: src/service.c:service_probe() input-hid profile probe failed for XX:XX:XX:XX:XX:XX
Sep 30 20:54:49 HOSTNAME bluetoothd[885]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
Sep 30 20:54:49 HOSTNAME bluetoothd[885]: profiles/input/server.c:confirm_event_cb() Refusing connection from XX:XX:XX:XX:XX:XX: unknown device
``` 

`pacman -Qs bluez` shows:
```shell
local/bluez 5.70-1
    Daemons for the bluetooth protocol stack
local/bluez-libs 5.70-1
    Deprecated libraries for the bluetooth protocol stack
local/bluez-qt5 5.110.0-1 (kf5)
    Qt wrapper for Bluez 5 DBus API
local/bluez-utils 5.70-1
    Development and debugging utilities for the bluetooth protocol stack
``` 

Issue #594 was closed with a comment implying that the issue was fixed in version 5.70, which was still in the Arch testing repos yesterday. Today the new version hit the official repos so I updated, rebooted, and tried again, but I am still getting the exact same errors.

I can reproduce this exact same issue on two different computers, both of which are up-to-date Arch Linux with KDE, both have different generations of intel wifi/bluetooth hardware, and both of which worked correctly with this controller when I last had time to game (~3 months ago). I also just updated the firmware for the controller using [Sony's updater](https://controller.dl.playstation.net/controller/lang/en/fwupdater.html) with a Windows computer, and the problem is still persisting.  The controller works correctly over USB in Arch, and the bluetooth worked correctly with my Windows machine, so I believe the issue is definitely with either bluez or KDE's software that depends on it. Since issue #594 seems to describe my issue exactly, I am reporting the bug here first.

Not sure what other info may be relevant or useful, but I am happy to provide any other info requested.

RealVishy, 2023-10-02:
Hey, I've encountered the same issue with my PS5 Controller failing to pair on version 5.70 of Fedora 38 KDE Spin, both on my laptop and desktop. 

It seems like this commit was intended to address the problem: [GitHub Commit](https://github.com/bluez/bluez/commit/670f0d0f618d3f122e6d95c34de53f59a343a15f?diff=unified), but I'm still experiencing the same issue as I did with version 5.69.

thyTwilightGoth, 2023-10-02:
I'm am also still having this issue, and my logs look identical to the OP's posted logs, strangely this seem to only effect Dualsense controllers, because my 8bitdo Pro 2 controller works just fine.

Vudentz, 2023-10-03:
Have you guys tried pairing with bluetoothctl? Do you guys can collect HCI traces with btmon.

Vudentz, 2023-10-03:
I wonder if this has something to do with:

https://patchwork.kernel.org/project/bluetooth/patch/20231003211442.3348427-1-luiz.dentz@gmail.com/

Absolutely-Free, 2023-10-04:
> Have you guys tried pairing with bluetoothctl?

I just tried running the pairing via bluetoothctl, see the relevant
[absolutelyFreeBtmonLogs.log](https://github.com/bluez/bluez/files/12798844/absolutelyFreeBtmonLogs.log)
[absolutelyFreeBtmonLogs.txt](https://github.com/bluez/bluez/files/12798847/absolutelyFreeBtmonLogs.txt)
 output below. I sanitized the logs following the pattern of "YY:YY:YY:YY:YY:YY" = the PC's bluetooth hardware controller, "XX:XX:XX:XX:XX:XX" = the Sony PS5 Dualsense Controller.

```shell
[bluetooth]# scan on
Discovery started
[CHG] Controller YY:YY:YY:YY:YY:YY Discovering: yes
[NEW] Device XX:XX:XX:XX:XX:XX DualSense Wireless Controller
[bluetooth]# connect XX:XX:XX:XX:XX:XX
Attempting to connect to XX:XX:XX:XX:XX:XX
[CHG] Device XX:XX:XX:XX:XX:XX Connected: yes
[CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: no
[CHG] Device XX:XX:XX:XX:XX:XX Connected: no
[bluetooth]# scan off
[CHG] Controller YY:YY:YY:YY:YY:YY Discovering: no
Discovery stopped

``` 

> Do you guys can collect HCI traces with btmon.

I repeated the process above with btmon running as root. I apologize for the noise in these logs, I live in an apartment building so there are many devices nearby and I didn't want to accidentally delete important info. I sanitized them following the same pattern I listed above.I attached them to this comment.

abnotus, 2023-10-04:
Having the same issue as well, can also confirm that I have no issue after downgrading back to 5.68

nfp0, 2023-10-07:
I am also having the same issue with 2 DualShock 4 controllers tested on 2 different Arch computers on bluez 5.70
Output is the same as other users posted above.
Downgrading to 5.68 fixes the issue.

Vudentz, 2023-10-09:
Any chance for one of you guys having this problem to bisect the patch which introduces it?

solisinvictum, 2023-10-09:
I have the same Problem. Version 5.70 and 5.69 dont work. 

Only Version 5.68 works fine.

Here are maaaaaaaany logs. Automatic with Kde. Over Terminal with bluetoothctl. Journal etc. Arch Linux.

[message4.txt](https://github.com/bluez/bluez/files/12850187/message4.txt)
[message3.txt](https://github.com/bluez/bluez/files/12850189/message3.txt)
[message1.txt](https://github.com/bluez/bluez/files/12850190/message1.txt)
[message.txt](https://github.com/bluez/bluez/files/12850191/message.txt)

Vudentz, 2023-10-09:
Looks like it failing because it cannot find any SDP record, now it really beats me why that wouldn't be the case for 5.68 but perhaps something has changed in the package in the meantime? Perhaps the cache location is wrong or something, what distro are you guys using and what is the content of bluetooth.service?

Vudentz, 2023-10-09:
I think I know what is going on, it is probably due to the following change 67a26abe53bf33422c17a3f63866e76864b92bc2, input-hid plugin probably requires input-hid to be resolved to be probed contrary to only when it is connected/connecting, so it fails to probe, I will try to come up with a fix for it in the upcoming days.

solisinvictum, 2023-10-09:
Arch Linux.

The Content of bluetooth.service is perhaps the same like for 5.68.

 >> now it really beats me why that wouldn't be the case for 5.68 but 
perhaps something has changed in the package in the meantime?

yeah, indeed. And i can say exactly what: Everything that got added and 
edited since Version 5.69 :D jk

Vudentz, 2023-10-10:
Could one if you guys please try with the above patch?

solisinvictum, 2023-10-11:
If you tell me how, i would try it!

nfp0, 2023-10-11:
I'm gonna give it a try in a bit.
@solisinvictum I'm gonna use the bluez-git AUR package and replace the source.

nfp0, 2023-10-11:
> Could one if you guys please try with the above patch?

@Vudentz I can confirm that fixes it! :slightly_smiling_face:

Vudentz, 2023-10-11:
@Absolutely-Free can you please check as well? @nfp0 perhaps you can add the package you have created so others people running the same distro can test with their peripherals.

nfp0, 2023-10-11:
Sure!
All I did was grab this from the AUR:
https://aur.archlinux.org/packages/bluez-git
and replace the source on the PKGBUILD file (line 34) with:
`"$pkgname::git+https://github.com/BluezTestBot/bluez.git#branch=791911"`

solisinvictum, 2023-10-11:
Thanks.

I can confirm that this fix it for dualsense (ps5) too.

Absolutely-Free, 2023-10-11:
> Sure! All I did was grab this from the AUR: https://aur.archlinux.org/packages/bluez-git and replace the source on the PKGBUILD file (line 34) with: `"$pkgname::git+https://github.com/BluezTestBot/bluez.git#branch=791911"`

Thanks for this @nfp0 !

> @Absolutely-Free can you please check as well? @nfp0 perhaps you can add the package you have created so others people running the same distro can test with their peripherals.

I can also confirm that this fix works! Thank you for your work on this @Vudentz , we all appreciate it!

closed 2023-10-11 by BluezTestBot

gabriele2000, 2023-10-20:
> Thanks for this @nfp0 !
>
>
> I can also confirm that this fix works! Thank you for your work on this @Vudentz , we all appreciate it!

Yeah, it's for arch... what about ubuntu-based distros?

NuVanDibe, 2023-10-25:
Not sure why this is closed. It was never fixed.

RealVishy, 2023-10-25:
> Not sure why this is closed. It was never fixed.

It was fixed in a recent patch but there hasn't been a new release, wait for a new release or downgrade to 5.68

borancar, 2023-12-26:
Still seems to be an issue on 5.71.

DrKlipper, 2024-01-27:
Is there any fix in the 5.72 now? Or does this pairing problem still exist?

gabriele2000, 2024-01-27:
> Is there any fix in the 5.72 now? Or does this pairing problem still exist?

it has been fixed a week later or something like that.

DrKlipper, 2024-01-28:
Does not work for me ... 
```
[bluetooth]# [NEW] Device 17:05:16:03:35:E1 MOCUTE-032S_A03-35E1
[bluetooth]# hci0 type 7 discovering off
pair 17:05:16:03:35:E1
Attempting to pair with 17:05:16:03:35:E1
[bluetooth]# hci0 device_flags_changed: 17:05:16:03:35:E1 (BR/EDR)
[bluetooth]#      supp: 0x00000000  curr: 0x00000000
[bluetooth]# hci0 17:05:16:03:35:E1 type BR/EDR connected eir_len 27
[CHG] Device 17:05:16:03:35:E1 Connected: yes
[MOCUTE-032S_A03-35E1]# hci0 new_link_key 17:05:16:03:35:E1 type 0x04 pin_len 0 store_hint 1
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 Bonded: yes
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 UUIDs: 00000000-deca-fade-deca-deafdecacaff
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 ServicesResolved: yes
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 Paired: yes
[MOCUTE-032S_A03-35E1]# Pairing successful
[MOCUTE-032S_A03-35E1]# hci0 17:05:16:03:35:E1 type BR/EDR disconnected with reason 2
[MOCUTE-032S_A03-35E1]# [CHG] Device 17:05:16:03:35:E1 ServicesResolved: no
[CHG] Device 17:05:16:03:35:E1 Connected: no
[bluetooth]# hci0 type 7 discovering on
[DEL] Device 50:EA:4A:4F:99:66 50-EA-4A-4F-99-66
hci0 type 7 discovering off
connect 17:05:16:03:35:E1 :16:03:35:E1
Attempting to connect to 17:05:16:03:35:E1
[bluetooth]# hci0 17:05:16:03:35:E1 type BR/EDR connected eir_len 27
[CHG] Device 17:05:16:03:35:E1 Connected: yes
[MOCUTE-032S_A03-35E1]# hci0 17:05:16:03:35:E1 auth failed with status 0x05 (Authentication Failed)
[MOCUTE-032S_A03-35E1]# Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[MOCUTE-032S_A03-35E1]# hci0 17:05:16:03:35:E1 type BR/EDR disconnected with reason 2
[CHG] Device 17:05:16:03:35:E1 Connected: no
```
This is a compiled 5.72 on a Raspberry Pi 3. 
The controller gets paired, connected and immediately disconnected.

gabriele2000, 2024-01-28:
> This is a compiled 5.72 on a Raspberry Pi 3. The controller gets paired, connected and immediately disconnected.

Can you try to compile [this](d994c9d45dad68bc060add77642fdc37834cd7c1)?
It's an "almost master" commit, it's of 3 days ago (because master is broken for me right now, so I deem this commit safer).

DrKlipper, 2024-01-29:
Doesn´t look well ... 
```
pi@Pi3Test:~ $ sudo bluetoothctl
[sudo] password for pi:
Waiting to connect to bluetoothd...[bluetooth]# hci0 new_settings: powered bondable ssp br/edr le secure-conn
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller B8:27:EB:4E:F7:C9 Pairable: yes
[bluetooth]# list
Controller B8:27:EB:4E:F7:C9 Pi3Test [default]
[bluetoo17:05:16:03:35:E15:16:03:35:E1
Attempting to connect to 17:05:16:03:35:E1
[bluetooth]# hci0 17:05:16:03:35:E1 type BR/EDR connected eir_len 22
[CHG] Device 17:05:16:03:35:E1 Connected: yes
[MOCUTE-032S_A03-35E1]# hci0 17:05:16:03:35:E1 auth failed with status 0x05 (Authentication Failed)
[MOCUTE-032S_A03-35E1]# Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[MOCUTE-032S_A03-35E1]# hci0 17:05:16:03:35:E1 type BR/EDR disconnected with reason 2
[CHG] Device 17:05:16:03:35:E1 Connected: no
[bluetooth]#
```
But I will do another test this evening. Will start from a clean installation to avoid any conflicts with some precompiled stuff. 

But anyway ... The first try was not successful - even after a reboot. 

BTW: Can I check if I actually run this specified version I compiled? 
`btmon -v` just reports 5.72

gabriele2000, 2024-01-29:
> BTW: Can I check if I actually run this specified version I compiled? `btmon -v` just reports 5.72

I don' think so.
I just remembered: did you try unpairing and pairing the controller again?
For pairing, remember that you have to hold the playstation logo and the "share" button (as I call it: the Select button)

DrKlipper, 2024-01-29:
Is this mentioned fix for PS controller only? 
I tried a small BT game controller - not a PS one. 
And to answer your question ... Yes a removed the device in bluetoothctl and die a new pair.

gabriele2000, 2024-01-29:
> Is this mentioned fix for PS controller only? I tried a small BT game controller - not a PS one. And to answer your question ... Yes a removed the device in bluetoothctl and die a new pair.

I honestly have no clue, I think it cover more than PS5 controllers, but I don't know about generic controllers.

DrKlipper, 2024-01-30:
Ok I can confirm that this compiled commit "d994c9d45dad68bc060add77642fdc37834cd7c1" works with 
* PS4 Controller
* PS5 Controller
But it won´t work with a very simple gamepad controller (https://www.amazon.de/auvisio-Gamepad-Bluetooth-Mini-Akku-Game-Controller-Fernbedienung-black/dp/B0179P8WMG)

I will track that in my initial post https://github.com/bluez/bluez/issues/605

The0neFreeMan, 2024-04-05:
Still not working for me in version 5.73. Currently downgrading all the way back to 5.68. I'm on Arch btw. I see earlier in the thread that it was supposedly fixed, but I've never had it work again with any version upgrade since it broke in 5.69.

lucasmb, 2024-04-19:
I confirm that it's working with bluez 5.75-1 with  `UserspaceHID=true` input.conf on arch btw @The0neFreeMan. Paired the ds5 using KDE bluetooth settings, it threw an error modal but still connected.

TableCoffee, 2024-04-23:
bluez 5.75-1 on arch/kde and still the same here as well. Trying to connect a Dualsense via KDE bluetooth applet results in an error, turning on the controller again it appears to connect, and shows connected in KDE panel but no game sees that it's connected.

JasonLandbridge, 2024-08-13:
The following fixed this for me: https://bbs.archlinux.org/viewtopic.php?pid=2120526#p2120526

TL;DR: Downgrade to version 5.68 with:

```bash
downgrade bluez bluez-libs bluez-utils 
```