bluez/bluez #618: Permission issue when bind L2CAP socket

katyo, 2023-10-04:
I implements Object Transfer Service (OTS) client so I need create L2CAP SEQPACKET socket and connect it to device for handling data transfer.

If I run my application from unprivileged user I have an issue with bind (permission denied) while from root it works as expected.
I tried change socket security level but it cannot help.

If I skips binding my socket works wrong in two aspects: first the readed data preceded by a two byte length which lesser or equals receive MTU, secord I able to read at most MTU bytes of data after that I cannot use same socket more.

Why is bind so matter? And how can I give sufficient privilegies to user to solve this issue?

Vudentz, 2023-10-04:
I think this is the reason:

```
	/* Restrict fixed, SIG assigned PSM values to CAP_NET_BIND_SERVICE */
	if (psm < L2CAP_PSM_LE_DYN_START && !capable(CAP_NET_BIND_SERVICE))
		return -EACCES;
```

What is the PSM you are trying to bind? I thought OTS PSM was freely defined but perhaps Im wrong on this. In case of OBEX we have bluetoothd handling the server socket and then we transfer over the control to the user application when that is authorized:

https://github.com/bluez/bluez/blob/master/doc/profile-api.txt

We may need to do some changes to accommodate LE profiles, but otherwise I think that is probably a better solution in terms of security then letting application bind to a SIG assigned PSM which is subject to qualification, etc.

katyo, 2023-10-05:
I bind socket to address with PSM = 0x0025.

```c
#define PSM_L2CAP_CID_OTS 0x0025

/* ... */

struct sockaddr_l2 bind_addr = {
    .l2_family      = AF_BLUETOOTH,
    .l2_psm         = PSM_L2CAP_CID_OTS,
    .l2_bdaddr      = *BDADDR_ANY,
    .l2_cid         = 0,
    .l2_bdaddr_type = BDADDR_LE_RANDOM,
};
```

katyo, 2023-10-05:
I tried to bind using L2CAP_PSM_LE_DYN_START for PSM and it works fine from unprivileged user for now.

closed 2023-10-06 by katyo