bluez/bluez #619: Expose bluetoothd version number over D-Bus

surban, 2023-10-06:
Currently there is no D-Bus method to query the version number of the running `bluetoothd`.

However, knowing the version number would allow libraries like BlueR to check for compatibility, determine supported features and work around known issues in certain versions of the daemon.

Calling `bluetoothd --version` and parsing the output is not be reliable, because

- the binary might have been replaced on disk without being restarted,
- the binary might not be accessible due to some form of isolation (container, non-standard installation location, etc.),
- spawning a sub-process is impossible due to security restrictions (SELinux, etc.).

Would you be open to accept a PR which exposes the version number over D-Bus?

CosminPerRam, 2024-02-05:
Bump, any blockages to be known for?

FeldrinH, 2025-09-13:
Still relevant as far as I know

CosminPerRam, 2025-12-26:
Yep, still relevant