bluez/bluez #625: [mesh] Scanning for new devices does not work

aozyumenko, 2023-10-11:
It seems that after the commit of 2023-03-31, the mesh was unable to search for new devices. 
"discover-unprovisioned on" command does not show any devices.
I definitely didn't understand the cause, but it looks like the problem is in the code:
    l_queue_isempty(pvt->io->rx_regs)
in functions: restart_scan() and recv_register().
In my case, the queue is always not empty and the scanning process does not start.

aozyumenko, 2023-10-12:
The scan is started either during mesh initialization or when the mesh_io_register_recv_cb() function is called.

During initialization, IO is created ( mesh_io_new() ) that runs the scan ( dev_init()->hci_init()->restart_scan() ) and then adds the IO to the nodes (hci_init() -> io->ready() ). Since the OI is added to nodes after restart_scan(), when the restart_scan() is called, the queue io->rx_regs remains empty and scanning does not start. 

When the mesh_io_register_recv_cb() function is called, receiver added to the io->rx_regs before io->api->reg() was called. Therefore, in the recv_register() function, the condition
    already_scanning = !l_queue_isempty(io->rx_regs);
will always be true.
In the previous version, when the recv_register() and mesh_io_register_recv_cb() functions were combined, this condition was checked **before** adding the receiver to the io->rx_regs queue. The value of this condition could be false and the scan would be triggered. I'm guessing this is the problem.

aozyumenko, 2023-10-12:
A simple patch to illustrate the problem. I used unused arguments to pass the status of the queue to the  io->api->reg()

`diff --git a/mesh/mesh-io-generic.c b/mesh/mesh-io-generic.c
index 00932ade7..bfc708fd3 100644
--- a/mesh/mesh-io-generic.c
+++ b/mesh/mesh-io-generic.c
@@ -781,7 +781,7 @@ static bool recv_register(struct mesh_io *io, const uint8_t *filter,
        bool already_scanning;
        bool active = false;

-       already_scanning = !l_queue_isempty(io->rx_regs);
+       already_scanning = (len != 0) ? true : false;

        /* Look for any AD types requiring Active Scanning */
        if (l_queue_find(io->rx_regs, find_active, NULL))
diff --git a/mesh/mesh-io.c b/mesh/mesh-io.c
index b953bf4cf..f61ce7ceb 100644
--- a/mesh/mesh-io.c
+++ b/mesh/mesh-io.c
@@ -218,6 +218,8 @@ bool mesh_io_register_recv_cb(struct mesh_io *io, const uint8_t *filter,
        l_free(rx_reg);
        l_queue_remove(io->rx_regs, rx_reg);

+       bool already_scanning = !l_queue_isempty(io->rx_regs);
+
        rx_reg = l_malloc(sizeof(struct mesh_io_reg) + len);
        rx_reg->cb = cb;
        rx_reg->len = len;
@@ -227,7 +229,7 @@ bool mesh_io_register_recv_cb(struct mesh_io *io, const uint8_t *filter,
        l_queue_push_head(io->rx_regs, rx_reg);

        if (io && io->api && io->api->reg)
-               return io->api->reg(io, filter, len, cb, user_data);
+               return io->api->reg(io, NULL, (uint8_t)already_scanning, cb, NULL);

        return false;
 }
`

inga-s, 2023-10-16:
You are correct: this bug has been introduced when some code duplication has been removed.
I suggest a slightly different fix (since this is only relevant to mesh-io-generic, so we preferably should confine the fix to that functionality):

--- a/mesh/mesh-io-generic.c
+++ b/mesh/mesh-io-generic.c
@@ -781,7 +781,7 @@ static bool recv_register(struct mesh_io *io, const uint8_t *filter,
        bool already_scanning;
        bool active = false;
 
-       already_scanning = !l_queue_isempty(io->rx_regs);
+       already_scanning = (l_queue_length(io->rx_regs) > 1);
 
        /* Look for any AD types requiring Active Scanning */
        if (l_queue_find(io->rx_regs, find_active, NULL))

Could you please verify that it works for you?

aozyumenko, 2023-10-17:
Yes, it works.
[image]

closed 2023-10-17 by BluezTestBot