bluez/bluez #627: Dual Shock and Dual Sense do not work

LibertyGM, 2023-10-17:
Please help me. Bluetooth used to work fine, but now I can't connect Dual Shock and Dual Sense.

paru -S bluez bluez-utils
sudo systemctl enable --now bluetooth.service

When adding a device writes:
Failed to configure the device

But the device itself appears in the list of added

When trying to connect writes:
br-connection-page-timeout

RealVishy, 2023-10-19:
This was already fixed recently, either follow the instructions in the below thread or revert to bluez 5.68 until a new version releases.

https://github.com/bluez/bluez/issues/614

LibertyGM, 2023-10-19:
> This was already fixed recently, either follow the instructions in the below thread or revert to bluez 5.68 until a new version releases.
> 
> #614

Thanks