bluez/bluez #628: Unable to get on D-Bus

navinagaraj, 2023-10-17:
```
# /usr/lib/bluetooth/bluetoothd -C -d -n
[  103.781400] Bluetooth: Core ver 2.22
[  103.787979] NET: Registered protocol family 31
[  103.799967] Bluetooth: HCI device and connection manager initialized
[  103.806389] Bluetooth: HCI socket layer initialized
[  103.812195] Bluetooth: L2CAP socket layer initialized
[  103.817336] Bluetooth: SCO socket layer initialized
bluetoothd[2640]: Bluetooth daemon 5.70
bluetoothd[2640]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
D-Bus setup failed: Connection ":1.0" is not allowed to own the service "org.bluez" due to security policies in the configuration file
bluetoothd[2640]: src/main.c:main() Unable to get on D-Bus
```

inga-s, 2023-10-17:
Did you try to run it as "sudo /usr/lib/bluetooth/bluetoothd ..."?

navinagaraj, 2023-10-30:
> Did you try to run it as "sudo /usr/lib/bluetooth/bluetoothd ..."?

yes

closed 2023-11-17 by navinagaraj