bluez/bluez #631: Establishing connections to multiple devices at once

jonas2515, 2023-10-23:
It seems that bluetooth adapters behave quite differently when trying to connect to more than a single device at the same time.

On the BCM4387 in an M1 Macbook Pro, only a single "Create Connection" HCI request will go through, and the subsequent ones will fail with `Command Status: Create Connection, Status: Hardware Failure` event as long as the adapter is trying to connect.

Meanwhile on the Marvell 88W8897 in a Surface Pro, all "Create Connection" requests go through (as in they return `Command Status: Create Connection, Status: Success`), but after a while we receive `Connect Complete` HCI event with "Limited Resources" status and a "Connect Failed" MGMT event.

I assume it's kind of expected and due to hardware/firmware differences that the command gets rejected immediately or later. What I'm wondering though is whether BlueZ should have special handling for these cases and try to connect again instead of failing?

jonas2515, 2023-10-24:
> What I'm wondering though is whether BlueZ should have special handling for these cases and try to connect again instead of failing?

I wrote a few fairly hacky patches to give this a try, posted as a PR (https://github.com/bluez/bluez/pull/633) because that's easier than linking to an email.

The approach of "interpreting" fairly broad socket errors as EAGAIN is probably not a good idea, but that seemed like the most straightforward way for a quick prototype.

Vudentz, 2023-10-24:
@jonas2515 have a look at how policy plugin works, it is possible to reconnect automatically using it when the profile driver return -EAGAIN see:

https://github.com/bluez/bluez/blob/master/plugins/policy.c

jonas2515, 2023-10-24:
Well, kind of.. I'm actually picking up the work I did for https://github.com/bluez/bluez/issues/294 last year right now, so that's the context of this issue actually.

The specific part where the policy plugin reconnects on -EAGAIN is only valid for audio related profiles (a2dp_sink, a2dp_source, avrcp_remote, avrcp_target, hfp_hs, hsp_hs). It seems better to me to handle this in generic code than in the policy plugin, and I don't see a reason why this behavior wouldn't be wanted for other devices.

Does a patch like https://github.com/bluez/bluez/pull/633/commits/cbda34e5c85ab5311f6efdc2df9a9bb58ee1b46c make sense to you?

Vudentz, 2023-10-25:
Well I kind had it explained the reasons not to start connecting everything on that issue and how it should be implemented following the response:

https://github.com/bluez/bluez/issues/294#issuecomment-1315898047

Or you have something against a simple ReconnectStatup=on/yes logic? As for https://github.com/bluez/bluez/commit/cbda34e5c85ab5311f6efdc2df9a9bb58ee1b46c I suppose what we should do is to serialize the connection procedure using cmd_sync queue in the kernel, I believe we already done something similar to LE links so it awaits for the respective event before attempting starting to do a new connection procedure.

That said it really unlike that there will be more than one or two devices connected when bluetoothd is stopped and we can just store that information in the filesystem with other innformation cached, then the next time bluetoothd start it would know what needs to be reconnected, anyway with LE Audio at least this is not a problem since the peripheral can indicate its presence via advertisements so we can use passive scanning.

Perhaps with the latest changes that probe driver when they are discovered we could instead use something similar to classic, although there is nothing like passive scanning on classic that we can leave on in the background scanning for peripherals, thus why it is really a bad idea to just connect everything as well which seems to be the trend at the moment when I restart bluetoothd I see pipewire attempting to connect to every single audio device currently paired, so even if we serialize the process it would make it very lengthy process which effectively would prevent other request to get thru as well.

jonas2515, 2023-10-26:
> Well I kind had it explained the reasons not to start connecting everything on that issue and how it should be implemented following the response:

Yup, and that made a lot of sense to me, I think the ReconnectStartup logic is definitely the way to go, once we have that, we can remove the "connect all" logic from pipewire.

About the connect to "more than one/two" devices at the same time case: IMO handing this case is also just a question of making life easier for the layers above: It's not unrealistic that a user clicks two devices in the bluetooth quick-connect menu of their DE, right now the second one would fail, which is unexpected if the user knows the device is powered on and in vicinity.

Thanks for the pointer towards the kernel, this indeed seems to fit better there: Kernel already has a state for "try again later" that is [BT_CONNECT2](https://elixir.bootlin.com/linux/latest/source/net/bluetooth/hci_conn.c#L2564), it looks quite doable to reuse that and try again in case of the relevant HCI errors.

jonas2515, 2023-10-26:
Turns out this case is actually already handled in the kernel and the "try again later" code I mentioned was later extended to handle the "Many controllers disallow HCI Create Connection while it is doing HCI Inquiry" case: https://github.com/torvalds/linux/commit/4c67bc74f016b0d360b8573e18969c0ff7926974

Looks like cards are simply behaving very differently here, so we might only have to extend this to handle "Hardware failure" and "Limited resources" errors

closed 2025-09-13 by github-actions[bot]