bluez/bluez #632: Soundcore Life P2 missing A2DP sinks in 5.69 (5.66 works fine)

Cat-Lady, 2023-10-24:
Hello bluez devs,
As in the title, when using version 5.69, Soundcore Life P2 are having bogus A2DP **Source** entry ("Sound Gateway", that does absolutely nothing) and (working) HSP/HFP + micropfone ones. There is no A2DP sink of any kind - no SBC, no AptX, no nothing.

Downgrading to version 5.66 (the previous one that is available on Debian repositories) brings back the proper audio sinks. I have no further info, but I am happy to provide any requested logs/perform tests, if directed what exactly is needed (output of which command, specifically).

Vudentz, 2023-10-24:
@Cat-Lady try with 5.70, 5.69 had some bugs although I don't recall if was A2DP or just HID that didn't work properly.

Cat-Lady, 2023-10-30:
Confirmed working again in 5.70. Thanks!

closed 2023-10-30 by Cat-Lady

Cat-Lady, 2024-02-23:
The issue re-appeared (exact same manner) in 5.71.