bluez/bluez #634: About write and notify in the reconnection of a BLE peripheral.

imammy-hacomono, 2023-10-25:
I am a beginner in BLE programming using bluez and dbus.

I am trying to establish communication between a Raspberry Pi as a BLE peripheral and other centrals. 
In this process, I am attempting to connect with centrals using the pairing feature, and the pairing configuration is set to 'just works' (NoInputNoOutput). 
I have prepared an Agent following the example in https://github.com/PunchThrough/espresso-ble/blob/master/app.py.

When I execute the script, start advertising, and establish a connection through pairing, both notifications and write without response work while maintaining the connection.
However, once I disconnect and attempt to reconnect, notify and write functions no longer work. 
I can resolve this issue by deleting pairing information from bluetoothctl and pairing again to maintain the connection.

However, this approach creates a problem when either the central or peripheral restart, resulting in a loss of connection, and communication cannot be resumed until re-pairing is performed.
Is there any workaround for this issue? I've been researching for about a week, but I haven't found any useful information and I'm quite stuck.

I will provide Python code as needed.

bluetoothctl v5.66

imammy-hacomono, 2023-10-25:
This is a log of btmon during reconnection.

~~~
Bluetooth monitor ver 5.66
= Note: Linux version 6.1.0-rpi4-rpi-v8 (aarch64)                                                                                                     0.165777
= Note: Bluetooth subsystem version 2.22                                                                                                              0.165788
= New Index: DC:A6:32:93:90:06 (Primary,UART,hci0)                                                                                             [hci0] 0.165791
= Open Index: DC:A6:32:93:90:06                                                                                                                [hci0] 0.165798
= Index Info: DC:A6:32:93:90:06 (Cypress Semiconductor)                                                                                        [hci0] 0.165802
@ MGMT Open: bluetoothd (privileged) version 1.22                                                                                            {0x0001} 0.165806
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                             #1 [hci0] 14.881783
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                    {0x0001} [hci0] 14.881835
        LE Address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
        Reason: Connection timeout (0x01)
> HCI Event: LE Meta Event (0x3e) plen 19                                                                                                  #2 [hci0] 14.887068
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 65
        Role: Peripheral (0x01)
        Peer address type: Public (0x00)
        Peer address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
        Connection interval: 200.00 msec (0x00a0)
        Connection latency: 0 (0x0000)
        Supervision timeout: 8100 msec (0x032a)
        Central clock accuracy: 0x07
@ MGMT Event: Device Connected (0x000b) plen 13                                                                                      {0x0001} [hci0] 14.928945
        LE Address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
        Flags: 0x00000000
        Data length: 0
> HCI Event: Vendor (0xff) plen 5                                                                                                          #3 [hci0] 14.887115
        55 00 00 41 00                                   U..A.           
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                                                                                #4 [hci0] 14.929090
        Advertising: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                #5 [hci0] 14.933477
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2                                                                           #6 [hci0] 14.933619
        Handle: 65 Address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
> HCI Event: Command Status (0x0f) plen 4                                                                                                  #7 [hci0] 14.934024
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> ACL Data RX: Handle 65 flags 0x02 dlen 27                                                                                                #8 [hci0] 15.088031
      ATT: Find By Type Value Request (0x06) len 22
        Handle range: 0x0001-0xffff
        Attribute type: Primary Service (0x2800)
          UUID: Vendor specific (36b40000-42ab-4add-b2dc-e178c2de6303)
> HCI Event: LE Meta Event (0x3e) plen 12                                                                                                  #9 [hci0] 15.287762
      LE Read Remote Used Features (0x04)
        Status: Unsupported Remote Feature / Unsupported LMP Feature (0x1a)
        Handle: 65 Address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
        Features: 0x3f 0x00 0x00 0x08 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Peripheral-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          Remote Public Key Validation
< ACL Data TX: Handle 65 flags 0x00 dlen 16                                                                                               #10 [hci0] 15.287875
      LE L2CAP: Connection Parameter Update Request (0x12) ident 1 len 8
        Min interval: 24
        Max interval: 40
        Peripheral latency: 0
        Timeout multiplier: 810
< ACL Data TX: Handle 65 flags 0x00 dlen 6                                                                                                #11 [hci0] 15.288132
      SMP: Security Request (0x0b) len 1
        Authentication requirement: Bonding, No MITM, SC, No Keypresses, CT2 (0x29)
[... 158 lines cut ...]
        Connection latency: 0 (0x0000)
        Supervision timeout: 8100 msec (0x032a)
> HCI Event: LE Meta Event (0x3e) plen 10                                                                                                 #49 [hci0] 94.588074
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
[... cut ...]
        Connection latency: 0 (0x0000)
        Supervision timeout: 8100 msec (0x032a)
> HCI Event: LE Meta Event (0x3e) plen 10                                                                                                #51 [hci0] 115.688132
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
        Handle: 65 Address: E5:BB:66:EE:DC:66 (OUI E5-BB-66)
        Connection interval: 200.00 msec (0x00a0)
        Connection latency: 0 (0x0000)
        Supervision timeout: 8100 msec (0x032a)

~~~

imammy-hacomono, 2023-10-25:
@Vudentz 
Would you be able to help with resolving this issue, if you don't mind? I'm really at my wit's end...

imammy-hacomono, 2023-10-30:
Changing "ControllerMode" to "le" in /etc/bluetooth/main.conf resolved the issue. I'm not entirely sure what the root cause was, but I'm closing this now. I hope this helps someone. 

I'm also frustrated with the lack of support.

closed 2023-10-30 by imammy-hacomono

Vudentz, 2023-10-31:
> Changing "ControllerMode" to "le" in /etc/bluetooth/main.conf resolved the issue. I'm not entirely sure what the root cause was, but I'm closing this now. I hope this helps someone.
> 
> I'm also frustrated with the lack of support.

Bad timing, Im on vacation for a few weeks, anyway good that you managed to resolve it yourself.