bluez/bluez #640: cannot perform directed advertising.

imammy-hacomono, 2023-10-31:
In the current bluez examples, there are only samples for undirected advertising.
Is there a way to perform direct advertising using a peer address (e.g., 11:22:33:44:55:66)?

imammy-hacomono, 2023-11-10:
@Vudentz 

Hello.
I want to broadcast advertisements containing the MAC address of the central device identified through pairing, with the intention of allowing connections only from a specific central device.

Alternatively, do you know of any other reliable approaches?

Vudentz, 2023-11-10:
@imammy-hacomono it requires the use of MGMT_OP_SET_ADVERTISING:

https://github.com/bluez/bluez/blob/00fdb61d56161f523e975b4c044030f4b40abb6d/tools/l2cap-tester.c#L998

We should probably attempt to enable it over D-Bus though and then have bluetoothctl> advertise <bdaddr> as an example how to to use directed advertising.

imammy-hacomono, 2023-11-10:
@Vudentz 
Thank you.
I am using python to control BLE peripherals via Dbus.
(I'm a newbie, based on samples and internet knowledge base...)
Do you know the specific way to enable direct advertise via Dbus?

closed 2025-08-05 by github-actions[bot]