bluez/bluez #642: DualShock 4 controller doesn't stay connected until the bluetooth service is reloaded

tognee, 2023-11-01:
I'm currently using Fedora 38 with kernel 6.5.8-200.fc38.x86_64 and a genuine ps4 controller.

I can connect the device correctly both through the Gnome GUI and bluetoothctl.
After the device is connected it stays on for a couple of seconds with a white light and then disconnect right after.
Doing power off, then power on doesn't fix the issue, but restarting the bluetooth service does.

I tried looking in journalctl and dmesg and I couldn't find anything related to this issue. On bluetoothctl it just says:
```
[CHG] Device 48:18:8D:54:8E:EB Connected: yes
[CHG] Device 48:18:8D:54:8E:EB ServicesResolved: yes
[CHG] Device 48:18:8D:54:8E:EB ServicesResolved: no
[CHG] Device 48:18:8D:54:8E:EB Connected: no
```

Web-eWorks, 2023-12-19:
Can confirm on Solus Linux with bluez 5.70 on kernel 6.6.7. Dualshock 4 controller connects briefly, then immediately disconnects with the above log from `bluetoothctl`. Running `systemctl restart bluetooth.service` allows it to fully connect.

clemenscodes, 2023-12-21:
Currently only bluez 5.68 allows me to connect a Dualshock 4 controller. For me, restarting the bluetooth service did not solve the issue on a different version than 5.68.

Isti115, 2023-12-29:
Oh, wow, thanks for the workaround! I can now pair the controller, restart the service, connect it again, wait for the service authorization prompt (or trust it preemptively) and get a proper connection.

I have previously dealt with this on Arch by simply downgrading the `bluez` package, but since moving to NixOS I didn't put in the effort to figure out how to override the version in the system config, and the `bluetooth` service restart is a viable temporary solution for me!

legluondunet, 2024-01-01:
I have the same issue on 
- Manjaro OS: Manjaro 23.1.1 with  Kernel: x86_64 Linux 6.6.8-2-MANJARO
- Bluez 5.71-3

I can add PS4 controller with Blutooth, use it as long as I need. But if I disconnect it, it will connect then immediatly disconnect. I have to removed it and add it again like a new controller each time I want to use it.
Restart Bluetooth server workaround does not work for me.
I can reproduce this issue with command line on blutoothctl:

```

[CHG] Device 98:B6:E9:5E:CF:07 Connected: yes
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0x0075 (117)
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Value:
[Wireless Controller]#   42 04 01 20 6f 21 0d 00 22 01 41 01 01 00 01 00  B.. o!..".A.....
[Wireless Controller]#   00 00 00 00 00 00 00 05                          ........        
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0xff19 (65305)
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Value:
[Wireless Controller]#   00 75 00 09 01 00 00 00 06 01 00 00 00 00 00 00  .u..............
[Wireless Controller]#   00 00 00 00 00 00 00 00                          ........        
[CHG] Device 98:B6:E9:5E:CF:07 Connected: no
[bluetooth]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0x0075 (117)

```

thank you for your help.

legluondunet, 2024-01-01:
I just downgraded to 5.68 and the issue is still here.

tognee, 2024-01-01:
@legluondunet I think that's a different issue, you should open a new issue with your specific details.

This issue is for the controller not connecting the first time and working the second time, yours is the opposite

legluondunet, 2024-01-01:
ok I deleted my comments and created a new issue, thanks.

sant123, 2025-05-28:
I'm experiencing the same with my DualShock 4, it works for the first time (paired) however for the next times it does not connect. I use this script as a workaround:

```sh
#!/bin/bash

# Replace this with your controller's Bluetooth MAC address
DS4_MAC=""

echo "[+] Resetting connection to DualShock 4 ($DS4_MAC)..."
echo "[*] Put your DualShock 4 in pairing mode (hold PS + Share until it blinks rapidly)..."
read -p "Press Enter once it's blinking rapidly..."

# Create command list for persistent session
{
    echo "remove $DS4_MAC"
    echo "scan on"
    sleep 5

    # Wait for the device to show up
    for i in {1..20}; do
        echo "devices"
        sleep 1
        if bluetoothctl devices | grep -q "$DS4_MAC"; then
            break
        fi
    done

    echo "scan off"
    echo "pair $DS4_MAC"
    echo "trust $DS4_MAC"
    echo "connect $DS4_MAC"
    sleep 3
} | bluetoothctl

echo "[✓] Re-pairing complete."
```