bluez/bluez #644: Cannot start advertising or re-register an advertisement after disconnecting BLE device

jdonahue58, 2023-11-02:
I am using sdbus-cpp to set up a gatt server and LE advertisement in bluez over dbus. I am able to register everything successfully and connect and send/receive messages on the client device. My problem occurs when I disconnect from the client - when I scan for the device after disconnecting, it no longer shows up in the scan. I can see in dbus that the `org.bluez.LEAdvertisement1` exists and is still active in the `LEAdvertisingManager1`:

```
  interface org.bluez.LEAdvertisingManager1 {
    methods:
      RegisterAdvertisement(in  o advertisement,
                            in  a{sv} options);
      UnregisterAdvertisement(in  o service);
    signals:
    properties:
      readonly y ActiveInstances = 0x01;
      readonly y SupportedInstances = 0x04;
      readonly as SupportedIncludes = ['tx-power', 'appearance', 'local-name'];
      readonly as SupportedSecondaryChannels = ['1M', '2M', 'Coded'];
  };
```
I have tried calling `UnregisterAdvertisement` and then `RegisterAdvertisement`, but I get an error with registering the advertisement that my parameters are invalid, even though it is calling the same function as during the advertisement setup.
```
error time=1698951345.782440 sender=:1.3 -> destination=:1.35 error_name=org.bluez.Error.Failed reply_serial=303
   string "Failed to register advertisement" 
```
``` 
bluetoothd[541]: bluetoothd[541]: ../bluez-5.65/src/advertising.c:add_client_complete() Failed to add advertisement: Invalid Parameters (0x0d)
```
The only solution that works is cycling the power cycling on the adapter, and then it shows up in the scan once again and I am able to connect.

My questions are: 
1. Should the advertisement automatically resume?
2. Is this expected behavior?
3. Is there an easier way to resume advertising after a disconnect event over dbus?

My bluez version is 5.65. 

Any help would be appreciated!
Thanks

jdonahue58, 2024-01-22:
@Vudentz Any feedback on this? Thanks

pclass-sensonix, 2024-01-31:
I'm seeing the same thing. Have you made any progress in figuring this out?

Vudentz, 2024-02-02:
@jdonahue58 advertising are re-enabled once disconnected, is there something that indicates otherwise? Do you have btmon traces? Here is the trace when using bluetootctl> advertise on:

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                     #659 [hci0] 15:50:08.735138
        Status: Success (0x00)
        Handle: 2048 Address: 54:6C:6F:06:29:0B (Resolvable)
          Identity type: Random (0x01)
          Identity: 80:39:8C:93:7C:E5 (Reserved)
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                              {0x0001} [hci0] 15:50:08.735177
        LE Address: 80:39:8C:93:7C:E5 (Samsung Electronics Co.,Ltd)
        Reason: Connection terminated by remote host (0x03)
< HCI Command: LE Set Extended Advertising Enable (0x08|0x0039) plen 6                                             #660 [hci0] 15:50:08.755277
        Extended advertising: Enabled (0x01)
        Number of sets: 1 (0x01)
        Entry 0
          Handle: 0x01
          Duration: 0 ms (0x00)
          Max ext adv events: 0
> HCI Event: Command Complete (0x0e) plen 4                                                                        #661 [hci0] 15:50:08.756025
      LE Set Extended Advertising Enable (0x08|0x0039) ncmd 2
        Status: Success (0x00)
```

So after disconnecting it immediately re-enable advertising.

jdonahue58, 2024-02-02:
Yes, I get an `Invalid HCI Command Parameters` error, which also appears after connecting over BLE. I tried the kernel patch given [here](https://stackoverflow.com/questions/61935284/invalid-params-issue-when-doing-hciconfig-hci0-reset) but that doesn't seem to get rid of the error or help in any way. 
```
> HCI Event: Disconnect Complete (0x05) plen 4          #1004 [hci0] 122.380912
        Status: Success (0x00)
        Handle: 24 Address: 60:48:A0:6D:AB:A2 (Resolvable)
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8    {0x0001} [hci0] 122.381112
        LE Address: 60:48:A0:6D:AB:A2 (Resolvable)
        Reason: Connection terminated by remote host (0x03)
< HCI Command: LE Set Extended.. (0x08|0x0036) plen 25  #1005 [hci0] 122.450306
        Handle: 0x00
        Properties: 0x0013
          Connectable
          Scannable
          Use legacy advertising PDUs: ADV_IND
        Min advertising interval: 1280.000 msec (0x0800)
        Max advertising interval: 1280.000 msec (0x0800)
        Channel map: 37, 38, 39 (0x07)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:00:00:00:00:00 (OUI 00-00-00)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
        TX power: Host has no preference (0x7f)
        Primary PHY: LE 1M (0x01)
        Secondary max skip: 0x00
        Secondary PHY: LE 1M (0x01)
        SID: 0x00
        Scan request notifications: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 5             #1006 [hci0] 122.451893
      LE Set Extended Advertising Parameters (0x08|0x0036) ncmd 2
        Status: Invalid HCI Command Parameters (0x12)
        TX power (selected): 0 dbm (0x00)
```

Vudentz, 2024-02-02:
@jdonahue58 what kernel version is this? I don't see anything wrong with the parameters that said it is quite strange that it would reprogram them instead of just re-enable but perhaps that has been fixed in the kernel a while back.

jdonahue58, 2024-02-02:
@Vudentz looks like mainline kernel 4.19.182

Vudentz, 2024-02-02:
> @Vudentz looks like mainline kernel 4.19.182

Yeah, no wonder it is not working, it is very unlikely you will get any sort of backported features for something this old:

```
commit 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d (tag: v4.19)
Author: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
Date:   Mon Oct 22 07:37:37 2018 +0100

    Linux 4.19
```

jdonahue58, 2024-02-02:
@Vudentz Do you know of a set of patches that fixes the issue?

closed 2025-08-05 by github-actions[bot]