bluez/bluez #648: Switch to gitlab or github for patch contribution workflow

jonas2515, 2023-11-07:
Disclaimer: Quite the long shot and possibly controversial, but putting this out there anyway as I think the discussion needs to happen. If the maintainers think this is the wrong place to discuss such a question, feel free to close this issue again.

The traditional kernel workflow of sending email with patches to mailing lists is a really high barrier to contributions and seems archaic for most developers nowadays. The arguments in favour and against it have been made a lot of times already, so I won't repeat them here.

Since BlueZ already has a Github based workflow for issue reporting, it seems fairly straightforward to also switch to Github for submitting patches. Are there plans to do that eventually, is it maybe already in the works, or are there specific reasons why it can't be done at this point?

Vudentz, 2023-11-07:
I don't oppose, that said I think we should look into having a hybrid approach where we reflect the pull requests to the mailing list and vice-versa, so we have proper email archives, etc, since there could be people still depending on having the tradition kernel workflow for automation.

Btw, we do have the intention of migrating the github CI into the bluez tree, they are currently maintained in @BluezTestBot, and with that perhaps we can attempt to handle PR requests to email which then creates patchwork items which should trigger CI automation, that said we currently don't have anyone really maintaining github automation so I hope someone can pickup these items.

jonas2515, 2023-11-07:
That's great to hear, so from how I understand it, two main things would have to be done:

- move the [CI workflow](https://github.com/BluezTestBot/bluez/blob/workflow/.github/workflows/ci.yml) over from the BluezTestBot fork to this repo
- add a workflow to this repo that runs on new PRs + changes to PRs, transforms the PR into a patch email and sends that to the linux-bluetooth mailing list. I wonder if [gitgitgadget](https://github.com/gitgitgadget/gitgitgadget) could be reused for this, it seems to target a slightly different use case, but it would solve the tricky parts like creating a v2 series on a new push to the PR.

Once these things are done, the [sync_patchwork.py](https://github.com/tedd-an/bzcafe/blob/main/sync_patchwork.py) script could be adapted to reply to new patches posted to linux-bluetooth, telling people to open a PR on github.

Vudentz, 2023-11-08:
> That's great to hear, so from how I understand it, two main things would have to be done:
>
> * move the [CI workflow](https://github.com/BluezTestBot/bluez/blob/workflow/.github/workflows/ci.yml) over [...]

That gitgitgadget looks promising indeed and I really like the idea of /submit and it does handle things like cover letter, and adding the changes between versions which is pretty cool.

> Once these things are done, the [sync_patchwork.py](https://github.com/tedd-an/bzcafe/blob/main/sync_patchwork.py) script could be adapted to reply to new patches posted to linux-bluetooth, telling people to open a PR on github.

Well Id keep the mailing list as primary workfollow, at least for some time until we stabilize things and people get more familiar, but I got the impression a lot of people will prefer to use github from the start.

Vudentz, 2024-03-18:
@evelikov @pv @tedd-an https://github.com/tedd-an/bzcafe/pulls perhaps we can start by moving the scripts under scripts/ subdir in the main tree so we can drop the use of Tedd's bzcafe tree entirely? Any volunteer to start that?

jonas2515, 2024-04-02:
> Looking at gitgitgadget - it also seems to require third-party service to send/receive emails. Cannot find mind details on replicating their setup either.

Yeah, I think replicating that setup will need a bit of work and definitely maintenance too, I'm not sure it's worth investing so much time into something that ultimately *always* worsens the github/gitlab experience.

> As long as we have very rapid turnaround on review/merges, I don't mind kicking something off - here or gitlab.

Hmm, do you mean that as in "simply do completely github based contribution workflow in parallel to mailing lists"?

jonas2515, 2024-04-09:
I'm happy to do a bunch of work too, but I'd more tend towards switching the primary way of contributing to GitHub/Gitlab immediately, I think first going "better GH support, but mailing list is still primary" to later "GH is primary, but mailing list still supported" adds a ton of extra work, while the contributor experience on GH would still be subpar.

Another question would be how kernel contributions would work, ideally they should also have a GH repo under the bluez org, I wonder if there's prior art from other kernel subsystems.

Vudentz, 2024-04-10:
> I'm happy to do a bunch of work too, but I'd more tend towards switching the primary way of contributing to GitHub/Gitlab immediately, I think first going "better GH support, but mailing list is [...]

Well we still need to at least process the mailing list for patches posted there, for the kernel there is no way around it since that is how kernel tree works.

Anyway, what do you guys need, I can add a new team with @evelikov and @jonas2515 and create a dedicate tree for migrating the bzcafe, etc, then have you guys as maintainers of that so you can make the changes necessary to bzcafe in parallel so when that becomes stable we switch to it, how about that?

Vudentz, 2025-04-30:
> I'm happy to do a bunch of work too, but I'd more tend towards switching the primary way of contributing to GitHub/Gitlab immediately, I think first going "better GH support, but mailing list is [...]

Ive moved the actions to bluez/bluez tree, so now the actions can be run on any PR done directly to bluez, and Im also taking advantage of GH action market place to add some more testing.

> Well we still need to at least process the mailing list for patches posted there, for the kernel there is no way around it since that is how kernel tree works.
>
> Anyway, what do you guys need, I can [...]

With the migration now done, it shall be simpler to address these things via action-ci tree, or with a third-party action.