bluez/bluez #650: Bluetooth Pairing unsuccessful with authentication failure on iOS devices on Raspi 4

bisahu, 2023-11-10:
I'm running into an issue where iOS devices fail to connect and pair with a Raspberry Pi operating as peripheral.

**System Configuration**

Bluez
$ bluetoothd --version
5.66

$ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

The issue is occurring on a GATT server created with the example [simple-agent](https://github.com/bluez/bluez/blob/master/test/simple-agent). To add further details:

1. The GATT server based on python, has a few "secure-read" & "secure-write" characteristics, which lead to the iOS devices invoking a pairing request.
2. Once a pairing pop-up is displayed on the iOS device, the user gets a choice to accept bonding by "yes" or "no".
3. Even after the user selects "yes", and it seems like the LTK is generated, the authentication fails and the device fails to connect to the device.

btmon output from when the iOS devices connects:

```
> ACL Data RX: Handle 64 flags 0x02 dlen 8                        #55 33.829126
      ATT: Write Request (0x12) len 3
        Handle: 0x002a Type: Vendor specific (5bc35bcf-d719-7986-dc48-7f6a0c9be08c)
          Data: 01
< ACL Data TX: Handle 64 flags 0x00 dlen 9                        #56 33.829421
      ATT: Error Response (0x01) len 4
        Write Request (0x12)
        Handle: 0x002a
        Error: Insufficient Authentication (0x05)
> HCI Event: Number of Completed Packets (0x13) plen 5            #57 33.829404
        Num handles: 1
        Handle: 64 Address: 4F:C3:56:99:5F:8C (Resolvable)
        Count: 2
> ACL Data RX: Handle 64 flags 0x02 dlen 11                       #58 33.889024
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses, CT2 (0x2d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey LinkKey (0x0b)
        Responder key distribution: EncKey IdKey LinkKey (0x0b)
< ACL Data TX: Handle 64 flags 0x00 dlen 11                       #59 33.889126
      SMP: Pairing Response (0x02) len 6
        IO capability: NoInputNoOutput (0x03)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, Legacy, No Keypresses (0x05)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey (0x03)
        Responder key distribution: EncKey (0x01)
> HCI Event: Number of Completed Packets (0x13) plen 5            #60 33.949288
        Num handles: 1
        Handle: 64 Address: 4F:C3:56:99:5F:8C (Resolvable)
        Count: 2
> ACL Data RX: Handle 64 flags 0x02 dlen 21                       #61 34.819084
      SMP: Pairing Confirm (0x03) len 16
        Confim value: aecf93759daa69112f14c84d11c3a6f5
< ACL Data TX: Handle 64 flags 0x00 dlen 21                       #62 34.819148
      SMP: Pairing Confirm (0x03) len 16
        Confim value: 129c5210db7d6d0707514e0541b02703
> ACL Data RX: Handle 64 flags 0x02 dlen 21                       #63 34.879117
      SMP: Pairing Random (0x04) len 16
        Random value: 1cc33b694877c40f560c5412e355e66c
< ACL Data TX: Handle 64 flags 0x00 dlen 21                       #64 34.879195
      SMP: Pairing Random (0x04) len 16
        Random value: 86633792102b9856fb0b8119e09c4c00
> HCI Event: Number of Completed Packets (0x13) plen 5            #65 34.939262
        Num handles: 1
        Handle: 64 Address: 4F:C3:56:99:5F:8C (Resolvable)
        Count: 2
> HCI Event: LE Meta Event (0x3e) plen 13                         #66 34.969693
[... 359 lines cut ...]
        Key: 8294e5c85eacbcd70072946eeac792b3
@ MGMT Event: New Long Term Key (0x000a) plen 37             {0x0001} 37.970961
        Store hint: Yes (0x01)
        LE Address: 60:7E:C9:DD:43:7B (Apple, Inc.)
        Key type: Unauthenticated legacy key (0x00)
        Central: 0x00
        Encryption size: 16
        Diversifier: 7f00
        Randomizer: f87e290894d0f27d
        Key: fef63ff012add1bd9e2a263823964e0e
> ACL Data RX: Handle 65 flags 0x02 dlen 8                       #140 37.971461
      ATT: Write Request (0x12) len 3
        Handle: 0x002a Type: Vendor specific (5bc35bcf-d719-7986-dc48-7f6a0c9be08c)
          Data: 01
< ACL Data TX: Handle 65 flags 0x00 dlen 9                       #141 38.012960
      ATT: Error Response (0x01) len 4
        Write Request (0x12)
        Handle: 0x002a
        Error: Insufficient Authentication (0x05)
```

Supplement queries:

1. I would ideally like the rasp pi to transmit as LE only mode, but unfortunately, changing the Controller Mode to "le" in `/etc/bluetooth/main.conf` doesn't change that. Is there a way, it could be achieved using python and dbus programatically?
2. Based on the example, I couldn't find an option to add advertising flags, specifically the "GeneralDiscoverable" and "BrEdrNotSupported" flags. Is there a way, it could be added using the example on python?

xzZero, 2023-11-25:
`secure-read` and `secure-write` force the pairing/bonding of the connection under LE Secure Connections, in which Numerical Comparison will be used. `NoInputNoOutput` can't be used for LE Secure Connections, it will trigger Numerical Comparison anyway. But because you are using the wrong Agent Capability, the pairing/bonding was failed. If you want to go with LE Secure Connections, just change the Agent Capability to `KeyboardDisplay`.

Supplement queries:

1. 
- Change Controller Mode to le would work until you restart the bluetooth service
```
sudo systemctl daemon-reload
sudo systemctl stop bluetooth.service
sudo systemctl restart bluetooth.service
```
- You can access the bluez management API [btmgmt ](https://helpmanual.io/help/btmgmt/)with [subprocess ](https://docs.python.org/3/library/subprocess.html)or using [btmgmt python wrapper](https://github.com/BOJIT/btmgmt)

2. When you are able to set the Raspberry Pi to LE only mode, the flag `BrEdrNotSupported` will be included in the advertising packet

dev-enphase, 2024-01-13:
How we can disable secure connection in bluez 5.66. I am doing through main.conf but it seems that bluetoothd not taking the settings?

Any ways to disable le secure connection?

dev-enphase, 2024-01-13:
I am facing issue where pairing works first time but when paired device is unpair from mobile then pairing never happens in noinputnooutput capability. Any solution for this?

closed 2025-08-04 by github-actions[bot]