bluez/bluez #652: Advertisement monitor matching no longer works for most data types

lopsided98, 2023-11-12:
https://github.com/bluez/bluez/commit/bc2ab4ba3c6103540b902802a3b156f769791fca broke advertisement monitor pattern matching for all the UUID and service data types. The advertisement monitor pattern matching uses the `ad_data.data` list, which used to contain all the advertising data fields, but now only contains fields that are not parsed. The parsed fields are instead added to the `ad_data.service_uuids` or `ad_data.service_data` lists, which are not used in pattern matching.

I think this can either be fixed by extending the pattern matching to look at `ad_data.service_uuids` and `ad_data.service_data`, or making it so all fields are still added to `ad_data.data`, even if they are parsed as well.

Vudentz, 2023-11-17:
@lopsided98 are you talking about adv monitor or scan pattern matching? It would help if you also include the exact steps to reproduce.

lopsided98, 2023-11-18:
You should be able to reproduce this in bluetoothctl. If you have any Apple devices nearby, you should be able to detect them with:
```
menu monitor
add-or-pattern 0 1 1a
```
The above should work even with BlueZ 5.69 and newer. On the other hand, if you then run:
```
remove-pattern all
add-or-pattern 0 255 4c
```
On BlueZ 5.68 and older, this will detect the same devices, but on newer versions, no devices will be detected.

You also likely need to make sure your Bluetooth adapter doesn't support the Microsoft vendor extensions, as I believe hardware pattern matching will prevent you from being able to reproduce the bug, but I don't have any hardware to verify this.

The difference between the two patterns is that the second data type (BT_AD_MANUFACTURER_DATA) appears in the switch statement below, while the first (BT_AD_FLAGS) does not:
https://github.com/bluez/bluez/blob/8b035b70f379e45a32c94579ec8d00e48070c21e/src/shared/ad.c#L366-L387

closed 2023-12-08 by BluezTestBot