bluez/bluez #654: Unable to pair Logitech K480 on bluez 5.70

RodoMa92, 2023-11-15:
I've tried to pair with both the KDE bluetooth gui and bluetoothctl and both fail, bluetoothctl report AuthenticationCanceled.

```
[NEW] Device F4:73:35:xx:xx:xxKeyboard K480
...
[Device]# pair F4:73:35:xx:xx:xx
Attempting to pair with F4:73:35:xx:xx:xx
[CHG] Device F4:73:35:xx:xx:xx Connected: yes
...
[CHG] Device F4:73:35:xx:xx:xx Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
```

Running on a Steam Deck with a Realtek RTL8822CE adapter. This keyboard was previously successfully paired to this device with no issues (with older version of Bluez). My smartphone is able to pair, connect and receive keyboard input just fine.

Leopard1907, 2023-11-17:
https://github.com/bluez/bluez/issues/605

Probably same issue. Latest blue-z release has borked device pairing. 0.5.68 works ok, i was able to dowgrade and connect with it, Arch Linux.

RodoMa92, 2023-12-14:
> Edit: I'm leaving all these words for posterity (and just in case I'm wrong) but it's starting to look like my issue is a kernel regression, not `bluez`. Stood up Ubuntu 20.04.6, built 477c41c and it paired perfectly...then went back to [the 5.70 tagged release](https://github.com/bluez/bluez/releases/tag/5.70) and that seems to be working perfectly too.
> 
> ---
> 
> This looks a lot like the issues I'm having with a [ZMK](https://github.com/zmkfirmware/zmk) ([Zephyr 3.2-based](https://github.com/zmkfirmware/zephyr)) BLE keyboard and Fedora 39 (for once, someone else with a Realtek adapter...) The symptoms aren't the same as #605.
> 
> The initial connect works, but I try to pair the device and it just doesn't take. Eventually something gives up (not sure if keyboard or computer) and `bluez` logs `Failed to pair: org.bluez.Error.AuthenticationCanceled`.
> If I switch to firmware that requires a passkey, I'm prompted for the passkey. Logging on the keyboard indicates it is being entered correctly and a flurry of data in `btmon` scrolls by but the result is the same.
> 
> It pairs happily on a live USB of **Ubuntu 20.04.6** LTS (bluez 5.53, kernel 5.15) I just happened to have around from May, but the current Ubuntu 22.04.3 LTS shows the same `AuthenticationCanceled`. I tried a few old CentOS RPMs (5.56, 5.66, 5.69) quickly and none worked; couldn't get all the way back to 5.52 for dependency reasons. @RodoMa92, do you remember what version of bluez (...and maybe what OS/kernel) used to work for you?
> 
> `dmesg` shows `Bluetooth: hci0: unexpected cc 0x0c7c length: 1 < 3` when attempting to pair. No idea if related, but I can say that doesn't show up on a successful pair with Ubuntu 20.04.6. I pasted this into the `bluetoothd -d` log file below where it happens.
> 
> `uname -a`:
> ```
> Linux raymond 6.6.4-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Dec  3 18:13:11 UTC 2023 x86_64 GNU/Linux
> ```
> [`lsusb -s 1:5 -v` for the Realtek adapter](https://github.com/bluez/bluez/files/13630581/lsusb-v_realtek.txt)
> 
> Keyboard's MAC address is `E6:82:B1:1B:6B:EE`:
> - [`bluetoothctl`](https://github.com/bluez/bluez/files/13630636/bluez_fedora_2023-12-10.txt)
> - [`bluetoothd -d`](https://github.com/bluez/bluez/files/13630635/bluetoothd_fedora_2023-12-10.log)
> - [`btmon`](https://github.com/bluez/bluez/files/13630637/btmon_fedora_2023-12-10.log) (onlookers note that this is in btsnoop format, not text like the others)
> 
> Some of these logs might be from unmatched attempts but all the passkeys were entered correctly. If you want more logs or the logs from the keyboard itself I can provide those. Ultimately the keyboard does seem to think it has paired successfully and saves the MAC address of the controller.
> 
> I at least have a fair amount of flexibility w/r/t flashing new firmware for my keyboard, so distro-hopping aside I might try to build new firmware using the (experimental) Zephyr 3.5 branch of ZMK.
> 
> Lastly: after pairing with Ubuntu 20.04.6, I can paste the `E6:82:B1:1B:6B:EE` cache and folder under `/etc/lib/bluetooth/<controller>` in Fedora 39, run `systemctl restart bluetooth.service` and everything works. Might be a viable solution for people who don't need to constantly un-pair and re-pair their keyboards due to active development (i.e. not me)

Oh, no, I hope the hell not it's a kernel regression, this will take ages to track down.

Sadly, I'm not sure about specific versions. If I have to guess it was a 6.5.x kernel rev + Bluez 5.67/68 (?). I'm sorry to not give you more pointers to this.

There is quite a bit of people with pairing issues on my distribution on Deck, so you might be getting somewhere on the kernel regression, but bisecting on Deck is a PITA. If you can so something like that on your end that would be lovely. I have a Realtek Bluetooth adapter, I can test if I can reproduce the same on my desktop in case.

RodoMa92, 2023-12-14:
Well, I hate to admit it @lesshonor, but you might be right. I've just tested pairing this keyboard on my desktop with a Realtek radio with VID:PID of 0bda:8771 and bluez 5.70 and it immediately prompt me for a keycode and paired first try. I'm happily typing it right now from it.

This quite might be a kernel related bluetooth regression. Fun.

RodoMa92, 2023-12-14:
@lesshonor if you want to add specific details to the kernel devs, the bug is [here](https://bugzilla.kernel.org/show_bug.cgi?id=218264).

RodoMa92, 2023-12-17:
Even copying manually the key associated to the same adapter under another OS will not work with an error on connect br-connection-create-socket. Not sure if it's useful.

Zelaf, 2023-12-17:
Well, been having issues on a Steam Deck as well connecting to a Keychron K7. A Corsair Katar PRO Wireless mouse has been working just fine. On my distro of Bazzite I've also had issues with connecting to a Bluetooth Speaker but seems to have been in working order this evening too.

```
[Living Room speaker]# show 14:D4:24:3C:CC:84 
Controller 14:D4:24:3C:CC:84 (public)
        Name: fedora
        Alias: fedora
        Class: 0x001c010c (1835276)
        Powered: yes
        PowerState: on
        Discoverable: no
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        Modalias: usb:v1D6Bp0246d0546
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x04 (4)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded
[Living Room speaker]# scan on
Discovery started
[CHG] Controller 14:D4:24:3C:CC:84 Discovering: yes
{Fluff...}
[NEW] Device DC:2C:26:3E:57:75 Keychron K7
{Fluff...}
[Living Room speaker]# scan off
Discovery stopped
{Fluff...}
[CHG] Controller 14:D4:24:3C:CC:84 Discovering: no
[Living Room speaker]# pair DC:2C:26:3E:57:75 
Attempting to pair with DC:2C:26:3E:57:75
[CHG] Device DC:2C:26:3E:57:75 Connected: yes
{Fluff...}
[CHG] Device DC:2C:26:3E:57:75 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
[Living Room speaker]# pair DC:2C:26:3E:57:75 
Attempting to pair with DC:2C:26:3E:57:75
[CHG] Device DC:2C:26:3E:57:75 Connected: yes
[CHG] Device DC:2C:26:3E:57:75 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
[Living Room speaker]# agent KeyboardOnly
Agent is already registered
[Living Room speaker]# default-agent
Default agent request successful
[Living Room speaker]# pair DC:2C:26:3E:57:75 
Attempting to pair with DC:2C:26:3E:57:75
[CHG] Device DC:2C:26:3E:57:75 Connected: yes
[CHG] Device DC:2C:26:3E:57:75 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
[Living Room speaker]# trust DC:2C:26:3E:57:75 
[CHG] Device DC:2C:26:3E:57:75 Trusted: yes
Changing DC:2C:26:3E:57:75 trust succeeded
[Living Room speaker]# pair DC:2C:26:3E:57:75 
Attempting to pair with DC:2C:26:3E:57:75
[CHG] Device DC:2C:26:3E:57:75 Connected: yes
[CHG] Device DC:2C:26:3E:57:75 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
[Living Room speaker]# connect DC:2C:26:3E:57:75 
Attempting to connect to DC:2C:26:3E:57:75
[CHG] Device DC:2C:26:3E:57:75 Connected: yes
[CHG] Device DC:2C:26:3E:57:75 Modalias: usb:v05ACp0250d0101
[CHG] Device DC:2C:26:3E:57:75 UUIDs: 00001000-0000-1000-8000-00805f9b34fb
[CHG] Device DC:2C:26:3E:57:75 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device DC:2C:26:3E:57:75 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device DC:2C:26:3E:57:75 ServicesResolved: yes
[CHG] Device DC:2C:26:3E:57:75 WakeAllowed: yes
[CHG] Device DC:2C:26:3E:57:75 Paired: yes
Connection successful
[Keychron K7]# pair DC:2C:26:3E:57:75 
Attempting to pair with DC:2C:26:3E:57:75
Failed to pair: org.bluez.Error.AlreadyExists
[Keychron K7]# devices Paired
Device D8:8C:79:77:76:4E Living Room speaker
Device F0:04:57:B4:65:DF KATAR PRO Wireless
Device F4:6A:D7:08:68:DE Xbox Wireless Controller
Device DC:2C:26:3E:57:75 Keychron K7
[Keychron K7]# devices Connected
Device D8:8C:79:77:76:4E Living Room speaker
Device F0:04:57:B4:65:DF KATAR PRO Wireless
Device DC:2C:26:3E:57:75 Keychron K7
[Keychron K7]# devices Bonded 
Device D8:8C:79:77:76:4E Living Room speaker
Device F0:04:57:B4:65:DF KATAR PRO Wireless
Device F4:6A:D7:08:68:DE Xbox Wireless Controller
[Keychron K7]# devices Trusted 
Device D8:8C:79:77:76:4E Living Room speaker
Device F0:04:57:B4:65:DF KATAR PRO Wireless
Device F4:6A:D7:08:68:DE Xbox Wireless Controller
Device DC:2C:26:3E:57:75 Keychron K7
[Keychron K7]# info DC:2C:26:3E:57:75 
Device DC:2C:26:3E:57:75 (public)
        Name: Keychron K7
        Alias: Keychron K7
        Class: 0x00002540 (9536)
        Icon: input-keyboard
        Paired: yes
        Bonded: no
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v05ACp0250d0101
[Keychron K7]# 
``

I am however unable to do any inputs on the keyboard at all.

RodoMa92, 2023-12-21:
Today the gods have said yes and I managed to correctly pair from the command line, both host and keyboard are bonded and connects automatically. However I do not get any input from the keyboard and this is printed on dmesg:

```
[ 6539.432356] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[ 6539.432370] Bluetooth: HIDP socket layer initialized
[ 6673.460129] hid-generic 0005:046D:B33D.0006: unknown main item tag 0x0
[ 6673.460148] hid-generic 0005:046D:B33D.0006: item fetching failed at offset 271/272
[ 6673.460171] hid-generic: probe of 0005:046D:B33D.0006 failed with error -22
[ 6706.511559] hid-generic 0005:046D:B33D.0007: unknown main item tag 0x0
[ 6706.511586] hid-generic 0005:046D:B33D.0007: item fetching failed at offset 271/272
[ 6706.511625] hid-generic: probe of 0005:046D:B33D.0007 failed with error -22
[ 6741.248981] hid-generic 0005:046D:B33D.0008: unknown main item tag 0x0
[ 6741.248997] hid-generic 0005:046D:B33D.0008: item fetching failed at offset 271/272
[ 6741.249017] hid-generic: probe of 0005:046D:B33D.0008 failed with error -22
```

Still bluez 5.70

cjcase, 2024-01-09:
> This looks a lot like the issues I'm having with a [ZMK](https://github.com/zmkfirmware/zmk) ([Zephyr 3.2-based](https://github.com/zmkfirmware/zephyr)) BLE keyboard and Fedora 39 (for once, someone [...]

For those of you experiencing this behavior with ZMK, I found that activating the "BLE Experimental" features for both connection and security seemed to fix this exact issue with Supervisor timeout to my keyboard. The related error codes that led me to this were the following:
* `btmon`
```
> HCI Event: Disconnect Complete (0x05) plen 4
        Status: Success (0x00)
        Handle: 2049 Address: 00:01:02:03:04:05 (Static)
        Reason: Connection Timeout (0x08)
```
* `bluetoothctl`
```
[bluetooth]# pair 00:01:02:03:04:05
Attempting to pair with 00:01:02:03:04:05
...
[CHG] Device 00:01:02:03:04:05 Connected: yes
[CarbonSplit]# [agent] Passkey: 063526
[CarbonSplit]# hci0 00:01:02:03:04:05 type LE Random disconnected with reason 1
[CHG] Device 00:01:02:03:04:05 Connected: no
[bluetooth]# Failed to pair: org.bluez.Error.AuthenticationCanceled
```

You can fix your firmware by adding the following to your ZMK `.conf` file:
```
# Bluetooth (https://zmk.dev/docs/config/bluetooth)
CONFIG_ZMK_BLE=y
CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y # connection and security experimental features
CONFIG_ZMK_BLE_PASSKEY_ENTRY=y # security hardening
```
I had wasted some time messing with the Supervisor timeout and was fearing some Kernel issue as I noticed the same as @lesshonor mentioned with the Linux Kernal and bluez version permutations.

Hope this saves some hours of debugging :grin:

RodoMa92, 2024-02-29:
Logs on the failing connection with my pair of Soundcore Space A40 here, @Vudentz let me know if you need anything else on it:
<details>
 <summary>bluetoothctl pair A4:77:58:xx:xx:xx</summary>

```Attempting to pair with A4:77:58:xx:xx:xx
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Connected: yes
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Bonded: yes
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00001000-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000110f-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0000112e-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00001203-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 00007034-0000-1000-8000-00805f9b34fb
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx UUIDs: 0cf12d31-fac3-4553-bd80-d6832e7b3936
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx ServicesResolved: yes
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Paired: yes
[Keyboard K480]# Pairing successful
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx ServicesResolved: no
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Connected: no
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Connected: yes
[Keyboard K480]# [NEW] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep2
[Keyboard K480]# [NEW] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep4
[Keyboard K480]# [DEL] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep2
[Keyboard K480]# [DEL] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep4
[Keyboard K480]# [NEW] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep1
[Keyboard K480]# [NEW] Endpoint /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep3
[Keyboard K480]# [NEW] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep3/fd17
[Keyboard K480]# [CHG] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep3/fd17 Delay: 0x000a (10)
[Keyboard K480]# [CHG] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep3/fd17 Volume: 0x0044 (68)
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx ServicesResolved: yes
[Keyboard K480]# [DEL] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep3/fd17
[Keyboard K480]# [NEW] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep1/fd18
[Keyboard K480]# [CHG] Transport /org/bluez/hci0/dev_A4_77_58_xx_xx_xx/sep1/fd18 Delay: 0x000a (10)
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx ServicesResolved: no
[Keyboard K480]# [CHG] Device A4:77:58:xx:xx:xx Connected: no
```
</details>

hci logs here 
[hcidump.tar.gz](https://github.com/bluez/bluez/files/14449779/hcidump.tar.gz)

RodoMa92, 2024-02-29:
Oh, interesting, I just noticed I have a kernel oops in dmesg:
```
[  118.152780] ------------[ cut here ]------------
[  118.152789] WARNING: CPU: 6 PID: 129 at net/bluetooth/hci_conn.c:620 hci_conn_timeout+0x61/0xb0 [bluetooth]
[  118.152897] Modules linked in: uinput xt_conntrack xt_MASQUERADE nf_conntrack_netlink xt_addrtype nft_compat snd_seq_dummy snd_hrtimer br_netfilter bridge stp llc overlay ip6table_nat tun intel_rapl_msr nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib bnep nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 snd_soc_acp5x_mach ip_set snd_acp5x_i2s snd_acp5x_pcm_dma nf_tables nfnetlink qrtr intel_rapl_common edac_mce_amd kvm_amd kvm irqbypass rapl sunrpc snd_sof_amd_acp63 snd_sof_amd_vangogh pcspkr snd_sof_amd_rembrandt btusb snd_sof_amd_renoir snd_sof_amd_acp rtw88_8822ce btrtl acpi_cpufreq snd_sof_pci btintel snd_sof_xtensa_dsp rtw88_8822c snd_sof rtw88_pci btbcm snd_sof_utils snd_hda_codec_hdmi btmtk rtw88_core snd_pci_ps snd_hda_intel snd_rpl_pci_acp6x snd_pci_acp6x bluetooth snd_pci_acp5x snd_rn_pci_acp3x snd_intel_dspcfg snd_intel_sdw_acpi snd_acp_config mac80211 snd_hda_codec snd_soc_acpi
[  118.153113]  i2c_piix4 snd_pci_acp3x libarc4 snd_hda_core cdc_acm snd_hwdep snd_soc_cs35l41_spi snd_soc_cs35l41 snd_soc_wm_adsp cs_dsp snd_soc_cs35l41_lib snd_soc_nau8821 regmap_spi snd_soc_core legion_laptop platform_profile snd_seq snd_compress ac97_bus snd_pcm_dmaengine snd_seq_device snd_pcm snd_timer binfmt_misc snd opt3001 ltrf216a soundcore industrialio vfat fat joydev brcmfmac brcmutil cfg80211 rfkill tcp_bbr sch_fq ryzen_smu(OE) ledtrig_timer loop zstd zram mmc_block amdgpu amdxcp drm_exec gpu_sched crct10dif_pclmul drm_buddy crc32_pclmul crc32c_intel i2c_algo_bit polyval_clmulni drm_suballoc_helper polyval_generic nvme sdhci_pci drm_display_helper cqhci ghash_clmulni_intel nvme_core sha512_ssse3 cec sdhci hid_multitouch drm_ttm_helper sha256_ssse3 sha1_ssse3 wdat_wdt mmc_core ccp video ttm nvme_auth wmi i2c_hid_acpi i2c_hid serio_raw hid_steam ff_memless v4l2loopback(OE) videodev mc scsi_dh_rdac scsi_dh_emc scsi_dh_alua ip6_tables ip_tables dm_multipath i2c_dev gcadapter_oc(OE) fuse evdi(OE) uas usb_storage
[  118.153376]  dwc3_pci dwc3 ulpi udc_core extcon_steamdeck leds_steamdeck steamdeck_hwmon steamdeck xhci_plat_hcd
[  118.153402] Unloaded tainted modules: nct6687(OE):1
[  118.153413] CPU: 6 PID: 129 Comm: kworker/u33:0 Tainted: G           OE      6.7.5-202.fsync.fc39.x86_64 #1
[  118.153422] Hardware name: Valve Jupiter/Jupiter, BIOS F7A0121 12/14/2023
[  118.153428] Workqueue: hci0 hci_conn_timeout [bluetooth]
[  118.153518] RIP: 0010:hci_conn_timeout+0x61/0xb0 [bluetooth]
[  118.153606] Code: 48 c7 c1 c5 e7 e6 c1 83 e8 01 83 f8 08 76 3d 48 89 ea 48 c7 c6 d3 e7 e6 c1 48 c7 c7 08 c0 e1 c1 e8 54 89 b1 cb 45 85 e4 79 c4 <0f> 0b 0f b6 83 31 fd ff ff 3c 01 74 1f be 13 00 00 00 3c 80 74 16
[  118.153613] RSP: 0018:ffffbade00547e50 EFLAGS: 00010286
[  118.153620] RAX: ffffffffc1db7640 RBX: ffff942c02901308 RCX: ffff942c08016428
[  118.153625] RDX: 0000000000000001 RSI: ffff942c02901310 RDI: ffff942c02901308
[  118.153630] RBP: ffff942c02901000 R08: ffff942c0dc88340 R09: ffff942c0dc88380
[  118.153635] R10: 0000000000000005 R11: 0000000000000100 R12: 00000000ffffffff
[  118.153640] R13: ffff942c08016400 R14: ffff942c10417615 R15: ffff942c02901310
[  118.153645] FS:  0000000000000000(0000) GS:ffff942f2fd80000(0000) knlGS:0000000000000000
[  118.153651] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  118.153657] CR2: 00007f56c689a000 CR3: 000000013fd26000 CR4: 0000000000350ef0
[  118.153681] Call Trace:
[  118.153688]  <TASK>
[  118.153693]  ? hci_conn_timeout+0x61/0xb0 [bluetooth]
[  118.153781]  ? __warn+0x81/0x130
[  118.153797]  ? hci_conn_timeout+0x61/0xb0 [bluetooth]
[  118.153888]  ? report_bug+0x171/0x1a0
[  118.153902]  ? handle_bug+0x3c/0x80
[  118.153913]  ? exc_invalid_op+0x17/0x70
[  118.153923]  ? asm_exc_invalid_op+0x1a/0x20
[  118.153938]  ? __pfx_hci_conn_timeout+0x10/0x10 [bluetooth]
[  118.154001]  ? hci_conn_timeout+0x61/0xb0 [bluetooth]
[  118.154055]  process_one_work+0x174/0x340
[  118.154064]  worker_thread+0x27b/0x3a0
[  118.154072]  ? __pfx_worker_thread+0x10/0x10
[  118.154076]  kthread+0xe8/0x120
[  118.154082]  ? __pfx_kthread+0x10/0x10
[  118.154087]  ret_from_fork+0x34/0x50
[  118.154094]  ? __pfx_kthread+0x10/0x10
[  118.154098]  ret_from_fork_asm+0x1b/0x30
[  118.154111]  </TASK>
[  118.154113] ---[ end trace 0000000000000000 ]---
```

Not sure if it's useful or not.

FredsMedia, 2024-03-11:
I have the same issue. Repeatedly trying to connect (_eventually_) showed the pin my Logitech K380 needed to pair and it seems to be working so far. If we can find a proper issue and fix it that would be awesome!

RodoMa92, 2024-03-11:
@Vudentz pairing with my bluetooth buds (which paired fine before) is still broken even with kernel 6.7.9

RodoMa92, 2024-04-07:
Still not working as of 6.7.11 + bluetoothctl 5.73

RodoMa92, 2024-06-04:
As of 5.76 it seems to behave again correctly on pairing, so this can be closed for now.

closed 2024-06-04 by RodoMa92