bluez/bluez #656: Multiple Scanning or Pattern use

p6x0016, 2023-11-18:
Hi I am creating a software that monitors glucose meters and they are around 100 devices. My problem is how I could filter out the devices in two or four devices only using designated MAC addresses or device local names / aliases.

I already made the software but it takes a long time to retrieve the target device advertisement. Please help, I don't have any ideas for solving my issue with my project.

bgix, 2023-11-19:
What kind of advertisements are the glucose meters sending? Are they LE or
are they BR/EDR? Are the advertisements sent with a specific AD type?

p6x0016, 2023-11-20:
Hi bgix, it is LE and in public random type.
The kernel is old 4.19
Buster OS and bluez built in is 5.50

What i want is to search only specific MAC addresses with one RPI for max of 4 devices. 100 devices and 26 RPI are the materials I am using.

I am having trouble in device discovery because it detects all nearby glucose meters and most of them sometimes doesnt retrieve its manufacturerData which the App operation rely on.

Sorry for my english btw.

bgix, 2023-11-20:
The latest Raspberry PI OS kernel v6.1.

There is no way to filter in the BT Controller for multiple MAC addresses.  With the Bluetooth Mesh changes added to v6.1, you can write your own user space app which will forward all received BT LE Advertisements that include an AD type (or list of AD types) of your choice.  This was added for BT Mesh, but there should be no reason you couldn't substitute your own AD Types for the ones used by Mesh.

After that, all MAC address filtering would need to be done by your user space code. Note that this is no less efficient for the Raspberry PI than if it were done in the kernel.  Basically you would be bypassing all of the user space BlueZ daemons in favor of your app talking to the kernel directly.

p6x0016, 2023-11-20:
Thanks bgix for your quick response. One of my last solution is upgrading all the RPI kernels which include bluez update to 5.70

Is it ok to do updating bluez or should I upgrade also the OS?

I am sorry I am still a newbie to linux.
I am thinking to use bluetoothctl pattern if this might help but i dont know yet how to secure only 4 MAC addresses per RPI.

bgix, 2023-11-20:
I would upgrade the entire Raspberry PI OS. kernel v4.19 is old enough that there will almost certainly be incompatibulities with user space pieces between it and v6.1

p6x0016, 2023-12-12:
Sorry for late update here. I already upgraded the OS Kernel version please see it here.

```
  Static hostname: rp01
  Icon name: computer
  Machine ID: b254ffecd33f4abd933898e895c6280d
  Boot ID: 66cde6c546fe487f96dafdc7930221f1
  Operating System: Raspbian GNU/Linux 10 (buster)
  Kernel: Linux 5.10.103-v7l+
  Architecture: arm
```

Although already been upgraded. I don't know why connection is somehow became more messy than older version.
the Bluez version I tried to use are 5.66 and 5.70 but both fail in doing multiple connection and further more it always doing an error of le-connection-abort-by-local

Now I am stuck on how I could do the kernel control then?
Like you said here "With the Bluetooth Mesh changes added to v6.1, you can write your own user space app to request the kernel (BT MGMT)"

I tried btmgmt and do some add devices but still all devices are shown whenever I try to scan.

by the way I am using Tmds.DBus in my RPI application. Current version is 0.14.0

923887863, 2024-01-18:
> Sorry for late update here. I already upgraded the OS Kernel version please see it here.
>
> ```
>   Static hostname: rp01
>   Icon name: computer
>   Machine ID: b254ffecd33f4abd933898e895c6280d
>   Boot ID: [...]

Hi, I have the same problem as you. Have you resolved your problem about 'le-connection-abort-by-local'?

closed 2025-08-04 by github-actions[bot]