bluez/bluez #659: isotest Broadcast Isochronous test

dabeixiaomibujiatang, 2023-11-22:
I'm tried to the isotest , the command of "isotest -s 00:00:00:00:00:00" can send Broadcast Isochronous frames, but it only has one BIS. So i tried command "isotest -s 00:00:00:00:00:00 -N 2" to send two BISs, but it error that "Can't connect: Address already in use". I tried "isotest -s 00:00:00:00:00:00 00:00:00:00:00:00 -N 2" and "isotest -s 00:00:00:00:00:00 00:00:00:00:00:01 -N 2", They still make mistakes. How do I send the correct command to make it have two BIS.

logs:
`# isotest -s 00:00:00:00:00:00
isotest[128]: mgmt socket: fd 3
isotest[129]: Exit
isotest[128]: mgmt_set_le: err 0
@ MGMT Open: isotest (privileged) version 1.22               {0x0002} 13.231536
isotest[128]: mgmt_set_experimental: err 0
@ MGMT Command: Set Low Energy (0x000d) plen 1        {0x0002} [hci0] 13.231562
        Low Energy: Enabled (0x01)
isotest[128]: Connecting 00:00:00:00:00:00 ...
@ MGMT Event: Command Complete (0x0001) plen 7        {0x0002} [hci0] 13.231570
      Set Low Energy (0x000d) plen 4
        Status: Success (0x00)
        Current settings: 0x000c0a81
          Powered
          BR/EDR
          Low Energy
          Secure Connections
          CIS Central
          CIS Peripheral
@ MGMT Command: Set Experimental Feature (0x004a) plen 17    {0x0002} 13.231619
        UUID: BlueZ Experimental ISO Socket
        Action: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 23              {0x0002} 13.231626
      Set Experimental Feature (0x004a) plen 20
        Status: Success (0x00)
        UUID: BlueZ Experimental ISO Socket
        Flags: 0x00000001
          Active
@ MGMT Close: isotest                                        {0x0002} 13.231656
< HCI Command: LE Set Extended Adv.. (0x08|0x0036) plen 25  #1 [hci0] 13.231800
        Handle: 0x01
        Properties: 0x0000
        Min advertising interval: 140.000 msec (0x00e0)
        Max advertising interval: 140.000 msec (0x00e0)
        Channel map: 37, 38, 39 (0x07)
        Own address type: Random (0x01)
        Peer address type: Public (0x00)
        Peer address: 00:00:00:00:00:00 (OUI 00-00-00)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
        TX power: Host has no preference (0x7f)
        Primary PHY: LE 1M (0x01)
        Secondary max skip: 0x00
        Secondary PHY: LE 2M (0x02)
        SID: 0x00
        Scan request notifications: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 5                 #2 [hci0] 13.233288
      LE Set Extended Advertising Parameters (0x08|0x0036) ncmd 1
        Status: Success (0x00)
        TX power (selected): 0 dbm (0x00)
< HCI Command: LE Set Advertising S.. (0x08|0x0035) plen 7  #3 [hci0] 13.233345
        Advertising handle: 0x01
        Advertising random address: 2C:4F:69:2C:9F:38 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                 #4 [hci0] 13.234783
      LE Set Advertising Set Random Address (0x08|0x0035) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Adve.. (0x08|0x0039) plen 6  #5 [hci0] 13.234822
        Extended advertising: Enabled (0x01)
        Number of sets: 1 (0x01)
        Entry 0
          Handle: 0x01
          Duration: 0 ms (0x00)
          Max ext adv events: 0
> HCI Event: Command Complete (0x0e) plen 4                 #6 [hci0] 13.236272
      LE Set Extended Advertising Enable (0x08|0x0039) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Periodic Adve.. (0x08|0x003e) plen 7  #7 [hci0] 13.236308
        Handle: 1
        Min interval: 70.00 msec (0x0038)
        Max interval: 70.00 msec (0x0038)
        Properties: 0x0000
> HCI Event: Command Complete (0x0e) plen 4                 #8 [hci0] 13.237885
      LE Set Periodic Advertising Parameters (0x08|0x003e) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Periodic Adve.. (0x08|0x003f) plen 3  #9 [hci0] 13.237918
        Handle: 1
        Operation: Complete ext advertising data (0x03)
        Data length: 0x00
> HCI Event: Command Complete (0x0e) plen 4                #10 [hci0] 13.239370
      LE Set Periodic Advertising Data (0x08|0x003f) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Periodic Adv.. (0x08|0x0040) plen 2  #11 [hci0] 13.239400
        Periodic advertising: Enabled (0x01)
        Handle: 1
> HCI Event: Command Complete (0x0e) plen 4                #12 [hci0] 13.240929
      LE Set Periodic Advertising Enable (0x08|0x0040) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Broadcas.. (0x08|0x0068) plen 31  #13 [hci0] 13.240962
        Handle: 0x00
        Advertising Handle: 0x01
        Number of BIS: 1
        SDU Interval: 10000 us (0x002710)
        Maximum SDU size: 40
        Maximum Latency: 10 ms (0x000a)
        RTN: 0x02
        PHY: LE 2M (0x02)
        Packing: Sequential (0x00)
        Framing: Unframed (0x00)
        Encryption: 0x00
        Broadcast Code: 00000000000000000000000000000000
> HCI Event: Command Status (0x0f) plen 4                  #14 [hci0] 13.242587
      LE Create Broadcast Isochronous Group (0x08|0x0068) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 21                  #15 [hci0] 13.383183
      LE Broadcast Isochronous Group Complete (0x1b)
        Status: Success (0x00)
        Handle: 0x00
        BIG Synchronization Delay: 912 us (0x000390)
        Transport Latency: 912 us (0x000390)
        PHY: LE 2M (0x02)
        NSE: 3
        BN: 1
        PTO: 0
        IRC: 3
        Maximum PDU: 40
        ISO Interval: 10.00 msec (0x0008)
        Connection Handle #0: 256
< HCI Command: LE Setup Isochrono.. (0x08|0x006e) plen 13  #16 [hci0] 13.383294
        Handle: 256
        Data Path Direction: Input (Host to Controller) (0x00)
        Data Path: HCI (0x00)
        Coding Format: Transparent (0x03)
        Company Codec ID: Ericsson Technology Licensing (0)
        Vendor Codec ID: 0
        Controller Delay: 0 us (0x000000)
        Codec Configuration Length: 0
        Codec Configuration: 
@ RAW Open: btmon (privileged) version 2.22                  {0x0002} 13.383393
@ RAW Close: btmon                                           {0x0002} 13.383405
> HCI Event: Command Complete (0x0e) plen 6                #17 [hci0] 13.384793
isotest[128]: Connected [00:00:00:00:00:00]
      LE Setup Isochronous Data Path (0x08|0x006e) ncmd 1
        Status: Success (0x00)
        Handle: 256`

Vudentz, 2023-11-22:
@dabeixiaomibujiatang support for multiple BISes was merged recently, so perhaps you need to update your kernel to the latest bluetooth-next.

dabeixiaomibujiatang, 2023-11-23:
@Vudentz 
Thanks

closed 2023-11-23 by dabeixiaomibujiatang